Kimplul/hid-tmff2 #88: T300 RS Driver Issue

FelipeMelon, 2024-04-01:
So, I tried updating the driver in windows virtual machine, however I ended up in my wheel being named Thrustmaster T300 RS bootloader and I don't know what to do now. I'd be grateful for any help.

Kimplul, 2024-04-01:
Pretty sure you just need to reattach the wheel to the Windows VM.

FelipeMelon, 2024-04-01:
So, I’ve tried it but I don’t think that the VM recognises that the usb is reattached it only works if I attach it before launching the VM

Kimplul, 2024-04-01:
Interesting. Last time I went through the update procedure the firmware updater waited until I reattached the wheel in its bootloader mode, and then it just finished normally. If you restarted the VM or the update procedure in the middle of this, I guess the installer might've gotten confused? Not sure. Anycase, there's a guide on Thrustmaster's site for what to do if you're stuck in the bootloader mode: https://ts.thrustmaster.com/download/accessories/Manuals/T300RS/T300RS_Bootloader_Method.pdf

FelipeMelon, 2024-04-01:
Alright, I’ll try that. Just in case, is there a way to reset the wheel to factory settings?

Kimplul, 2024-04-01:
As in resetting to the firmware version that the wheel came with in the box? No, not that I'm aware of. The Thrustmaster tools don't seem to have any way to select previous firmware versions.

FelipeMelon, 2024-04-01:
Ok, so it seems like windows VM doesn't recognise the bootloader as a usb device even after switching to different usb ports. It is however recognised on my linux system.

Kimplul, 2024-04-01:
Do you mean that you can't attach the bootloader device to the VM or that Windows/the update utility doesn't recognize the device?

FelipeMelon, 2024-04-01:
I was able to attach it to VM but windows itself didn't recognise it.

Kimplul, 2024-04-01:
Just to clarify, are you referring to the red text in step 7 of the pdf? The firmware update program is not showing anything? Or that the device doesn't show up in device manager under Windows?

Which virtual machine are you using, `virt-manager`? How are you attaching the USB device?

FelipeMelon, 2024-04-01:
I am using virt-manager, I switched from Virtualbox and at least now I am getting an error message when I am trying to redirect usb device (the t300 bootloader) and it says:spice client error quark:Could not redirect USB Device [044f:b66c0] at 1-41: error resetting device LIBUSB_ERROR_NOT_FOUND (0) so I think that's the key problem

Kimplul, 2024-04-01:
> I am using virt-manager, I switched from Virtualbox and at least now I am getting an error message when I am trying to redirect usb device (the t300 bootloader) and it says:spice client error [...]

Seems reasonable. I've personally used `virt-manager` so I know that it at least CAN work, but can't really say I've seen anything like that unfortunately. Googling around I can see some bug reports about something similar, but they've been patched so not entirely sure what this is about. Which distro and version of `virt-manager`/`qemu` are you using?

FelipeMelon, 2024-04-01:
I am using Linux Mint 21.2 Cinnamon, and virt-manager version is 4.0.0

Kimplul, 2024-04-01:
I seem to be running `virt-manager` 4.1.0 on Debian Testing, but I can try out Mint and see if I can replicate.

FelipeMelon, 2024-04-01:
Did you see any progress?

Kimplul, 2024-04-01:
Not yet, as I'm unfortunately not home right now, but will be tomorrow. Probably should've mentioned that, come to think of it.

FelipeMelon, 2024-04-01:
Yeah that's no problem, I have no idea what to do anyway.

FelipeMelon, 2024-04-02:
I have solved the issue, by downloading Windows 10 on a separate drive and getting the firmware over there. No solution found for the VM problem though.

closed 2024-04-02 by FelipeMelon

Kimplul, 2024-04-02:
Very pragmatic.

Unfortunately, I wasn't able to replicate, so really not sure what the VM issue might be. Could be worth opening a bug report for?

FelipeMelon, 2024-04-02:
I'll submit a bug-report relatively soon, I'm taking a break after troubleshooting for a whole day. I really hope it will help some ppl in the future. :)