Kimplul/hid-tmff2 #89: T300 RS GT Buttons Not Working

FelipeMelon, 2024-04-02:
So, I have went with the issue to the Oversteer git but they weren't able to resolve it.
Here's what I've learned

This is what happens after I run evtest and select the t300:
nput driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x44f product 0xb66e version 0x111
Input device name: "Thrustmaster Thrustmaster T300RS Racing wheel"
Supported events:
Event type 0 (EV_SYN)
Event type 1 (EV_KEY)
Event code 288 (BTN_TRIGGER)
Event code 289 (BTN_THUMB)
Event code 290 (BTN_THUMB2)
Event code 291 (BTN_TOP)
Event code 292 (BTN_TOP2)
Event code 293 (BTN_PINKIE)
Event code 294 (BTN_BASE)
Event code 295 (BTN_BASE2)
Event code 296 (BTN_BASE3)
Event code 297 (BTN_BASE4)
Event code 298 (BTN_BASE5)
Event code 299 (BTN_BASE6)
Event code 300 (?)
Event type 3 (EV_ABS)
Event code 0 (ABS_X)
Value 32812
Min 0
Max 65535
Fuzz 255
Event code 1 (ABS_Y)
Value 1023
Min 0
Max 1023
Fuzz 3
Event code 2 (ABS_Z)
Value 1021
Min 0
Max 1023
Fuzz 3
Event code 5 (ABS_RZ)
Value 1021
Min 0
Max 1023
Fuzz 3
Event code 16 (ABS_HAT0X)
Value 0
Min -1
Max 1
Event code 17 (ABS_HAT0Y)
Value 0
Min -1
Max 1
Event type 4 (EV_MSC)
Event code 4 (MSC_SCAN)
Event type 21 (EV_FF)
Event code 80 (FF_RUMBLE)
Event code 81 (FF_PERIODIC)
Event code 82 (FF_CONSTANT)
Event code 83 (FF_SPRING)
Event code 84 (FF_FRICTION)
Event code 85 (FF_DAMPER)
Event code 86 (FF_INERTIA)
Event code 87 (FF_RAMP)
Event code 88 (FF_SQUARE)
Event code 89 (FF_TRIANGLE)
Event code 90 (FF_SINE)
Event code 91 (FF_SAW_UP)
Event code 92 (FF_SAW_DOWN)
Event code 96 (FF_GAIN)
Event code 97 (FF_AUTOCENTER)

Buttons:
After clicking L3:
Event: time 1712072433.914725, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000b
Event: time 1712072433.914725, type 1 (EV_KEY), code 298 (BTN_BASE5), value 1
Event: time 1712072433.914725, -------------- SYN_REPORT ------------
Event: time 1712072434.007721, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000b
Event: time 1712072434.007721, type 1 (EV_KEY), code 298 (BTN_BASE5), value 0
Event: time 1712072434.007721, -------------- SYN_REPORT ------------

After clicking R3:
Event: time 1712072472.800726, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000c
Event: time 1712072472.800726, type 1 (EV_KEY), code 299 (BTN_BASE6), value 1
Event: time 1712072472.800726, -------------- SYN_REPORT ------------
Event: time 1712072472.916729, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000c
Event: time 1712072472.916729, type 1 (EV_KEY), code 299 (BTN_BASE6), value 0
Event: time 1712072472.916729, -------------- SYN_REPORT ------------

No other buttons show anything.

Kimplul, 2024-04-02:
Wow, you seem to be really unlucky.

I haven't run into just a couple buttons working before, usually it's more all-or-nothing.
There is technically support for the PS4 mode, and I would be curious to know what happens if you try it out. Unplug/replug the wheel, maybe even restart your computer to be sure the mode switch is complete.

Assuming the hardware isn't broken or the steering wheel is not properly installed or something, the only real reason I can think of as to why just a couple buttons work would be that the `rdesc` that this driver has doesn't match your T300 for whatever reason. I only have the RS, not the GT, so technically speaking different wheels, but other people seem to have gotten the GT to work and not entirely sure why that would be the issue. I updated the firmware of my wheel, but it seems to have stayed at version 34, so I don't think it's an issue of a new firmware mucking things up.

FelipeMelon, 2024-04-02:
Changing into PS4/PS5 mode makes only wheel turning and ffb and the two buttons work. Oversteer also stops recognising it.

Kimplul, 2024-04-03:
Alright, interesting. Thanks for the info, sounds more or less expected. I guess the obvious follow up question is did all buttons work on Windows?

Next step would likely be to check that the wheel is actually sending out updates for the buttons. Try running `sudo usbhid-dump --address X:Y --entity=all`, where `X` is the bus and `Y` is the device number for the wheel. You can get `X` and `Y` from `lsusb`. You should see a bunch of hex numbers, and when you press buttons, some of those numbers should change. For now it doesn't really matter what the numbers mean, but check that all buttons change some number. This should at least tell us at which end of the USB cable the problem occurs. 

I'll try out the wheel a bit in my Mint VM and see if I can replicate, probably tomorrow.

FelipeMelon, 2024-04-03:
Pressing the buttons except the ones I stated that work, doesn't change any of the numbers. It's the same on windows.

Kimplul, 2024-04-03:
That indicates the wheel itself isn't sending any input. Maybe try reinstalling the firmware and reseating the steering wheel, could be due to poor contact. Is it a new wheel?

FelipeMelon, 2024-04-04:
Yes, I managed to fix it by reseating the wheel, seems like I didn't push until the very end with the wheel! I'm so glad it wasn't a software issue.

closed 2024-04-04 by FelipeMelon

FelipeMelon, 2024-04-04:
On the sidenote, is it possible to solve the issue of ets2 on proton only recognising the wheel but not the shifter (as in the input section)? I can always switch to native with which there are no problems but I sometimes need DX11 for map mods.

Kimplul, 2024-04-04:
Not that I'm aware of, unfortunately. Multiple input devices seem to sometimes be rather confusing for some games, I guess we can add `ets2` to the list. All advice I can give is to try out different settings in game. Some games seem to also benefit from https://github.com/Kimplul/hid-tmff2/wiki#installing-the-windows-drivers-manually, could be worth a shot.

FelipeMelon, 2024-04-04:
Yeah, I'll let it go, I don't need it that much.