Kimplul/hid-tmff2 #93: Ferrari SF1000 Edition Wheel

Wanama78, 2024-04-14:
Hi, I use the SF1000 wheel on the T300RS.
Now some buttons don't work with the driver.

I uploaded some Wireshark data, with a representation of the wheel and the buttons,
which have no function.
Which can be found here:
[(https://github.com/Wanama78/tm300rs_adv_SF1000)](https://github.com/Wanama78/tm300rs_adv_SF1000)

I could do the detection in "tminit {0x02, 0x0a, 0x0005, "Thrustmaster T300 Ferrari SF1000 Edition"}",
Installing drivers, that's about it as far as my options for doing more with a driver are concerned.

At this point I would like to thank you for your work on this driver, it's a lot of fun to use it.

If further data is needed to solve the problem, I will try to provide it.

[image]

Kimplul, 2024-04-14:
Hello, thanks for the captures. With some luck, this is probably just a case of adding another fixed `rdesc` like the ones here: https://github.com/Kimplul/hid-tmff2/blob/414f1f03f1ef9d51829d10b92e03bf1bca3bb966/src/tmt300rs/hid-tmt300rs.c#L87

I would still need the output of `lsusb`, specifically the `ID` of the wheel, and the `rdesc` of the wheel.

If the attachment works the same as other ones encountered so far, you should be able to find it with
```
cat /sys/kernel/debug/hid/XXXX\:VID\:PID.YYYY/rdesc
```

The `ID` part of `lsusb` gives `VID:PID`, which should help you find the file. `XXXX/YYYY` are some constants that may change from boot to boot, so you may have to look around a bit unfortunately.

Wanama78, 2024-04-14:
Thanks for the quick reply.
Is this the data you need?

rdesc:

> 05 01 09 05 a1 01 15 00 25 01 35 00 45 01 75 01 95 0d 05 09 19 01 29 0d 81 02 95 03 81 01 05 01 25 07 46 3b 01 75 04 95 01 65 14 09 39 81 42 05 01 09 30 15 00 26 ff 0f 35 00 46 ff 0f 65 00 75 0c 95 01 81 02 26 ff 00 46 ff 00 09 31 09 32 09 35 75 08 95 03 81 02 06 00 ff 09 20 09 21 09 22 09 23 09 24 09 25 09 26 09 27 09 28 09 29 09 2a 09 2b 95 0c 81 02 0a 21 26 95 08 b1 02 0a 21 26 91 02 26 ff 03 46 ff 03 09 2c 09 2d 09 2e 09 2f 75 10 95 04 81 02 c0

ID 044f:b65d ThrustMaster, Inc. Thrustmaster FFB Wheel

Kimplul, 2024-04-14:
Good catch, I was thinking more about the case with drivers. With the F1 attachment, there is a way to enter an 'advanced mode' with the wheel that enabled extra features of the attachment if I recall correctly, does that also apply to this attachment? I'm asking because

> ID 044f:b66e ThrustMaster, Inc. Thrustmaster T300RS Racing wheel

`b66e` is the same `PID` as the base T300 in PS3 mode, which makes it a bit more tricky to separate a wheel without an attachment (well, with the default attachment at least) from one without one. There is a USB query command that we can use, but at least currently it is only accessible after the `rdesc` tweak.

Wanama78, 2024-04-14:
So I didn't have to set this when I tested it in Windows, the display is then also controlled there, which I only managed to do in Linux with F1-21 via UDP.

Kimplul, 2024-04-14:
Oh okay, in that case the `rdesc` that you captured is just https://github.com/Kimplul/hid-tmff2/blob/414f1f03f1ef9d51829d10b92e03bf1bca3bb966/src/tmt300rs/hid-tmt300rs.c#L87 i.e. not the one we want.

Could you capture the USB traffic of plugging in the wheel? It should send over the `rdesc` we're interested in as part of the HID initialization transfers, something like `SETUP ...` in Wireshark. As you've probably already noticed, plugging in the wheel is a two-stage process, with the `init` stage kicking the wheel to start again but properly this time. In this case we're interested in the second stage traffic.

Wanama78, 2024-04-14:
I haven't shortened it now.

[https://github.com/Wanama78/tm300rs_adv_SF1000/blob/main/init_wheel.pcapng](https://github.com/Wanama78/tm300rs_adv_SF1000/blob/main/init_wheel.pcapng)

Wanama78, 2024-04-15:
Here you look at it, steering wheel integrated into Linux without a driver, nothing works, I found it strange, I was sure that I could control all the buttons, even without a driver.

Then I remembered that I had already tested it and Windows was running in the VM and I had only started the first stage of detection.

lsusb now shows:
ID 044f:b696 ThrustMaster, Inc. Thrustmaster Advanced Mode Racer

what the switch is, possible?

And rdesc, throws this out:

> 05 01 09 04 a1 01 09 01 a1 00 85 07 09 30 15 00 27 ff ff 00 00 35 00 47 ff ff 00 00 75 10 95 01 81 02 09 31 26 ff 03 46 ff 03 81 02 09 35 81 02 09 36 81 02 75 08 26 ff 00 46 ff 00 09 40 81 02 09 41 81 02 09 33 81 02 09 34 81 02 09 32 81 02 09 37 81 02 05 09 19 01 29 1a 25 01 45 01 75 01 95 1a 81 02 75 06 95 01 81 03 05 01 09 39 25 07 46 3b 01 55 00 65 14 75 04 81 42 65 00 81 03 85 0a 06 00 ff 09 0a 75 08 95 3f 26 ff 00 46 ff 00 91 02 85 02 09 02 81 02 09 14 85 14 81 02 c0 c0

plus all the other stuff that follows underneath.

Yes, I also made a Wireshark recording, i.e.:
Starts at 233.
[https://github.com/Wanama78/tm300rs_adv_SF1000/raw/main/init_wheel_new.pcapng](https://github.com/Wanama78/tm300rs_adv_SF1000/raw/main/init_wheel_new.pcapng)

ID 044f:b65d ThrustMaster, Inc. Thrustmaster FFB Wheel
to:
ID 044f:b696 ThrustMaster, Inc. Thrustmaster Advanced Mode Racer

It can also be overridden, now controlling every key except for
RX and RY – axis, which work in ACC.

Maybe this will help

Kimplul, 2024-04-15:
> Here you look at it, steering wheel integrated into Linux without a driver, nothing works, I found it strange, I was sure that I could control all the buttons, even without a driver.

If I understand you correctly, you mean that without this driver installed, all buttons showed up? Yep, I guess that's expected. The `rdesc` is a description of what kind of input data the wheel is going to send to the computer, and contains stuff like 'I have x buttons, please map them to buttons 0, 1, 2, 3 ...`. The `rdesc` also happens to contain information for sending FFB commands to the wheel, though with Thrustmaster wheels they're in a non-standard set, which means that this driver modifies the `rdesc` to get FFB working. This is why we needed the wheel's unmodified `rdesc` for the buttons.

 > what the switch is, possible?

(Again, if I understand you correctly) Yeah, sounds like the switch that the F1 attachment also requires to enable the wheel's advanced features. Not entirely sure how you activated the mode, have you looked at your wheel's manual?

> It can also be Oversteer, now controlling every key except for
RX and RY – axis, which work in ACC.

Sounds like an Oversteer issue, I guess they haven't seen a wheel use those axes before.

> Yes, I also made a Wireshark recording, i.e.:
> Starts at 233.
> https://github.com/Wanama78/tm300rs_adv_SF1000/raw/main/init_wheel_new.pcapng
>
> ID 044f:b65d ThrustMaster, Inc. Thrustmaster FFB Wheel
> to:
> ID 044f:b696 ThrustMaster, Inc. Thrustmaster Advanced Mode Racer

`b696` is the PID used by the T248, and interestingly, the `rdesc` I fished out of your captures (thanks again) is the exact same as the one the T248 uses. As such, I would expect this driver to work with your wheel in the `Advanced Mode Racer` mode, the driver just thinks it's dealing with a T248. Does force feedback work?

Kimplul, 2024-04-15:
Oh, by the way, could you merge your changes in `hid-tminit` to https://github.com/scarburato/hid-tminit? Would be useful for other people with the attachment.

Wanama78, 2024-04-15:
Yes, force feedback works, but the keys in question don't work with your drivers.

Of course, nothing works in Linux without a driver, and the entire steering wheel is not recognized.

If I have a Windows VM running and carry out the first recognition there, I get all the buttons and the steering wheel recognized under Linux.

But what isn't there is force feedback, because there is no driver.

Kimplul, 2024-04-15:
So under Linux, with this driver installed, and your wheel in the `Advanced Mode Racer` mode, force feedback works, but not the buttons? What does `dmesg` show when you attach the `Advanced Mode Racer` wheel?

Kimplul, 2024-04-15:
> tmff2 0003:044F:B66E.0016: force feedback for T300RS

Your wheel isn't in the 'Advanced Mode Racer' mode, which what we're currently mostly interested in.

Wanama78, 2024-04-15:
I recorded it like I mean it.

[init wheel Youtube](https://youtu.be/CD1F1t6yyU0)

Kimplul, 2024-04-15:
Just to check, did you have this driver installed when you recorded that? This driver should accept a PID of `b696`, but doesn't seem like it's being probed for whatever reason.

Wanama78, 2024-04-16:
The driver was blacklisted by me.
As shown in the video, the driver is not loaded either.
Loading starts from the Windows VM.

As I described it here.
[https://github.com/Kimplul/hid-tmff2/issues/93#issuecomment-2057649842](https://github.com/Kimplul/hid-tmff2/issues/93#issuecomment-2057649842)

Kimplul, 2024-04-16:
> It seems like the detection you initially proposed ("tminit {0x02, 0x0a, 0x0005, "Thrustmaster T300 Ferrari SF1000 Edition"}") might not be the one we want to use, I'll look through your captures again and see if I can spot something.

Yeah, I think `0x0005` should actually be `0x000b`, packet number 184 in https://github.com/Wanama78/tm300rs_adv_SF1000/raw/main/init_wheel_new.pcapng uses it. Try making that change and see what happens.

Wanama78, 2024-04-16:
> Oh, okay, sorry, in that case I'm afraid I misunderstood your comment thoroughly. 

Everything is fine, I translate in Google, sometimes nothing useful comes out.

> Also, I ask again,

I have left an entry in the hid_tminit project.

hid_thrustmaster; hid_tminit; hid_tmff_new, sind alle auf der Schwarzen Liste

> So do the initialization in Windows, then bring the Advanced Mode Racer over to Linux. From the information I've seen the wheel should work, with buttons and force feedback.

Test Windows drivers, then exit Windows again.
Nothing, no force feedback, buttons yes.
Linux hid-tmff-new just loaded, nothing happens.
Linux hid-tminit and hid-tmff-new loaded, then hid-tminit overwrites everything.

> Yeah, I think 0x0005 should actually be 0x000b, packet number 184 in https://github.com/Wanama78/tm300rs_adv_SF1000/raw/main/init_wheel_new.pcapng uses it. Try making that change and see what happens.

Then the steering wheel is no longer recognized.

I also tried this in hid-tminit:

> static const struct hid_device_id thrustmaster_devices[] = {
>	{ HID_USB_DEVICE(0x044f, 0xb65d)},
>	{ HID_USB_DEVICE(0x044f, 0xb696)},
>	{}
> };

Unfortunately nothing, except after the first Windows init.

Wanama78, 2024-04-16:
Ok, if the hid-tmff-new module is loaded before hid-tminit, what I did in Windows can also be done in Linux.

```shell
[ 7537.965573] usb 3-2: new full-speed USB device number 25 using xhci_hcd
[ 7538.125665] usb 3-2: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 7538.125669] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7538.125671] usb 3-2: Product: Thrustmaster FFB Wheel
[ 7538.125672] usb 3-2: Manufacturer: Thrustmaster
``` 

I now load hid-tminit:

```shell
[ 7724.107000] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-2/3-2:1.0/0003:044F:B65D.00A6/input/input195
[ 7724.107096] hid-thrustmaster 0003:044F:B65D.00A6: input,hidraw6: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:2a:00.3-2/input0
[ 7724.130080] hid-thrustmaster 0003:044F:B65D.00A6: Wheel with (model, attachment) = (0x2, 0xa) is a Thrustmaster T300 Ferrari SF1000 Edition. attachment_found=1
[ 7724.131233] usb 3-2: USB disconnect, device number 25
[ 7724.134147] hid-thrustmaster 0003:044F:B65D.00A6: Success?! The wheel should have been initialized!
[ 7724.794902] usb 3-2: new full-speed USB device number 26 using xhci_hcd
[ 7724.961076] usb 3-2: New USB device found, idVendor=044f, idProduct=b696, bcdDevice= 7.00
[ 7724.961080] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7724.961082] usb 3-2: Product: Thrustmaster Advanced Mode Racer
[ 7724.961083] usb 3-2: Manufacturer: Thrustmaster
[ 7724.979244] input: Thrustmaster Thrustmaster Advanced Mode Racer as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-2/3-2:1.0/0003:044F:B696.00A7/input/input196
[ 7724.979349] tmff2 0003:044F:B696.00A7: input,hidraw6: USB HID v1.11 Joystick [Thrustmaster Thrustmaster Advanced Mode Racer] on usb-0000:2a:00.3-2/input0
[ 7724.993118] tmff2 0003:044F:B696.00A7: force feedback for T248
``` 

However, the driver is not loaded correctly.
Unfortunately I cannot unload hid-tmff-new.
Then the steering wheel starts doing strange things.

Kimplul, 2024-04-16:
Alright, I'll try to be as clear as I can, hopefully Google Translate can translate this properly. Thrustmaster wheels are weird, because they have a two-stage bootup. In the first stage, they all show up as a generic FFB device, with USB PID `0xb65d`. That's the same `0xb65d` that you see in `hid-tminit`. `hid-tminit` checks which wheel it actually is, and sends a command to the wheel to reboot as the correct wheel. `hid-tminit` does NOT have any force feedback support, all force feedback is handled by this module, `hid-tmff-new`.

Please don't blacklist/unblacklist modules, it makes it very difficult for me to follow what's going on. You should have `hid_thrustmaster` blacklisted, as it might get in the way of testing, but that's the exception.

Please explain clearly what you mean by

> Then the steering wheel is no longer recognized.

Changing `0x0005` to `0x000b` should not have any effect on if the wheel is 'being recognized', as it just changes what command gets sent out to the wheel during initialization, so I'm very confused by this. Is there any output in `dmesg`? Did you forget to ublacklist `hid-tminit`?

---

I don't want to sound rude, I appreciate that you're participating in this project, but I do need to understand what you're trying to say, and I feel like Google Translate is letting us both down in this regard. Similarly, I'm afraid I need more specific descriptions.

> However, the driver is not loaded correctly.

What do you mean? How would the driver be loaded correctly? Is `dmesg` showing something?

> Unfortunately I cannot unload hid-tmff-new.

Why do you want to unload `hid-tmff-new`?

> Then the steering wheel starts doing strange things.

What things? Is it vibrating?

Wanama78, 2024-04-16:
> Alright, I'll try to be as clear as I can, hopefully Google Translate can translate this properly.

I'll try another program!

> Thrustmaster wheels are weird, because they have a two-stage bootup.

I understand that the steering wheel starts in 2 steps and hid-tminit takes over.

> hid-tminit checks which wheel it actually is, and sends a command to the wheel to reboot as > the correct wheel. hid-tminit does NOT have any force feedback support, all force feedback > is handled by this module, hid-tmff-new.

I understand what the individual drivers do, roughly.

> Please don't blacklist/unblacklist modules, it makes it very difficult for me to follow what's going > on. You should have hid_thrustmaster blacklisted, as it might get in the way of testing, but > that's the exception.

The hid-thrustmaster is always on my blacklist.
The two hid-tminit and hid-tmff-new are not on the blacklist.
Only when I test something in Windows as shown.

> Changing 0x0005 to 0x000b should not have any effect on if the wheel is 'being recognized', as it just changes what command gets sent out to the wheel during initialization, so I'm very confused by this.

If I set the value to 0x000b, only the hid-tminit is loaded but not hid-tmff-new.
This is then output.

```shell
[15575.401557] usb 3-2: new full-speed USB device number 57 using xhci_hcd
[15575.558693] usb 3-2: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[15575.558698] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[15575.558701] usb 3-2: Product: Thrustmaster FFB Wheel
[15575.558703] usb 3-2: Manufacturer: Thrustmaster
[15575.591896] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-2/3-2:1.0/0003:044F:B65D.00C5/input/input229
[15575.591985] hid-thrustmaster 0003:044F:B65D.00C5: input,hidraw6: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:2a:00.3-2/input0
[15575.613692] hid-thrustmaster 0003:044F:B65D.00C5: Wheel with (model, attachment) = (0x2, 0xa) is a Thrustmaster T300 Ferrari SF1000 Edition. attachment_found=1
``` 

I think we will manage it.

Kimplul, 2024-04-16:
> If I set the value to 0x000b, only the hid-tminit is loaded but not hid-tmff-new.

Okay, good, thank you for the clarification. What does `lsusb` show after `hid-tminit` is loaded?

Wanama78, 2024-04-16:
Here you can see what is loaded.

```shell
lsmod | grep hid_t
hid_tminit             16384  0
usbhid                 77824  1 hid_tminit
``` 

lsusb:

> ID 044f:b65d ThrustMaster, Inc. Thrustmaster FFB Wheel

Kimplul, 2024-04-16:
Is the wheel in some kind of bootloop maybe? What does `dmesg` show? You can follow it for a bit with `dmesg -w`, if the wheel is bootlooping you should see some repeated messages.

Wanama78, 2024-04-16:
No, nothing seems to happen anymore.

If I switch back to 0x0005, everything loads correctly again.

Kimplul, 2024-04-16:
Alright, let's take a step back. There are a couple questions you didn't answer a bit earlier, let's get back to them now.

When you got the following `dmesg` output:

> [ 7724.107000] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-2/3-2:1.0/0003:044F:B65D.00A6/input/input195
> [ [...]

What did you do to get this kind of output? Can you get back to this state?
This output is what I'm expecting to see, the wheel booting into the `Advanced ...` mode, and it being picked up as a T248. That matches with my understanding of the situation, pretty much everything else currently doesn't.

I had these questions earlier, please answer them.

> However, the driver is not loaded correctly.

What do you mean? How would the driver be loaded correctly? Is `dmesg` showing something? (please include `dmesg` output if it is)

> Unfortunately I cannot unload hid-tmff-new.

Why do you want to unload hid-tmff-new?

> Then the steering wheel starts doing strange things.

What things? Is it vibrating?

Wanama78, 2024-04-17:
> Alright, let's take a step back. There are a couple questions you didn't answer a bit earlier, let's get back to them now.
> When you got the following dmesg output:

> What did you do to get this kind of output? Can you get back to this state?
This output is what I'm expecting to see, the wheel booting into the Advanced ... mode, and it being picked up as a T248. That matches with my understanding of the situation, pretty much everything else currently doesn't.

In this case I tested whether hid-tmff-new can be loaded before hid-tminit.
The hid-tminit is then executed but not the hid-tmff-new.

If I now unload hid-tmff-new again, the steering wheel is gradually turned to the right on its own. This is the strange behavior I was referring to.

This is what is shown to me in this test.
```shell
[ 7724.107000] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-2/3-2:1.0/0003:044F:B65D.00A6/input/input195
[ 7724.107096] hid-thrustmaster 0003:044F:B65D.00A6: input,hidraw6: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:2a:00.3-2/input0
[ 7724.130080] hid-thrustmaster 0003:044F:B65D.00A6: Wheel with (model, attachment) = (0x2, 0xa) is a Thrustmaster T300 Ferrari SF1000 Edition. attachment_found=1
[ 7724.131233] usb 3-2: USB disconnect, device number 25
[ 7724.134147] hid-thrustmaster 0003:044F:B65D.00A6: Success?! The wheel should have been initialized!
[ 7724.794902] usb 3-2: new full-speed USB device number 26 using xhci_hcd
[ 7724.961076] usb 3-2: New USB device found, idVendor=044f, idProduct=b696, bcdDevice= 7.00
[ 7724.961080] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7724.961082] usb 3-2: Product: Thrustmaster Advanced Mode Racer
[ 7724.961083] usb 3-2: Manufacturer: Thrustmaster
[ 7724.979244] input: Thrustmaster Thrustmaster Advanced Mode Racer as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-2/3-2:1.0/0003:044F:B696.00A7/input/input196
[ 7724.979349] tmff2 0003:044F:B696.00A7: input,hidraw6: USB HID v1.11 Joystick [Thrustmaster Thrustmaster Advanced Mode Racer] on usb-0000:2a:00.3-2/input0
[ 7724.993118] tmff2 0003:044F:B696.00A7: force feedback for T248
``` 

That's all there is to it, I'll test it again tonight.

Wanama78, 2024-04-17:
As long as the hid-tminit does not work properly, I cannot switch to extended mode.
With 0x0005 I remain in normal mode and the extended data cannot be loaded.
With 0x000b only the detection in hid-tminit is done but not switched to extended mode.
And therefore the hid-tmff-new is not loaded.

Here are my adjustments to the SF1000 wheel.
If I apply this with 0x0005, the Bootloader becomes active.
I then have to switch it off again in Windows.

``` c
static u8 t300rs_rdesc_sf1000_fixed[] = {
0x05, 0x01, /* Usage page (Generic Desktop) */
0x09, 0x04, /* Usage (Joystick) */
0xa1, 0x01, /* Collection (Header Application) */
0x09, 0x01, /* Usage (Pointer) */
0xa1, 0x00, /* Collection (Physical) */
0x85, 0x07, /* Report ID (7) */
0x09, 0x30, /* Usage (X) */
0x15, 0x00, /* Logical minimum (0) */
0x27, 0xff, 0xff, 0x00, 0x00, /* Logical maximum (65535) */
0x35, 0x00, /* Physical minimum (0) */
0x47, 0xff, 0xff, 0x00, 0x00, /* Physical maximum (65535) */
0x75, 0x10, /* Report size (16) */
0x95, 0x01, /* Report count (1) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x09, 0x31, /* Usage (Y) */
0x26, 0xff, 0x03, /* Logical maximum (1023) */
0x46, 0xff, 0x03, /* Physical maximum (1023) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x09, 0x35, /* Usage (Rz) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x09, 0x36, /* Usage (Slider) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x75, 0x08, /* Report size (8) */
0x26, 0xff, 0x00, /* Logical Maximum (255) */
0x46, 0xff, 0x00, /* Physical Maximum (255) */
0x09, 0x40, /* Usage (Vx) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x09, 0x41, /* Usage (Vy) */ //Wanama
0x81, 0x02, /* Input (Variable, Absolute) */
0x09, 0x33, /* Usage (Rx) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x09, 0x34, /* Usage (Ry) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x09, 0x32, /* Usage (Z) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x09, 0x37, /* Usage (Dial) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x05, 0x09, /* Usage Page (Button) */
0x19, 0x01, /* Usage Minimum (0x01) */
0x29, 0x1a, /* Usage Minimum (0x1a) */
0x25, 0x01, /* Logical Maximum (1) */
0x45, 0x01, /* Physical Maximum (1) */
0x75, 0x01, /* Report size (1) */
0x95, 0x1a, /* Report Count (26) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x75, 0x06, /* Report size (6) */
0x95, 0x01, /* Report Count (1) */
0x81, 0x03, /* Input (Variable, Absolute, Constant) ? */
0x05, 0x01, /* Usage Page (Generic Desktop Controls) */
0x09, 0x39, /* Usage (Hat switch) */
0x25, 0x07, /* Logical Maximum (7) */
0x46, 0x3b, 0x01, /* Physical Maximum (315) */
0x55, 0x00, /* Unit Exponent (-16) */
0x65, 0x14, /* Unit (Eng Rot, Angular Pos) */
0x75, 0x04, /* Report size (4) */
0x81, 0x42, /* Input (Variable, Absolute, NullState) */
0x65, 0x00, /* Unit (0x00) */
0x81, 0x03, /* Input (Variable, Absolute, Constant) */
0x85, 0x0a, /* Report ID (0x0a) */
0x06, 0x00, 0xff, /* Usage Page (Vendor) */
0x09, 0x0a, /* Usage (Vendor) */
0x75, 0x08, /* Report size (8) */
0x95, 0x3f, /* Report count (63) */
0x26, 0xff, 0x00, /* Logical maximum (255) */
0x46, 0xff, 0x00, /* Physical maximum (255) */
0x91, 0x02, /* Output (Variable, Absolute) */
0x85, 0x02, /* Report ID (2) */
0x09, 0x02, /* Usage (Vendor) */
0x81, 0x02, /* Input (Variable, Absolute) */
0x09, 0x14, /* Usage (Vendor) */
0x85, 0x14, /* Report ID (0x14) */
0x81, 0x02, /* Input (Variable, Absolute) */
0xc0, /* End collection */
0xc0, /* End collection */
};
```

Kimplul, 2024-04-17:
> If I apply this with 0x0005, the Bootloader becomes active.

This generally means that the firmware on the wheel crashed, which I guess I'm not entirely surprised about. If I'm understanding you correctly, you've taken the `rdesc` of the wheel in advanced mode and applied it to the T300 mode. I don't really know if this means that the HID subsystem of Linux tries to send a command to the wheel that then crashes it or if it's something else, but I don't think that strategy is going to work. You need to use the `rdesc` of the 'advanced ...' mode with the wheel in its `advanced ...` mode, as the wheel itself is *very* dumb and you can tell it to boot as a T150 if you want to (I tried it on my wheel once, the wheel crashed into the bootloader after a while if I recall correctly).

As I see it, we really do need to get the wheel into the `advanced ...` mode. I have no idea why `0x000b` isn't working for you, I've looked through the captures and there's nothing that stands out to me, so I don't understand why it wouldn't boot into the `advanced ...` mode.

What's more, I'm getting contradictory messages here. Maybe it's your translator or something, no clue, but when you say

> With 0x0005 I remain in normal mode and the extended data cannot be loaded.
> With 0x000b only the detection in hid-tminit is done but not switched to extended mode.

that clearly doesn't seem to be the case, as you provided a log of the wheel entering into `advanced ...` mode earlier:

> [ 7724.107000] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-2/3-2:1.0/0003:044F:B65D.00A6/input/input195
[ 7724.107096] hid-thrustmaster 0003:044F:B65D.00A6: input,hidraw6: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:2a:00.3-2/input0
[ 7724.130080] hid-thrustmaster 0003:044F:B65D.00A6: Wheel with (model, attachment) = (0x2, 0xa) is a Thrustmaster T300 Ferrari SF1000 Edition. attachment_found=1
[ 7724.131233] usb 3-2: USB disconnect, device number 25
[ 7724.134147] hid-thrustmaster 0003:044F:B65D.00A6: Success?! The wheel should have been initialized!
[ 7724.794902] usb 3-2: new full-speed USB device number 26 using xhci_hcd
[ 7724.961076] usb 3-2: New USB device found, idVendor=044f, idProduct=b696, bcdDevice= 7.00
[ 7724.961080] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7724.961082] usb 3-2: Product: Thrustmaster Advanced Mode Racer
[ 7724.961083] usb 3-2: Manufacturer: Thrustmaster
[ 7724.979244] input: Thrustmaster Thrustmaster Advanced Mode Racer as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-2/3-2:1.0/0003:044F:B696.00A7/input/input196
[ 7724.979349] tmff2 0003:044F:B696.00A7: input,hidraw6: USB HID v1.11 Joystick [Thrustmaster Thrustmaster Advanced Mode Racer] on usb-0000:2a:00.3-2/input0
[ 7724.993118] tmff2 0003:044F:B696.00A7: force feedback for T248

This log shows `hid-tminit` (which shows up in `dmesg` as `hid-thrustmaster`, very confusing, I know) booting the wheel into the `advanced ...` mode. So there's something that's getting lost in transmission.

Wanama78, 2024-04-17:
> that clearly doesn't seem to be the case, as you provided a log of the wheel entering into advanced ... mode earlier:

Yes only because I loaded hidtmff-new before hid-tminit , which does not initialize the wheel completely.

The hid-tmff-new then no longer applies, as you can see here.

> [ 7724.993118] tmff2 0003:044F:B696.00A7: force feedback for T248

Although force feedback is written, it does not exist.

Kimplul, 2024-04-17:
> Yes only because I loaded hidtmff-new before hid-tminit , which does not initialize the wheel completely.

`hid-tmff-new` doesn't do any initialization, it doesn't care about `0xb65d`, manually loading drivers doesn't do anything as Linux will automatically load available drivers that match the device. Loading `hid-tmff-new` before `hid-tminit` doesn't really do anything, it just moves `hid-tmff-new` into the kernel to wait for a device to show up.

> The hid-tmff-new then no longer applies, as you can see here.

`tmff2` is what `hid-tmff-new` shows up as in `dmesg`. Historical reasons. So you're looking at `hid-tmff-new` output.

Wanama78, 2024-04-19:
Am I right in assuming that you no longer have the T300RS yourself?

Kimplul, 2024-04-19:
No, I still have my T300. I don't have any attachments to it, though.

Wanama78, 2024-04-21:
> No, I still have my T300. I don't have any attachments to it, though.

Ok, that's a pity, would have made it easier.

So I put on the standard wheel and recorded the init in Windows.
[GT_init_wheel_win Wireshark](https://github.com/Wanama78/tm300rs_adv_SF1000/raw/main/GT_init_wheel_win.pcapng)

Here is also the init to the SF1000 wheel.
[SF1000_init_wheel_win Wireshark](https://github.com/Wanama78/tm300rs_adv_SF1000/raw/main/SF1000_init_wheel_win.pcapng)

I have adapted the driver so that I only get the hid-tminit and the wheel no longer gets stuck in the bootloader.

This is just to test if the hid-tminit is working properly.
I think I need to extend this section in hid-tminit.c.

But what exactly goes in there, no plan.
```c
static const u8 setup_0[] = { 0x42, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
static const u8 setup_1[] = { 0x0a, 0x04, 0x90, 0x03, 0x00, 0x00, 0x00, 0x00 };
static const u8 setup_2[] = { 0x0a, 0x04, 0x00, 0x0c, 0x00, 0x00, 0x00, 0x00 };
static const u8 setup_3[] = { 0x0a, 0x04, 0x12, 0x10, 0x00, 0x00, 0x00, 0x00 };
static const u8 setup_4[] = { 0x0a, 0x04, 0x00, 0x06, 0x00, 0x00, 0x00, 0x00 };
static const u8 *const setup_arr[] = { setup_0, setup_1, setup_2, setup_3, setup_4 };
static const unsigned int setup_arr_sizes[] = {
	ARRAY_SIZE(setup_0),
	ARRAY_SIZE(setup_1),
	ARRAY_SIZE(setup_2),
	ARRAY_SIZE(setup_3),
	ARRAY_SIZE(setup_4)
};
```
Unfortunately the developer seems to be hard to reach.
Maybe you have time and would like to have a look to see if there is anything to see.

Kimplul, 2024-04-21:
> Ok, that's a pity, would have made it easier.

I can still test with the base T300, of course.

> But what exactly goes in there, no plan.

I added that bit. For whatever reason, the wheel got stuck in the bootloader without those packets, but I have no idea why or what they do. So far, they haven't been causing any issues for other wheels than the T300, so they've stayed.

Wanama78, 2024-04-21:
> I added that bit. For whatever reason, the wheel got stuck in the bootloader without those packets, but I have no idea why or what they do. So far, they haven't been causing any issues for other wheels than the T300, so they've stayed.

Yes, it seems that another entry is missing.
Now I just have to find out which one.

Kimplul, 2024-04-21:
They are `URB_INTERRUPT out` in Wireshark, for example packet 30 in `GT_init_wheel_win.pcapng`.