Kimplul/hid-tmff2 #100: No Input with T300RS in ACC

roomfullofcommits, 2024-06-05:
I'm on PopOS, trying to run my T300RS in Assetto Corsa Competizione.

I installed the dependencies and then the driver with DKMS, then booted into Windows to update my firmware to v34, I was on v29 before.
I tried it in ACC on pop and I'm pretty sure the game already recognized the wheel at that point and loaded the correct profile. However, none of my buttons and pedals worked and I got no input with the wheel. The wheel also always tried to center itself like it does by default, not sure if that is how it's supposed to behave in menus.
I then tried installing the windows drivers in the proton prefix and I think they installed correctly, but I'm not sure. Is there a way to check that?
After that I had no change in behavior in ACC. I did notice though that in PS4 mode at least my buttons worked, but not my wheel and pedals.
The blacklisting thing didn't work for me, it says `bash: /etc/modprobe.d/hid_thrustmaster.conf: Permission denied`.
During that whole process I rebooted multiple times and nothing changed.

`sudo dmesg -w` logs this:
<details>
  <summary>Log</summary>
[Full log in pastebin](https://pastebin.com/CYCKHLCU)

``` 
. . . . . . . . . . .
. . . . . . . . . . .
[   21.074133] usb 7-4: 4:1: All rates were zero
[   21.074143] usb 7-4: 4:2: All rates were zero
[   21.074149] usb 7-4: 4:3: All rates were zero
[   21.074153] usb 7-4: 4:4: All rates were zero
[   21.074158] usb 7-4: 4:5: All rates were zero
[   21.074163] usb 7-4: 4:6: All rates were zero
[   21.261204] usbcore: registered new interface driver snd-usb-audio
[   24.921287] logitech-hidpp-device 0003:046D:4074.000A: HID++ 4.2 device connected.
[   33.581260] rfkill: input handler enabled
[   36.849235] rfkill: input handler disabled
[   89.500664] xone-dongle 3-3:1.0: xone_mt76_suspend_radio: suspended
[  228.802646] usb 3-4: USB disconnect, device number 5
[  231.802958] usb 3-4: new full-speed USB device number 11 using xhci_hcd
[  231.974998] usb 3-4: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  231.975004] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  231.975007] usb 3-4: Product: Thrustmaster FFB Wheel
[  231.975009] usb 3-4: Manufacturer: Thrustmaster
[  232.010847] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:05:00.3/usb3/3-4/3-4:1.0/0003:044F:B65D.000B/input/input31
[  232.071258] hid-thrustmaster 0003:044F:B65D.000B: input,hidraw2: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:05:00.3-4/input0
[  232.089473] Error: Driver 'hid-thrustmaster' is already registered, aborting...
[  232.090056] hid-thrustmaster 0003:044F:B65D.000B: Wheel with model id 0x206 is a Thrustmaster T300RS
[  232.091162] usb 3-4: USB disconnect, device number 11
[  232.093066] hid-thrustmaster 0003:044F:B65D.000B: Success?! The wheel should have been initialized!
[  232.782965] usb 3-4: new full-speed USB device number 12 using xhci_hcd
[  232.950045] usb 3-4: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[  232.950055] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  232.950060] usb 3-4: Product: Thrustmaster T300RS Racing wheel
[  232.950064] usb 3-4: Manufacturer: Thrustmaster
[  232.963398] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:05:00.3/usb3/3-4/3-4:1.0/0003:044F:B66E.000C/input/input32
[  232.963668] tmff2 0003:044F:B66E.000C: input,hidraw2: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:05:00.3-4/input0
[  232.967017] tmff2 0003:044F:B66E.000C: force feedback for T300RS
[  526.816430] usb 3-4: USB disconnect, device number 12
[  529.316353] usb 3-4: new full-speed USB device number 13 using xhci_hcd
[  529.485770] usb 3-4: New USB device found, idVendor=044f, idProduct=b66d, bcdDevice= 1.00
[  529.485777] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  529.485780] usb 3-4: Product: Thrustmaster Racing Wheel FFB
[  529.485782] usb 3-4: Manufacturer: Thrustmaster
[  529.503051] input: Thrustmaster Thrustmaster Racing Wheel FFB as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:05:00.3/usb3/3-4/3-4:1.0/0003:044F:B66D.000D/input/input33
[  529.503364] tmff2 0003:044F:B66D.000D: input,hiddev4,hidraw2: USB HID v1.10 Gamepad [Thrustmaster Thrustmaster Racing Wheel FFB] on usb-0000:05:00.3-4/input0
[  529.506737] tmff2 0003:044F:B66D.000D: force feedback for T300RS
[  608.391662] usb 3-4: USB disconnect, device number 13
[  610.570562] usb 3-4: new full-speed USB device number 14 using xhci_hcd
[  610.742560] usb 3-4: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  610.742568] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  610.742571] usb 3-4: Product: Thrustmaster FFB Wheel
[  610.742574] usb 3-4: Manufacturer: Thrustmaster
[  610.758730] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:05:00.3/usb3/3-4/3-4:1.0/0003:044F:B65D.000E/input/input34
[  610.758920] hid-thrustmaster 0003:044F:B65D.000E: input,hidraw2: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:05:00.3-4/input0
[  610.780602] hid-thrustmaster 0003:044F:B65D.000E: Wheel with model id 0x206 is a Thrustmaster T300RS
[  610.781673] usb 3-4: USB disconnect, device number 14
[  610.783616] hid-thrustmaster 0003:044F:B65D.000E: Success?! The wheel should have been initialized!
[  611.135210] Error: Driver 'hid-thrustmaster' is already registered, aborting...
[  611.474654] usb 3-4: new full-speed USB device number 15 using xhci_hcd
[  611.657618] usb 3-4: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[  611.657625] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  611.657628] usb 3-4: Product: Thrustmaster T300RS Racing wheel
[  611.657630] usb 3-4: Manufacturer: Thrustmaster
[  611.671793] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:05:00.3/usb3/3-4/3-4:1.0/0003:044F:B66E.000F/input/input35
[  611.671985] tmff2 0003:044F:B66E.000F: input,hidraw2: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:05:00.3-4/input0
[  611.674560] tmff2 0003:044F:B66E.000F: force feedback for T300RS
[  801.782934] usb 3-4: USB disconnect, device number 15
```

</details>

And I feel the effects with `fftest`.
Let me know I there's more info I can provide, I think in the meantime I'll try using the wheel in other games and see how that goes.

Kimplul, 2024-06-06:
Hi, thanks for the detailed report. This sounds a lot like #56, unfortunately that one still hasn't really been resolved. Looks like the driver is correctly installed and responds like it should to `fftest`, so this is (probably) something in either Proton or ACC itself.

I would recommend trying out different Proton versions, would be cool if some newer version of Proton fixed things but I wouldn't hold my breath. Unfortunately I just went on holiday and can't really test anything on my end for a bit over a week, but I'd be happy to try and help otherwise.

  > The blacklisting thing didn't work for me, it says bash: /etc/modprobe.d/hid_thrustmaster.conf: Permission denied.

Whoops, wrong command in the README, good thing you pointed it out. Should be
```
echo 'blacklist hid_thrustmaster' | sudo tee /etc/modprobe.d/hid_thrustmaster.conf
```

Fixed it in the README.

Kimplul, 2024-06-06:
You could check that registry keys associated with Thrustmaster show up, like here: https://github.com/Kimplul/hid-tmff2/issues/46#issuecomment-1198933162

Though, that's just checking that the driver installed, not necessarily if it is having any effect. Here's a list of some games that seem to benefit from the driver being installed, you could maybe do an A/B test, before vs after installing the driver: https://github.com/Kimplul/hid-tmff2/issues/46#issuecomment-1202526300

roomfullofcommits, 2024-06-06:
I had absolutely no luck testing today. I tried Dirt 4 and my wheel didn't show up before or after installing the windows drivers. Or rather, it showed up as an Xbox Controller and none of the input worked.

I was unsure before if the driver installed correctly for ACC, but I was able to uninstall it with the installer, so that seemed to work.

In https://github.com/Kimplul/hid-tmff2/issues/46#issuecomment-1202526300 it sounds like both Dirt 4 and ACC should work as well as on Windows.

As for Proton versions, I didn't try a lot of different ones today because I was busy getting steam installed correctly and installing a up to date version of protontricks, that didn't have an effect on wheel functionality though. I tried Experimental, 9-0-1, 8-0-5, and Proton-GE 9-7. None of them made a difference for me for ACC or Dirt, maybe I'll try going further back tomorrow.

tbh I'm pretty discouraged right now and I feel like there's something more basic wrong apart from individual games.

roomfullofcommits, 2024-06-07:
No good news again, unfortunately. I read through #54 and tried some stuff, but none of it worked for me.

I followed yacinbm's experiment of running `protontricks -v -c "wine control"` with `--no-bwrap` and unlike them I didn't get input in wine control.

I tried adding a udev rule for the wheel, but it didn't help.

I tried running the protontricks wine control command without the Steam runtime like described [here](https://github.com/Matoking/protontricks/issues/216#issuecomment-1501092876), but the run script didn't generate for me, so I don't know if the Steam runtime is the problem (but it seems pretty likely at this point. like, what else is there?).

I tried older proton versions and went as far back as 4.11-13 because I read somewhere that someone had problems starting with 5-something, but nothing.

I was using the Steam .deb version from Steam's website and tried using the flatpak instead, but no change.

Kimplul, 2024-06-08:
What does Steam's game controller input menu say about the wheel? Try disabling Steam Input if it isn't already. If that doesn't work, it really does seem like some bug in the Steam runtime. May or may not be related to https://github.com/ValveSoftware/Proton/issues/6674.

roomfullofcommits, 2024-06-08:
# OH. MY GOD
I mistyped the file name of the udev rule. I had `[...]-rules` instead of `[...].rules`. I'll still have to test if it works in other games but I have input in Dirt 4 and ACC.

Can I send you a few bucks or gift you a game for your awesome work on the driver and your time?

Kimplul, 2024-06-09:
> I mistyped the file name of the udev rule. I had [...]-rules instead of [...].rules. I'll still have to test if it works in other games but I have input in Dirt 4 and ACC.

Hah, good catch. Seems to be a common enough issue that I might add it to the list of common issues. Might also be a good idea to set up some database of which games are expected to work and how well and workarounds, most of the opened issues tend to be about game support these days.

> Okay, I still have problems in ETS2. I can't use the native version, because the graphics are glitchy and I'm getting no input with the proton version, but the game sees the wheel. I have the Windows driver installed and tried disabling Steam Input.

Interesting, ETS2 has showed some other issues before but not missing input completely. I can't remember if I tested ETS2 via Proton or natively, but I believe it might've been natively. I should check that when I get back, it's also been a while since I last played it and I believe it's gotten some updates to how input is handled, could've introduced some new issues.

> Can I send you a few bucks or gift you a game for your awesome work on the driver and your time?

Assuming ETS2 hasn't soured your mood too badly, I would be happier if you used that money to make a donation to your charity/software project/whatever of choice :)

Kimplul, 2024-06-16:
Alright, ETS2 worked fine for me both natively and under Proton 9.0.2. Just a sanity check, you did run the input wizard after selecting the wheel as the input device?

Kimplul, 2025-02-01:
Hi @NickChiapputo, thanks for the report. Just to add to your impressive list of references, there's also https://github.com/Kimplul/hid-tmff2/issues/132, where we found out that the udev `uaccess` tag that seems to be required for some systems (particularly the SteamDeck) for input to be detected in Steam/Proton, seems to also cause some other systems to stop detecting input. Fun, eh?`

TL:DR; SDL, which Steam/Wine/Proton uses, has a feature where it prefers its own userspace HID drivers for some common devices. To use these drivers, the `uaccess` tag has to be set for the device. For some reason, at least some versions of SDL seem to incorrectly detect the T300 as a device that it should use a userspace driver for, but there isn't actually a driver that understands the wheel and we end up with broken input. I haven't pinned down which versions of SDL act this way, but at least `2.30.11` has worked for me on Debian Testing, with `2.30.7` being 'broken' on Gentoo. Ubuntu 24 Oracular seems to be on `2.30.6`, which is close to the known 'broken' Gentoo, so I guess it's possible you're being affected by the same root cause as well?

I somewhat recently added the `uaccess` tag to the default `udev` rules, so if that's what's causing your issues I might have to revert and provide some separate `sudo make fix-steamdeck` or something, at least until some known good SDL versions start getting widespread usage, presumably in a couple of years or so.

> Previously, most proton games I had to run with the PROTON_DUMP_DEBUG_COMMANDS=1 %command% command and then run with the generated ./tmp/proton_<name>/run executable. 

Interesting flow, seems pretty annoying. Seems like something that would be worth opening up an issue for :)

Wonder if Steam messes with library loading, for example it could come with a 'broken' version of SDL and you running the command manually would load another, 'working' SDL library? Not sure.

> I didn't want to create a new issue since the behavior I am seeing is (more or less, I don't think there's one that is exactly the same other than this issue) [...]

Sure, this is fine. I wouldn't have minded a new issue either, but there's value to adding more context to an existing discussion as well.  

> [...] f the other reports seem to have: [  241.640731] usbhid 5-1:1.0: URB to change wheel mode seems to have failed, error code -2

That's (unfortunately) kind of expected, when the wheel restarts itself it does so **very** aggressively and kind of breaks the USB protocol while doing so, which causes the USB subsystem to report an error (even though everything is 'fine'). This error, however, seems to vary quite a lot, you can see in https://github.com/Kimplul/hid-tminit/blob/eebcfce75900bb793f32ccd463367c46567eed8d/tminit.c#L152 that a few known error values are checked but every now and then a new one pops up. Might be worth just ignoring the errors outright, but that's not exactly good practice either (what if a real error happens?).

NickChiapputo, 2025-02-02:
> Just to add to your impressive list of references, there's also https://github.com/Kimplul/hid-tmff2/issues/132, where we found out that the udev uaccess tag that seems to be required for some systems

Yep, I was looking at #132 but didn't see any change with, without, or modifying the udev rules so I didn't include that with my report.

> TL:DR; SDL, which Steam/Wine/Proton uses, has a feature where it prefers its own userspace HID drivers for some common devices.

That seems to be it! Ubuntu 24 noble came with 2.30.0 so I removed that and built 2.30.11. That didn't work either, so I went back to 2.0.20 which I (probably) had before the distro upgrade and it works exactly as before! I started making a table to list what versions were working to hopefully help you pin down what versions are happier...but nothing past 2.24.2 works for me (at least in Proton 8.0-5 which is the only version I can get to work with the wheel). It's possible there is something else on my system that is not particularly up to date that interacts poorly with the newer versions. But here's the table anyway:

| SDL2 Version  | Works (for me) |
| ------------- | ------------- |
|2.0.20|✅|
|2.0.22|✅|
|2.24.0|✅|
|2.24.1|✅|
|2.24.2|✅|
|2.26.0|❌|
|2.26.1|❌|
|2.26.2|❌|
|2.26.3|❌|
|2.26.4|❌|
|2.26.5|❌|
|2.28.0|❌|
|2.28.1|❌|
|2.28.2|❌|
|2.28.3|❌|
|2.28.4|❌|
|2.28.5|❌|
|2.30.0|❌|
|2.30.1|❌|
|2.30.2|❌|
|2.30.3|❌|
|2.30.4|❌|
|2.30.5|❌|
|2.30.6|❌|
|2.30.7|❌|
|2.30.8|❌|
|2.30.9|❌|
|2.30.10|❌|
|2.30.11|❌|

Thank you for pointing me in the right direction!

> Interesting flow, seems pretty annoying. Seems like something that would be worth opening up an issue for :)

It is certainly not ideal and the option to generate that executable is gone in proton 9+, so I have to stay back on 8. I just move the executable to under my user directory and use that as a launcher instead. Fairly painless, fortunately. There is an open issue on this here: https://github.com/ValveSoftware/Proton/issues/6674, but I am not sure how actively anybody is searching for a solution/cause.

Kimplul, 2025-02-02:
Thanks for the list of SDL versions, greatly appreciated. I'm somewhat surprised 2.30.11 doesn't work, either there was some flaw in my testing setup or there's some third, as of yet unknown variable that's different between Ubuntu and Debian. Still, very useful information.

I suppose the next step would be to start digging into why the wheel is being treated as if it should be accessed from userspace. I might also go ahead and revert my `udev` rule changes as mentioned previously, since based on your reports they might be actively harmful for a large(ish) number of users, at least until I we figure out what the deal with SDL is. Could maybe also open up an issue in SDL for it, I'll have a look tomorrow after work.

Kimplul, 2025-02-03:
I opened up a PR in SDL with what I believe is a fix, seemed to work on my machine at least, let's see what the maintainers think of it. I also separated out the `uaccess` rules to be specific to the SteamDeck, since I believe that's the main distro that requires it. Still not entirely sure why, figuring that out is still a TODO.

Tebro, 2025-02-28:
I had this issue with ACC, after many hours of trying  various things I found this project, and it fixed my issues instantly https://gitlab.com/openirseny/protopedal, just had to bind everything in ACC by hand.