Kimplul/hid-tmff2 #101: T300 RS no constant forces in games

anto31ad, 2024-06-07:
Hi,

I've been trying to get my T300 RS (GT edition) to work on Linux Mint 21.3.

I have installed the driver twice so far, first time manually, second time through the DKMS script.

I have tried the wheel in the following games
- *Speed Dreams*: no ffb
- *Race07*: ffb only on impacts and curb-riding and/or when the Spring effect is on (activated/deactivated through oversteer)
- *Automobilista*: no ffb

I tried several things, among which I
- blacklisted `hid-thrustmaster` in `blacklist.conf`
- installed the windows driver in the Race07 and Automobilista Proton prefix (before realizing that this is pointless)

Then I used fftest and everything seems to work fine since running effects 0 (periodic) and 1 (constant) does indeed make the wheel do what you would expect it to do; But it seems like the games I tried won't use such effects, and I am lead to believe that no other game will behave differently.

I browsed through closed issues and other people seemed to have solved their issue after your fixes, but I guess it's not my lucky day lol.

One thing I noticed is that when I run dmesg I still see hid-thrustmaster popping up (even though it should be blacklisted):

```
-5: new full-speed USB device number 7 using xhci_hcd
[   50.455144] usb 3-5: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[   50.455149] usb 3-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   50.455150] usb 3-5: Product: Thrustmaster FFB Wheel
[   50.455152] usb 3-5: Manufacturer: Thrustmaster
[   50.466347] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/0003:044F:B65D.0004/input/input15
[   50.466494] hid-thrustmaster 0003:044F:B65D.0004: input,hidraw3: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-5/input0
[   50.484459] hid-thrustmaster 0003:044F:B65D.0004: Wheel with (model, attachment) = (0x2, 0x6) is a Thrustmaster T300RS. attachment_found=1
[   50.484727] hid-thrustmaster 0003:044F:B65D.0004: Success?! The wheel should have been initialized!
[   50.484745] usb 3-5: USB disconnect, device number 7
[   51.154891] usb 3-5: new full-speed USB device number 8 using xhci_hcd
[   51.303461] usb 3-5: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[   51.303465] usb 3-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   51.303466] usb 3-5: Product: Thrustmaster T300RS Racing wheel
[   51.303467] usb 3-5: Manufacturer: Thrustmaster
[   51.305128] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/0003:044F:B66E.0005/input/input16
[   51.305351] hid-generic 0003:044F:B66E.0005: input,hidraw3: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-5/input0
[   51.394402] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/0003:044F:B66E.0005/input/input17
[   51.454188] tmff2 0003:044F:B66E.0005: input,hidraw3: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-5/input0
[   51.454594] tmff2 0003:044F:B66E.0005: force feedback for T300RS
```
And running `lsmod` with the wheel plugged in looks like this

```
Module                  Size  Used by
hid_tmff_new           53248  0
...
hid_generic            16384  0
usbhid                 65536  2 hid_tminit
hid                   151552  4 hid_tminit,usbhid,hid_generic,hid_tmff_new
...
```

What should I try next?

Thank you for your efforts.

Kimplul, 2024-06-08:
Hi, thanks for the report. I believe this is the first time anyone's mentioned *Speed Dreams* or *Race07*, so I don't really know how well they should behave with this driver. *Automobilista* should still work from my experience, so something funky is going on. Looks like you've installed the module correctly, as `fftest` seems to work as expected.
What does Steam's controller input menu say about the wheel?

> One thing I noticed is that when I run dmesg I still see hid-thrustmaster popping up (even though it should be blacklisted):

That's normal, though rather confusing. `hid-tminit` that is currently installed by this driver as a dependency reports its name as 'hid-thrustmaster' in the logs (https://github.com/scarburato/hid-tminit/blob/9375f6c7d83af5dd6c8b8fe30351d0f36043b20a/hid-tminit.c#L392), but the module files it is installed as are named 'hid-tminit'. The blacklist works on the file names, not the reported names. So your blacklist is working, the naming situation is just a bit confusing right now. Should be fixed in the near future.

anto31ad, 2024-06-08:
I will check the Steam controller input menu soon

In the mean time, I wanted to point out a few things:

As I said the first time I installed the driver  (a couple of months ago I believe) I went for the manual approach. I tested the wheel through Oversteer, running the linearity tests and such;
some days after that day I was in a call with a friend and wanted to show him some footage of a sim, so I downloaded Speed Dreams (I played it briefly on Windows in the past) but I couldn't set it up, so without further ado I downloaded Race07 since it's lightweight and run it for 5m;  the force feedback seemed to be working since I could feel resistance when turning, and pulling from curbs and impacts. 

Then some time passed up to some days ago when I wanted to go for a spin.
I fired up Race07 and realised that other than curbs and impacts I could just feel the spring effect which I had left enabled;

At this point I though that maybe I could try a fresh install of the driver, but didn't know how to uninstall (i tried make uninstall inside the project directory) so I thought that maybe doing a reinstall would overwrite every module file.

But the wheel "felt" exactly the same.

That's when I tried various things like those I mentioned in the previous comment.
So initially I hadn't even blacklisted the hid-thrustmaster driver (about this at the bottom)

I think it's of note mentioning the following:

Speed dreams had a huge deadzone in the center, so I followed your tip on setting up a udev rule in `/etc/udev/rules.d/99-joydev.rules`:
```
SUBSYSTEM=="input", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b66e", RUN+="/usr/bin/evdev-joystick --evdev %E{DEVNAME} --deadzone 0"
```
This didn't seem to produce any effect until yesterday when I went for another try and the deadzone was gone.

Now, Oversteer has always required sudo so far to access the wheel properties (and so did fftest) but apparently this is not meant to happen.

The Oversteer's readme says this:

> By default, the udev rules will be installed at `/usr/local/lib/udev/rules.d` when installing to prefix `/usr/local or /lib/udev/rules.d` when installing to any other prefix.  (...) Older rules might be already installed at `/etc/udev/rules.d` or `/lib/udev/rules.d`. You may need to remove these files manually in case you're experiencing issues with permissions.

Since `/etc/udev/rules.d/99-joydev.rules` apparently worked, could it be that my Mint system looks for rules in /etc/udev/rules.d/ instead of the other places?
And is it possible that also games don't have the permission to use the force feedback?
Though, Race07 partially working kind of invalidate this hypothesis.

### About blacklisting hid-thrustmaster

In your ReadMe you wrote

> Until the updated hid-tminit is [upstreamed](https://github.com/scarburato/hid-tminit), you might want to blacklist the kernel module hid-thrustmaster. Do this with
> `echo 'blacklist hid_thrustmaster' | sudo tee /etc/modprobe.d/hid_thrustmaster.conf`

but the driver is called hid-thrustmaster, with the dash, not the underscore (found in `/lib/modules/$(uname -r)/kernel/drivers/hid/`)  Is this a mistake, or is it irrelevant?

anto31ad, 2024-06-08:
Is this the steam's controller input menu?

[image]

If so, it does not seem to detect the wheel.

Kimplul, 2024-06-08:
> At this point I though that maybe I could try a fresh install of the driver, but didn't know how to uninstall (i tried make uninstall inside the project directory) so I thought that maybe doing a reinstall would overwrite every module file.

Yep, that is effectively what happens. Generally you would also want to restart your system to make sure the changes take effect. There are ways to avoid rebooting as well, but generally rebooting is just the easiest/safest way.

> Speed dreams had a huge deadzone in the center, so I followed your tip on setting up a udev rule in /etc/udev/rules.d/99-joydev.rules: [...]

Same thing, generally a reboot is in order. I guess that could be made more clear in the README.

> Now, Oversteer has always required sudo so far to access the wheel properties (and so did fftest) but apparently this is not meant to happen. [...]

Oversteer generally installs the correct udev files automatically so you don't have to run it as root after that, there should be a popup that asks if you want to install the files the first time you launch Oversteer. At least when installed from GitHub the last time I installed it. `fftest` shouldn't require root access, at least it doesn't on my Debian machines and I don't think I've done anything special to allow that to happen.

Could be that something weird is going on with your permissions, though I don't think that that really explains much, as programs generally wouldn't use some effects but not others, they'd use no effects at all, as you correctly deduced.

> [...] driver is called hid-thrustmaster, with the dash, not the underscore (found in /lib/modules/$(uname -r)/kernel/drivers/hid/) Is this a mistake, or is it irrelevant?

For whatever reason the kernel module subsystem uses `-` in some contexts and `_` in others, the blacklist just happens to be one where underscores seem to generally be used. Not sure why but probably some kind of backwards compatibility thing.

> Is this the steam's controller input menu? [...]

Yep, and not detecting anything is in this case is what I want to see, Steam doesn't think the wheel is an Xbox controller or something. Could you still try running your games with Steam Input disabled? You can disable Steam Input on a per-game basis, `Properties > Controller`.

anto31ad, 2024-06-08:
I tried disabling steam input per game but nothing changed.

However, in Automobilista control setting I noticed that by disabling "auto steering rotation" I cannot change the rotation and it's stuck at 240.

In fact, forcefeedback aside the steering angle felt weird compared to last time I played the game on Windows, as if there was a mismatch between the T300 rotation and the ingame rotation,

Here's a screen

[image]

Kimplul, 2024-06-08:
> In fact, forcefeedback aside the steering angle felt weird compared to last time I played the game on Windows, as if there was a mismatch between the T300 rotation and the ingame rotation, [...]

Seems plausible, I suppose? The method Linux uses to change the steering angle is by writing the angle to a special file, and I don't think Proton is aware of how to handle that. So the wheel receives inputs from Linux, that is working with some angle, and the wheel reinterprets that to be some other angle, maybe.

I'm not sure if the steering wheel angle setting is something standardized into DirectInput or if it might be something provided by `libtm`, a library that some games seem to use to handle Thrustmaster wheels. Unfortunately that library is a black box, and I have no knowledge of how it works internally.

Anyway, thanks for checking. Do you have a way to update the firmware of the wheel? The driver should complain if it detects a firmware version too old to work for certain, but it could be that something's messed up the check for whatever reason.

I'm currently away from home and I can't check how Automobilista is behaving on my computer for at least a week, sorry.

anto31ad, 2024-06-08:
Don't worry! Take your time :)

Yesterday I checked the firmware and even reinstalled it for mistake but it was already at version 34, which seems to be the latest.

In the past hour
- I tried to execute Speed Dreams with elevated priviledges but still no force feedback to be felt.
- I tried solving the permission problem with Oversteer by changing the location of the rules and editing the .rules file of thrustmaster in ways that were described in some closed issues in Oversteer.

I was thinking about installing another distro (maybe Debian or Fedora) to see if the issue persists, but that would be quite time consuming.

Kimplul, 2024-06-08:
>  was thinking about installing another distro (maybe Debian or Fedora) to see if the issue persists, but that would be quite time consuming.

Maybe try using a virtual machine, I fairly regularly spin up throwaway VMs for testing different things. I would recommend `virt-manager`, fairly nice GUI tool, it generally creates VMs with reasonable defaults. You (probably) won't be able to run any heavier games, but at least Race07 looks to maybe be lightweight enough to test.

I didn't even realize Speed Dreams was open source, actually really cool. I'll have to look into how it implements its FFB, could maybe shed some light onto other games as well.

anto31ad, 2024-06-10:
I will try VMs once I have some time to spare, thanks for the tips :)

About Speed Dreams, yeah, very cool to see! Seems to be a very old project but still in development, I read they're changing up the heads of the projects and planning, among various things, to finally move from Subversion to Git, for their version control ([news post here](https://www.speed-dreams.net/en/major-changes-in-the-speed-dreams-project/)).

Then I wanted to ask you some questions, @Kimplul , but they're slightly out of topic in regards to this issue, so I thought it's best to leave out of this space. It's about getting started with linux driver development. Where can I reach you?

Kimplul, 2024-06-10:
> Where can I reach you?

Email at kimi.h.kuparinen@gmail.com if you want a private but asynchronous conversation, maybe a GitHub discussion if you don't mind it being public. For more direct contact, I do still have a Discord account, `kimplu`, but I've been meaning to try out some more open alternatives likes Matrix or something, wouldn't mind having an excuse to finally make an account.

anto31ad, 2024-06-10:
> maybe a GitHub discussion if you don't mind it being public

Actually I don't mind it being public, because it can be of use for others.

Since I was not familiar enough with GitHub discussions i did a quick search that showed me that there should be a discussion section in the repo. I guess it isn't enabled here because there was no reason up to now. Or it's not available for everyone.

> but I've been meaning to try out some more open alternatives likes Matrix or something, wouldn't mind having an excuse to finally make an account

I don't mind this too, seems like  an interesting alternative.

Kimplul, 2024-06-10:
> Since I was not familiar enough with GitHub discussions i did a quick search that showed me that there should be a discussion section in the repo. I guess it isn't enabled here because there was no reason up to now. Or it's not available for everyone.

Oh, sorry, haven't used them myself, I just saw the button for them and thought they might be interesting to try out, didn't realize they had to explicitly be enabled. Should be online now.

anto31ad, 2024-06-10:
They are indeed! I'll write a post as soon as I can,

Kimplul, 2024-06-16:
I looked at speed dreams, turns out that the game requested an effect with a duration of 1 second, started playing it with a replay count of effectively infinite and then just kept updating it. I thought I handled these situations properly, but the short effect duration triggered a bug that caused the driver to ignore the play count and stop the effect after one second. Speed dreams now has at least some FFB on my machine with commit cafd0f46535df3674098754c6c693cd34274810c.

anto31ad, 2024-07-03:
> Speed dreams now has at least some FFB on my machine with commit...

I tested the wheel some days ago but forgot to let you know that the wheel came alive with Speed Dreams! Didn't test extensively due to limited time, but plan to in the upcoming weeks.

Unfortunately the issues with Race07 and AMS with my system are still there;

Moreover, not sure if I made a mistake, but the constant force effect (no. 1) in fftest didn't work this time. I need to try again before confirming, maybe it was just a temporary hiccup.