Kimplul/hid-tmff2 #106: BeamNG sees wheel but does not detect input

krims0n32, 2024-06-17:
I cannot get my T300RS to work with BeamNG. The game sees the wheel but does not detect any input at all. It does detect input on my Virpil controls however. If I set the wheel to PS4 mode input does work but some buttons and the shifter pedals do not work. Any ideas?

Kimplul, 2024-06-17:
> I see there are similar issues reported already regarding proton, sorry for not checking those first.

No worries. Sounds like it's OK to close this issue, but feel free to chime in on the other ones.

> I am supposed to set the wheel to PC mode correct?

PC mode is generally recommended, but there is support for the PS4 mode as well, just less tested.

closed 2024-06-17 by Kimplul

krims0n32, 2024-06-18:
I tried PS4/PS5 mode again and now it works all of a sudden. Tried ACC and BeamNG. Force feedback also works.

Thanks for your work on this driver!