Kimplul/hid-tmff2 #108: Wrong wheel input in My Summer Car (T300RS)

swagboss2011, 2024-06-27:
OS: EndeavourOS
Did everything in the readme, wheel works as expected in the games I've tested so far:
- pCARS2 running with Bottles
- BeamNG running with Proton GE
- ETS2 running with Proton GE
But on MSC (tried both bottles and proton, with driver installed in the prefix) only the right axis of the wheel gives input and the pedals only work on half of the axis, buttons work as expected.

https://github.com/Kimplul/hid-tmff2/assets/170236378/6db1527e-e681-4f23-b1d5-358eebe08243

Kimplul, 2024-06-27:
I'm guessing you already tried different Proton versions?

Somewhat interesting that all tested methods produce the same issue, might indicate a bug in the game itself.

swagboss2011, 2024-06-28:
I've tried different versions, but on protondb people used proton 6.3-8 to get their steering wheels to work, I couldn't get the game to launch with that version. Guess I'll just play this one on windows.

hellfur, 2024-07-07:
I've been having the same issue with Farming Simulator 2022. Exactly as described the wheel works when turning right, but not left (turning left makes the wheel lock hard right in Farming Sim 2022). Pedals only have half the axis.

Not sure if it's somehow engine-based, but thought I should +1 this issue with a different game.

hellfur, 2024-07-08:
I can also confirm that Proton 6.3 works nearly correctly for Farming Simulator 22. It doesn't detect my clutch pedal, but it does allow the wheel to turn both directions properly, and the other pedals respond properly as well. So that narrows it down to some interaction with newer Proton versions for these two games.

If there's anything you'd like me to run to try to get more detail, let me know.

jsee0, 2024-07-15:
I've been trying to get my summer car to work with force feedback for quite a while now. The issue is that my summer car(and maybe Farming Sim too but I don't play it so I can't be sure) expects the range of of the all axis to go from -topValue to topValue with the default that it is being provided for the steering axis ranging from 0 to 65535. I believe that is why only the right half of the steering range works. Although you would expect a different behaviour, as it shouldn't just stop at the left half if this is the case. I'm not sure if proton or the driver does something with this range but it seems like it. To fix this I used [these](https://www.reddit.com/r/MySummerCar/comments/16fyhpr/comment/k058v2z/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button) commands, with the only difference being that the max value for pedals on the t300 seems to be 1023 and not 255.
This worked fine except for the fact I had no force feedback, but it was something.

Now the good news: I managed to get everything working with force feedback on proton version 4.11-13. I remember the force feedback not working before with this version but I just launched the game to check and I have no clue as to why, because I don't remember making any changes, but it works now.

I haven't played much in this version other than just now to check if force feedback works, so there may be more issues but for now I'm gonna enjoy msc without having to boot up windows.

timechips, 2024-08-08:
> OS: EndeavourOS
> Did everything in the readme, wheel works as expected in the games I've tested so far:
> - pCARS2 running with Bottles
> - BeamNG running with Proton GE
> - ETS2 running with Proton GE
> But [...]

Hello.
I'm sorry to ask but I have been trying to make the wheel t248 work in bottles for a couple hours now and I can't figure out how. I'm kindly asking just for some simple directions. How did you manage to make it work?

swagboss2011, 2024-08-08:
> Hello. I'm sorry to ask but I have been trying to make the wheel t248 work in bottles for a couple hours now and I can't figure out how. I'm kindly asking just for some simple directions. How did you manage to make it work?

Worked straight out of the box with no tweaking required, games just detected the wheel no problem like on windows.