Kimplul/hid-tmff2 #111: How to redo the installation?

janot92, 2024-07-19:
Hey.
Firstly thank you for the work you did here, doing Thrustmasters work yourself. Ive recently switched completely to Linux and was pleasantly suprised that even my T300RS could be working.
However Im strugling a bit.
The wheel is working, however the FFB seems to be "dull" and I cant change anything using Oversteer. I think I might have screwed up when building the kernel module, I tried the manual method since the DKMS didnt work. Later I learned that distribution that I use, Fedora, needs to install DKMS manually as well.
So now I built the module twice using both methods and Im not sure if that could be causing any issues. However I have no idea how to clear all of the previous stuff I did.
Wheels is correctly recognized and I can bind all the buttons and pedals in Assetto Corsa just fine. Its only FFB thats "dead", and reading here makes me think that shouldnt be the case for T300.
Running the fftest command returns this:
`Device /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick opened
Features:
  * Absolute axes: X, Y, RZ, Throttle, Hat 0 X, Hat 0 Y, 
    [63 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: 
    Force feedback periodic effects: 
    [00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
  * Number of simultaneous effects: 0

Uploading effect #0 (Periodic sinusoidal) ... Error:: Function not implemented
Uploading effect #1 (Constant) ... Error: Function not implemented
Uploading effect #2 (Spring) ... Error: Function not implemented
Uploading effect #3 (Damper) ... Error: Function not implemented
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Function not implemented
Uploading effect #5 (Weak rumble, with light motor) ... Error: Function not implemented`

Thanks in advance!

Kimplul, 2024-07-20:
Hello, it seems like the driver isn't being loaded or is running into some issue that stops it from working, an outdated firmware version comes to mind.

Could you post the output of `sudo dmesg -w` when you plug in the wheel? I probably only need the last 10 or lines or so, should have some messages with `tmff` in them.

janot92, 2024-07-20:
Hmm, Ill be honest I havent used my wheel in a long time, but I think last time I updated the firmware was last year. Here the output:
`[ 1332.004585] usb 1-6: new full-speed USB device number 5 using xhci_hcd
[ 1332.308135] usb 1-6: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 1332.308140] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1332.308142] usb 1-6: Product: Thrustmaster FFB Wheel
[ 1332.308144] usb 1-6: Manufacturer: Thrustmaster
[ 1332.358171] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-6/1-6:1.0/0003:044F:B65D.0006/input/input31
[ 1332.358298] hid-thrustmaster 0003:044F:B65D.0006: input,hidraw5: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:03:00.0-6/input0
[ 1332.378562] Loading of module with unavailable key is rejected
[ 1332.380834] hid-thrustmaster 0003:044F:B65D.0006: Wheel with model id 0x206 is a Thrustmaster T300RS
[ 1332.382012] usb 1-6: USB disconnect, device number 5
[ 1332.383109] hid-thrustmaster 0003:044F:B65D.0006: Success?! The wheel should have been initialized!
[ 1333.069581] usb 1-6: new full-speed USB device number 6 using xhci_hcd
[ 1333.437271] usb 1-6: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[ 1333.437276] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1333.437278] usb 1-6: Product: Thrustmaster T300RS Racing wheel
[ 1333.437280] usb 1-6: Manufacturer: Thrustmaster
[ 1333.455110] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-6/1-6:1.0/0003:044F:B66E.0007/input/input32
[ 1333.455337] hid-generic 0003:044F:B66E.0007: input,hidraw5: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:03:00.0-6/input0
[ 1333.469671] Loading of module with unavailable key is rejected
`

Kimplul, 2024-07-20:
Looks like the module isn't being loaded, presumably due to `Loading of module with unavailable key is rejected`. This is related to module signing, something I thought DKMS handles automagically for the user. I guess I'll need to review how DKMS works, I'll get back to you on that.

There are some quicker ways around the issue, including signing by hand, but it's a bit cumbersome: https://www.redhat.com/sysadmin/secure-boot-systemtap

Another option is to disable SecureBoot, this generally allows unsigned modules to be loaded, but *maybe* makes your system less secure.

janot92, 2024-07-20:
So this will happen regardless if I install the module manually or with DKMS?
Signing by hand, yeah Ill have to pass, my head hurts whenever I see certs :D
Ill try to disable SecureBoot and will report back

Kimplul, 2024-07-20:
> So this will happen regardless if I install the module manually or with DKMS?

Oh, I might've misunderstood your initial message. I thought you had already installed via DKMS and that the driver wasn't working, but I think you meant that the install script itself didn't work?

DKMS *should* handle everything automatically and would at least theoretically be preferable to a manual installation.
The manual installation is also a bit clumsy because you'll have to reinstall the driver anytime your kernel gets updated, DKMS takes care of that as well.

> But I guess that might be since Im using the flatpak version...

Sounds plausible. Oversteer should prompt you the first time you run it to install some udev files that give it permission to access the wheel, sounds like something flatpak apps tend to have issues with due to the sandboxing that they try to implement. Try installing from source and see what happens.

janot92, 2024-07-20:
No no, sorry English is not my first language so I might not have written that correctly.
I did install the driver via DKMS but first time I tried the manual install. In both cases the only error was the SSL one that you explained in the readme.
And yes I've built Oversteer straight from GitHub and it works correctly now. Altho for some reason the deadline for the pedals is like 50% in the app, however in Assetto Corsa it's fine.
Thanks again, I'll keep an eye on when you update regarding the certs.

-------- Pôvodná správa --------
20. 7. 2024 12:49, Kimplul  napísal/a:

Kimplul, 2024-07-22:
Right, just had a look at the DKMS README and they have a section specifically on SecureBoot: https://github.com/dell/dkms?tab=readme-ov-file#secure-boot

TL;DR: DKMS does automatically sign modules, but the user apparently has to tell SecureBoot that DKMS's signature can be trusted.

I guess I must've glossed over that part last time I looked at DKMS, should probably add a note about it somewhere.

Kimplul, 2024-11-03:
Doing some winter cleaning, I believe this can be closed since we got the FFB working and I learned something new about DKMS. :)

closed 2024-11-03 by Kimplul