Kimplul/hid-tmff2 #112: T300RS GT not recognised outside Oversteer?

8L4YD, 2024-07-30:
Hi there,

I picked up this Thrustmaster T300RS GT wheel earlier today, I've been trying to get it set up by installing this driver and Oversteer. I planned to use the DKMS method to install but copied the method for the manual installation by accident, not knowing what to do I did the DKMS method after this. I'm not sure if this is the issue.

My system is running Fedora 40, kernel 6.9.11 (latest update).

Below is the last section of my output for ```sudo dmesg -w```;

```
[   33.217767] wlp5s0: authenticate with d4:35:1d:a9:33:2f (local address=5a:45:7c:99:db:e5)
[   33.219012] wlp5s0: send auth to d4:35:1d:a9:33:2f (try 1/3)
[   33.251420] wlp5s0: authenticated
[   33.253059] wlp5s0: associate with d4:35:1d:a9:33:2f (try 1/3)
[   33.257813] wlp5s0: RX AssocResp from d4:35:1d:a9:33:2f (capab=0x1411 status=0 aid=5)
[   33.272657] wlp5s0: associated
[  101.416048] hid_tmff_new: loading out-of-tree module taints kernel.
[  101.416053] hid_tmff_new: module verification failed: signature and/or required key missing - tainting kernel
[  151.582236] usb 1-4.3: new full-speed USB device number 8 using xhci_hcd
[  151.806934] usb 1-4.3: New USB device found, idVendor=044f, idProduct=b66d, bcdDevice= 1.00
[  151.806941] usb 1-4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  151.806944] usb 1-4.3: Product: Thrustmaster Racing Wheel FFB
[  151.806946] usb 1-4.3: Manufacturer: Thrustmaster
[  151.841232] input: Thrustmaster Thrustmaster Racing Wheel FFB as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-4/1-4.3/1-4.3:1.0/0003:044F:B66D.000B/input/input28
[  151.841441] tmff2 0003:044F:B66D.000B: input,hiddev98,hidraw7: USB HID v1.10 Gamepad [Thrustmaster Thrustmaster Racing Wheel FFB] on usb-0000:02:00.0-4.3/input0
[  151.852917] tmff2 0003:044F:B66D.000B: force feedback for T300RS
[  284.892773] usb 1-4.3: USB disconnect, device number 8
[  288.890365] usb 1-4.3: new full-speed USB device number 9 using xhci_hcd
[  289.149362] usb 1-4.3: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  289.149369] usb 1-4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  289.149372] usb 1-4.3: Product: Thrustmaster FFB Wheel
[  289.149374] usb 1-4.3: Manufacturer: Thrustmaster
[  289.227210] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-4/1-4.3/1-4.3:1.0/0003:044F:B65D.000C/input/input29
[  289.227338] hid-thrustmaster 0003:044F:B65D.000C: input,hidraw7: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:02:00.0-4.3/input0
[  289.254312] hid-thrustmaster 0003:044F:B65D.000C: Wheel with (model, attachment) = (0x2, 0x6) is a Thrustmaster T300RS. attachment_found=1
[  289.256579] hid-thrustmaster 0003:044F:B65D.000C: Success?! The wheel should have been initialized!
[  289.500319] usb 1-4.3: USB disconnect, device number 9
[  289.901343] usb 1-4.3: new full-speed USB device number 10 using xhci_hcd
[  290.222366] usb 1-4.3: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[  290.222375] usb 1-4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  290.222378] usb 1-4.3: Product: Thrustmaster T300RS Racing wheel
[  290.222381] usb 1-4.3: Manufacturer: Thrustmaster
[  290.257748] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-4/1-4.3/1-4.3:1.0/0003:044F:B66E.000D/input/input30
[  290.257989] tmff2 0003:044F:B66E.000D: input,hidraw7: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:02:00.0-4.3/input0
[  290.269330] tmff2 0003:044F:B66E.000D: force feedback for T300RS
[  338.779611] wlp5s0: disconnect from AP d4:35:1d:a9:33:2f for new auth to d6:35:1d:a9:33:37
[  338.838738] wlp5s0: authenticate with d6:35:1d:a9:33:37 (local address=5a:45:7c:99:db:e5)
[  338.839980] wlp5s0: send auth to d6:35:1d:a9:33:37 (try 1/3)
[  338.869069] wlp5s0: authenticated
[  338.869566] wlp5s0: associate with d6:35:1d:a9:33:37 (try 1/3)
[  338.871336] wlp5s0: RX ReassocResp from d6:35:1d:a9:33:37 (capab=0x1011 status=0 aid=1)
[  338.882844] wlp5s0: associated
[  338.927005] wlp5s0: Limiting TX power to 23 (23 - 0) dBm as advertised by d6:35:1d:a9:33:37
[  399.927779] show_signal_msg: 53 callbacks suppressed
[  399.927782] supertuxkart[4486]: segfault at 7efec64ac000 ip 00007efec64ac000 sp 00007fff815451e8 error 14 in libcap.so.2.70[7efec6832000+3000] likely on CPU 8 (core 0, socket 0)
[  399.927794] Code: Unable to access opcode bytes at 0x7efec64abfd6.
[ 1051.128530] gldriverquery[5129]: segfault at 0 ip 0000000000000000 sp 00007fff9f11c8d8 error 14 in gldriverquery[5636b8600000+14000] likely on CPU 1 (core 1, socket 0)
[ 1051.128540] Code: Unable to access opcode bytes at 0xffffffffffffffd6.
[ 1051.850340] gldriverquery[5306]: segfault at 8763e08 ip 0000000008763e08 sp 00000000ffacd45c error 15 likely on CPU 7 (core 7, socket 0)
[ 1051.850347] Code: 84 08 03 00 00 00 00 00 00 00 40 00 00 00 20 4a 84 08 04 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 <00> 00 00 00 81 00 00 00 08 51 73 08 d8 c7 d2 f7 00 00 00 00 00 00
```

The output of ```lsmod``` shows both ```hid_tminit``` and ```hid_tmff_new``` loaded but not being used by anything.

The tests in Oversteer seem to work fine, I've seen ```fftest``` mentioned but I'm not sure what this is or how to access it.

I'm not sure if I've done something wrong or what to do from here. The wheel doesn't show up as a controller in PC mode in Steam, and in PS mode (this seemingly has PS4 and PS5), the buttons don't work in Oversteer and the pedals are mixed around. I've tested BeamNG and F1 2018, with Steam Input both enabled and disabled, and neither seem to receive any input from the wheel, nor does Steam. I've tried the ```protontricks``` method of running the driver installer in both of these games, choosing restart at the end. I'm not sure whether or not I missed a step here.

Please feel free to ask any more questions, I'm still learning the ins and outs of Linux and I'm aware that I could have left some crucial information out.

Thanks in advance!!

Kimplul, 2024-07-30:
> [...] I'm not sure if this is the issue.

I doubt it, Linux doesn't really care if there are multiple identical drivers installed, at most printing a warning.

> [...] I've seen fftest mentioned but I'm not sure what this is or how to access it.

`fftest` is a command line tool that outputs some very basic effects to the wheel, generally a good first step in debugging. It looks like you can install it on Fedora through the package `linuxconsoletools`. Give it a try and see what happens, the following should work for your wheel:
```
fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick
```

> The wheel doesn't show up as a controller in PC mode in Steam, [...]

Do you mean that the wheel doesn't show up in the Steam controller menu? That would be normal, Steam apparently doesn't count wheels as controllers. Generally I would recommend sticking to the PC mode, some games might be confused by the PS mode showing up on PC. To the games, the different modes actually show up as different wheels, which is reflected in the `Generic FFB ...` name as well.

Just a sanity check, have you checked that the wheel attachment is installed properly on the base? The symptoms you're describing sound somewhat similar to #89, that one was fixed by reseating the wheel.

Have you tried games not on Steam? There have been some issues with Steam itself borking the input for whatever, reason, there's some previous discussion on it in #54 and #61. Speed Dreams is a good candidate: https://www.speed-dreams.net/

I see you've tried running SuperTuxKart, but apparently it crashed. Most likely unrelated to the wheel, since `gldriverquery` also seems to be crashing, but seems like you might have other issues with your installation than just the wheel not working :) There's some discussion on it over in https://github.com/ValveSoftware/steam-for-linux/issues/8853, with a claim that installing `SDL` fixes it: https://github.com/negativo17/steam/issues/9

> Please feel free to ask any more questions, I'm still learning the ins and outs of Linux and I'm aware that I could have left some crucial information out.

You were plenty informative, really very well put together issue.

8L4YD, 2024-08-05:
Hi Kimplui, I much appreciate your quick and detailed reply. I have been away the last couple days but have just done some testing. 

I was able to run ```fftest``` with no problems, and Speed Dreams did work with the wheel after setting bindings (there was a large deadzone however, not sure if that's expected behaviour). The wheel is correctly seated. 

I had a read through #54 , made sure I had wine installed and ran ```wine control```. The T300RS showed up initially under an xinput device, I then hit override and it showed in the main box next to the 'T500 Gear Shift' (my TH8A which has worked flawlessly). However, I can't see it detecting any input. When I go into the ```dinput``` menu, I get this in the terminal:

```
01fc:fixme:dinput:hid_joystick_device_try_open device usage ffc0:0001 not implemented!
01fc:fixme:dinput:hid_joystick_device_try_open device usage 0001:0000 not implemented!
01fc:fixme:dinput:hid_joystick_device_try_open device usage 008c:0001 not implemented!
```

This leads me to think that it's a problem with Wine. I believe I have 9.5 staging installed, but Protontricks is also throwing a bunch of architecture errors atm, it's kind of become a mess recently. I'm not sure what to do at this point.

And I believe that error could've just been from me Alt+F4ing STK lol.

Kimplul, 2024-08-05:
> [...] (there was a large deadzone however, not sure if that's expected behaviour). [...]

Did you already add a udev rule about deadzones? https://github.com/Kimplul/hid-tmff2?tab=readme-ov-file#common-issues-and-notes

> [...] (my TH8A which has worked flawlessly) [...]

Have you tried out what happens without the TH8A plugged in? Some applications don't handle multiple input devices particularly well, unfortunately.

> This leads me to think that it's a problem with Wine.

Possible, but I'm not quite convinced yet. Wine 9.5 is newer than what I have installed, I might have a go at it and see if I can replicate this. I'll report back.

8L4YD, 2024-08-07:
Hi again,

I tried adding the udev rule but it didn't seem to help with the deadzone. I did add it for PS4 normal mode, I'm not sure if that was the right move. I can only get the AppImage to work, the flatpak in the repo crashes when loading a race.

I also tried it without the shifter and it exhibited the same behaviour. BeamNG exhibits the same symptoms as ```wine control```, detecting it as a controller but it isn't able to receive any inputs from the device.

I'm interested to see what you come up with. If you need any further details about my configuration for testing, let me know :)

krims0n32, 2024-08-07:
Chiming in here, I have the same wheel which worked fine a week or two ago but now I have the same issue as OP. Oversteer works, games do see the wheel but no input is received. Tried ACC, BeamNG and Eurotruck simulator. In PS4/PS5 mode some buttons work but the axis do not. Tried various Proton versions but to no avail. I have some doubts it is related to Proton though as it worked before. My udev rules also haven't changed. My wheel is on the latest firmware.

Kimplul, 2024-08-07:
@krims0n32 A couple questions, if you don't mind:

Which distro are you using?
Did you already read through #61 and #54?
Have you tried games outside of Steam?

krims0n32, 2024-08-08:
@Kimplul Thanks for the quick reply.

I'm on Archlinux. I have read through the other tickets and tried various stuff but no luck sofar. fftest works, Oversteer works, Non-Steam works (tried BeamNG linux native (beta), wheel works apart from FFB but that might be a beta issue).

udev:
`KERNEL=="hidraw*", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b66d", MODE="0660", TAG+="uaccess"`

`wine control joy.cpl`

shows the wheel but no input is registered, no buttons no axis. Switching to PS4/5 mode some buttons do register but still no axis. So it seems to be a wine issue but I am super confused as it worked perfectly a few weeks ago (albeit only in PS4/5 mode). Proton uses its own wine version right? It does not rely on system-wine? Anything else I can check?

krims0n32, 2024-08-08:
Well what can I say, I updated my kernel and moved by udev rules further back (had 60, now 99), wheel works again in wine. No idea what's going on here, like I said it worked fine before.

There is something quirky going on still though. When I have my wheel connected and reboot or poweroff/poweron my PC, the wheel does not get initialized properly (there is no calibration happening on boot) and no wheel input is detected in wine. Only after disconnecting/reconnecting the USB cable calibration is done and the wheel input is ok in wine again. Let me know if I can debug this further.

Kimplul, 2024-08-08:
Pretty weird. Do you happen to know what the previous kernel version was? Do you have any other udev rules?

> When I have my wheel connected and reboot or poweroff/poweron my PC, the wheel does not get initialized properly (there is no calibration happening on boot)

Is the wheel still showing up, for example in `lsusb`? Is there anything about the wheel in `dmesg`?

Kimplul, 2024-08-08:
Alright, I finally got around to installing Fedora 40, it indeed came with Wine 9.5 and I'm afraid I can't reproduce. All buttons work on my wheel (note that it's not a GT version, just the plain RS) in `wine control`, as well as Oversteer. I haven't tried Steam yet, I believe it installs some Udev rules and there *might* be some kind of weird cross-influencing going on but that's probably what I'll do next.

I also tried booting up an old Arch VM I had lying around, with the same results. Things just worked, although since Arch is rolling release I can't be entirely sure I didn't skip over something like a broken kernel version.

@8L4YD @krims0n32 can you remember if you've installed anything else on your machines? Doesn't have to be a super accurate list, just wondering if there might be some package that's messing with permissions or something.

Kimplul, 2024-08-08:
Installed steam as described in https://docs.fedoraproject.org/en-US/gaming/proton/, no change.

krims0n32, 2024-08-09:
@Kimplul strange. I spent a couple of hours debugging just now.

The wheel shows up in lsusb as 044f:b66e after reboot or cold boot and oversteer works, evtest works, fftest works. Except wine/proton which sees the wheel but does not detect input until after I replug the usb cable. I do see some difference between reboot and replug in the journalctl output, When replugging I see hid-thrustmaster doing some stuff, even though it is blacklisted (confirmed with modprobe --showconfig|grep blacklist). Weird, but the wheel works after replug.

Other stuff I tried:
Uninstall steam => no change
Add TAG+="seat" to udev rules => no change
Add entire vendor class to udev rules => no change
Call udevadm trigger after boot => no change

So I suspect it is not udev related, also because my other controllers (Virpil) work just fine in the wine control panel after a reboot. Instead it looks like the wheel does not get initialized on boot in the same way it gets initialized when replugging after bootup. This then causes wine to not see the wheel as a wheel or something like that.

I have attached relevant journalctl output of boot and replug, maybe you can make something out of it.

[reboot.txt](https://github.com/user-attachments/files/16559806/reboot.txt)
[replug.txt](https://github.com/user-attachments/files/16559807/replug.txt)