Kimplul/hid-tmff2 #115: T128 0 when not pressing pedals

Henry-Hiles, 2024-08-13:
Right now the pedals report 1 when not pressed down, and -1 when pressed, which breaks some games like trackmania. Would there be a way for me to make it 0 when not pressed? Thanks!

Kimplul, 2024-08-14:
You might want to look at https://manpages.debian.org/testing/joystick/evdev-joystick.1.en.html, it should allow you to set the input range.

Out of curiosity, what exactly happens when the value is above 0? Most games tend to do just fine with the default range mentioned in the above link and I'm wondering if it might be possible to set the range in-game rather than globally, as that might have some knock-on consequences in other games.

Henry-Hiles, 2024-08-14:
> You might want to look at https://manpages.debian.org/testing/joystick/evdev-joystick.1.en.html, it should allow you to set the input range.
>
> Out of curiosity, what exactly happens when the value is [...]

Trackmania doesn't support analog steering, and thinks that whenever the value is not 0, the pedal is pressed, and therefore it should accelerate / brake. I found this out with a trackmania community support, and there is no way to fix it in game :) I'll check out evdev, thanks!;

Chris92de, 2024-08-14:
To clarify, Trackmania does support analog steering, however not analog acceleration/braking.
It supports keybinds for analog acceleration and brake, but only to provide compatibility with Gamepads that use analog triggers (Dualshock 4, Dualsense, Xbox 360/One/Series controller, etc). Actual acceleration/braking is always completely digital, either you accelerate/brake or you don't.

Henry-Hiles, 2024-08-14:
> To clarify, Trackmania does support analog steering, however not analog acceleration/braking. It supports keybinds for analog acceleration and brake, but only to provide compatibility with Gamepads [...]

Oops yeah I said the wrong thing, edited now 📃

Henry-Hiles, 2024-08-16:
> You might want to look at https://manpages.debian.org/testing/joystick/evdev-joystick.1.en.html, it should allow you to set the input range.
>
> Out of curiosity, what exactly happens when the value is [...]

hmm, this doesn't seem to do anything. I set it on the right axis and it does nothing. However, if i look at evtest, the evtest value for these axis neer goes below 0 anyways so now im confused

Henry-Hiles, 2024-08-16:
Its not just trackmania though, something like https://hardwaretester.com/gamepad picks it up as -1 too.

Henry-Hiles, 2024-08-18:
I think combined pedals can maybe fix this for trackmania, ill wait for that.

Kimplul, 2024-08-20:
Hello, sorry it took a while to respond, I've had a couple busy weeks lately. Anycase, I snooped around a bit and looks like the hardware tester uses the `joydev` Linux subsystem, which has its own way to set input ranges. I probably should've mentioned that earlier, it's technically speaking deprecated in favour of `evdev` but I guess it still pops up here and there. Trackmania presumably uses it, haven't tested it myself but the following configuration gives me 0 at no input at least in the hardware tester you linked:

```
jscal -s 11,1,255,32768,32768,21844,21844,1,3,1023,1023,1394469,1394469,1,0,112,142,5534751,5534751,1,0,112,142,5534751,5534751,1,0,112,142,5534751,5534751,1,3,1023,1023,1394469,1394469,1,3,1023,1023,1394469,1394469,1,0,112,142,5534751,5534751,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912,1,0,112,142,5534751,5534751 /dev/input/js0
```

Add this to a script and run it whenever you start Trackmania, should hopefully work. I maybe wouldn't recommend setting this as a global setting since it is kind of a hack and probably breaks other games.

Kimplul, 2024-08-20:
> Oh thanks, I'll try it, but I wonder why trackmania and the website both still worked when I modprobe -r joydev?

Website broke for me when I removed the `joydev` module after connecting the wheel. However, unloading first and then connecting the wheel automatically reloads the module, I'm guessing that's what happened here.

Henry-Hiles, 2024-08-20:
Oh okay thanks! I'll test it rn!

Henry-Hiles, 2024-08-20:
> Website broke for me when I removed the `joydev` module after connecting the wheel. However, unloading first and then connecting the wheel automatically reloads the module, I'm guessing that's what [...]

No, 100% it still works after. Is this different per-browser? I'm using firefox.

Kimplul, 2024-08-20:
That's odd. Firefox 115 for me. Could maybe be that you're triggering a reload and I'm not, unsure why/how though. Maybe worth cheking if the module has reappered in `lsmod`.

I'm not 100% sure if Linux forcefully rips out modules whose files are still open in some program. I believe it does, but there's a possibilty that if the gamepad detector is currently using the `joydev` subsystem it might silently be left in place. I believe I removed the module first and then opened the test page, and it wouldn't detect my wheel.

Henry-Hiles, 2024-08-20:
I didn't check lsmod, but i did see if the file existed in /dev/input/by-id, and it got removed when I used modprobe. Ill check lsmod and try opening it after later today.

Henry-Hiles, 2024-08-20:
That command also doesnt impact gamepad tester for me.

Henry-Hiles, 2024-08-20:
It does affect jstest though

Henry-Hiles, 2024-08-20:
Oh, your script does affect chromium though, just not firefox!

Henry-Hiles, 2024-08-20:
It also doesnt affect trackmania. Both must use evdev. If it matters, TrackMania runs through proton.

Henry-Hiles, 2024-08-20:
Yep Firefox uses evdev, that's why: https://bugzilla.mozilla.org/show_bug.cgi?id=1680982

Chromium must not.

I did try evdev-joystick, but I wasn't able to make it work.

Kimplul, 2024-08-21:
Ah, that explains it. I use debian's `firefox-esr`, i.e. the **E**xtended **S*upport **R**elease, which apparently doesn't include that patch.

Anyway, so back to `evdev`, this works for the hardware tester on my machine:
```
evdev-joystick --evdev /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Racing_Wheel_FFB-event-joystick --axis 1 --minimum 0 --maximum 2046
evdev-joystick --evdev /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Racing_Wheel_FFB-event-joystick --axis 6 --minimum 0 --maximum 2046
evdev-joystick --evdev /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Racing_Wheel_FFB-event-joystick --axis 5 --minimum 0 --maximum 2046
evdev-joystick --evdev /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Racing_Wheel_FFB-event-joystick --axis 6 --minimum 0 --maximum 2046
```

The actual range of the pedals is `[0 - 1023]`, so we tell `evdev` that we want to pretend the range is `[0 - 2046]`. This means that the top position of `1023` is registered as being in the middle of the range, giving us a 0 in hardwaretester. Again, probably breaks other games, probably good to place in a script and potentially make another one that resets the pedals, etc.

Also, reload the page. Apparently the modifications don't take effect immediately after executing the above commands, rather once a device is reopened by a client.

Henry-Hiles, 2024-08-21:
Unfortunately, it doesn't work for trackmania, because it needs to go from 0 when not pressed to a positive number when pressed, whereas this goes from 0 to negative, so the pedals are never pressed according to trackmania

Kimplul, 2024-08-21:
That's interesting, thought that anything non-zero would work based on earlier comments. Anyway, try swapping `--minimum` and `--maximum` and see what happens. If we're lucky it'll just invert the range.

Henry-Hiles, 2024-08-21:
> That's interesting, thought that anything non-zero would work based on earlier comments. Anyway, try swapping `--minimum` and `--maximum` and see what happens. If we're lucky it'll just invert the range.

Yeah I thought that too until recently. I'll try swapping those soon, thanks!

Henry-Hiles, 2024-08-24:
```
evdev-joystick --evdev /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Advance_Racer-event-joystick --axis 1 --minimum 2046 --maximum 0
evdev-joystick --evdev /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Advance_Racer-event-joystick --axis 5 --minimum 2046 --maximum 0
```
This appears to work perfectly, tysm!!

closed 2024-08-24 by Henry-Hiles