Kimplul/hid-tmff2 #117: Multiple Undefined errors when trying to compile hid-tmff2/deps/hid-tminit/hid-tminit-new.ko

mattmcadoo, 2024-08-25:
```
$ make
make -C deps/hid-tminit KDIR="/lib/modules/6.10.6-gentoo-x86_64/build"
make[1]: Entering directory '/home/mcadoo/git/hid-tmff2/deps/hid-tminit'
make -C /lib/modules/6.10.6-gentoo-x86_64/build M=/home/mcadoo/git/hid-tmff2/deps/hid-tminit modules
make[2]: Entering directory '/usr/src/linux-6.10.6-gentoo'
  CC [M]  /home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit.o
  CC [M]  /home/mcadoo/git/hid-tmff2/deps/hid-tminit/tminit.o
  LD [M]  /home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.o
  CC [M]  /home/mcadoo/git/hid-tmff2/deps/hid-tminit/usb-tminit.o
  LD [M]  /home/mcadoo/git/hid-tmff2/deps/hid-tminit/usb-tminit-new.o
  MODPOST /home/mcadoo/git/hid-tmff2/deps/hid-tminit/Module.symvers
ERROR: modpost: "__hid_register_driver" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
ERROR: modpost: "memcpy" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
ERROR: modpost: "kfree" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
ERROR: modpost: "usb_get_intf" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
ERROR: modpost: "hid_hw_stop" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
ERROR: modpost: "usb_interrupt_msg" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
ERROR: modpost: "usb_put_dev" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
ERROR: modpost: "__stack_chk_fail" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
ERROR: modpost: "usb_get_dev" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
ERROR: modpost: "usb_submit_urb" [/home/mcadoo/git/hid-tmff2/deps/hid-tminit/hid-tminit-new.ko] undefined!
WARNING: modpost: suppressed 35 unresolved symbol warnings because there were too many)
make[4]: *** [scripts/Makefile.modpost:145: /home/mcadoo/git/hid-tmff2/deps/hid-tminit/Module.symvers] Error 1
make[3]: *** [/usr/src/linux-6.10.6-gentoo/Makefile:1886: modpost] Error 2
make[2]: *** [Makefile:240: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-6.10.6-gentoo'
make[1]: *** [Makefile:4: all] Error 2
make[1]: Leaving directory '/home/mcadoo/git/hid-tmff2/deps/hid-tminit'
make: *** [Makefile:16: deps/hid-tminit] Error 2
```
commit 580cabe8de447f18ba2a655a096989f6cfdb8cb8
6.10.6-gentoo-x86_64 kernel

Kimplul, 2024-08-25:
Did you configure your kernel yourself? I had a go at installing Gentoo, but the pre-configure `gentoo-kernel` seems to be at version 6.6.47 and has a different suffix (`-gentoo-dist` versus your `-gentoo-x86_64`). `gentoo-kernel` seems to build the module just fine.

mattmcadoo, 2024-08-25:
Yes, it's a kernel configuration I've tailored to my system.  Using the latest sys-kernel/gentoo-sources at 6.10.6.

mattmcadoo, 2024-08-25:
kernel config

mattmcadoo, 2024-08-25:
kernel config
[kernel-config-6.10.6-gentoo-x86_64.gz](https://github.com/user-attachments/files/16742304/kernel-config-6.10.6-gentoo-x86_64.gz)

Kimplul, 2024-08-26:
Thanks, I can try playing around with the config and see if I can replicate. I'm assuming there's some parameter that's not set, possibly `HID_something` or maybe something related to modules, not sure. I'm unfortunately a bit busy at the moment but I hope I can get to it by the end of the week. Of course, feel free to look around yourself if you feel like it.

mattmcadoo, 2024-08-27:
Has GCC 14.2 been tested on compiling the module?

Kimplul, 2024-08-27:
Yep, at least Debian Testing's `gcc (Debian 14.2.0-1) 14.2.0` works.

Kimplul, 2024-08-27:
Okay, I snooped around a little, as a treat. I noticed that `Modules.symvers` in `/usr/src/linux` (with your config + 6.10.6 sources eselected) was empty for whatever reason. `Modules.symvers` is the file that `modpost` uses to link external modules (if I understand it correctly). Running `./scripts/mod/modpost -M -m -o Module.symvers  -T modules.order -t vmlinux.o` from `/usr/src/linux` populates `Module.symvers` and the module compilation succeeds, but it's still a bit unclear to me why this doesn't happen automatically.

Kimplul, 2024-08-27:
Alright, `TRIM_UNUSED_KSYMS` was `=y`, switched it to be `=n`, that seems to get the module compiling as it should. `TRIM_UNUSED_KSYMS` just adds the `-t` option to `scripts/mod/modpost`, as the name suggests, trimming out unused exports. In this case I guess it trims out all exports.

mattmcadoo, 2024-08-28:
Nice!  I'm looking forward to trying it out when you get things pushed over.

Kimplul, 2024-08-29:
`TRIM_UNUSED_KSYMS` is in your kernel config. You should change it, recompile, reinstall, `grub-mkconfig`, etc. Nothing I can really do about it. Guess I was about vague.

Kimplul, 2024-11-03:
Doing some winter cleanup, I never got a confirmation that my suggestion worked but if it didn't, feel free to reopen.

closed 2024-11-03 by Kimplul