Kimplul/hid-tmff2 #118: games not responding to wheel

ledaquepangarcon, 2024-09-03:
I installed the tmff2 driver, and the windows driver via proton along with it. Oversteer takes the inputs perfectly, but all games I try give no response. How do I go about troubleshooting this issue? I'm running the games via proton on my steam deck in desktop mode.

Kimplul, 2024-09-03:
Sounds like it might be a Wine/Proton/Steam issue, there are a few existing issues about stuff in that wheelhouse, see #90 and 112 in particular. For starters, maybe check if games or programs outside of Steam work, for example Speed Dreams is an open-source racing game that reportedly works well.

ledaquepangarcon, 2024-09-04:
Thanks for the swift reply. I have installed Speed Dreams. The built-in controller configurator in Speed Dreams does recognise the inputs from my wheel and pedals, but when I start a race, the wheel and pedals do nothing. The wheel does vibrate though, as Speed Dreams simulates the engine resonation at low rpms. One thing that I forgot to mention in my first post is that I'm using a thrustmaster T300. I'll read through #90 and #112 when I have time later today.

ledaquepangarcon, 2024-09-04:
I indeed seem to have an issue very similar to #90 and #112, but both threads have no conclusive solution yet. Perhaps the steam deck software has a quirk somewhere that messes with the driver?

ledaquepangarcon, 2024-09-21:
Hey I’ve got an update. I tried running some games with the wheel again, and ets2 works! (?) beamng and AC are still unresponsive.
The wheel does have a large deadzone in ets2, 90° to the left and to the right. Adding the udev rule from the readme does not fix it, sadly

Kimplul, 2024-09-22:
> ets2 works

Are you still running all games through Proton? ETS2 has a Linux native port (and I guess BeamNG has as well), I would expect that the Linux native version works since it doesn't rely on Proton. Otherwise, interesting.

> 90° to the left and to the right

That's really quite massive, weird. Can't remember running into anything similar before, but you could try setting the deadzone via `joydev`, see https://github.com/Kimplul/hid-tmff2/issues/42#issuecomment-1318632097. My understanding is that `joydev` is deprecated, but it seems to pop up here and there, this could be one of those situations. Wine/Proton uses `evdev` since ages ago, I believe.

rcardenes, 2025-05-23:
Hi all,

@ledaquepangarcon: I got a new wheel a few days ago and I've done a lot of fiddling. I also **had** a huge deadzone for my T128X while playing ETS2 (something like -20 to 20 deg from center).

This was a bit mysterious, because I can see the whole range of motion in, say, Oversteer... but today I finally tried `jstest` and, to my surprise, it showed the deadzone as well!

That got me thinking that **maybe** it was about the correction coefficients, which tmff2's default udev file sets to the same values as for the T248. So I got to play with them, and after some 30 minutes of adjusting... I managed it. No deadzone and full range of motion.

@Kimplul we might want to change the values for the T128/T128X.