Kimplul/hid-tmff2 #119: T128 refusing to work with Steam games even after installing drivers through protontricks (version 1.11.1)

m4xed0ut, 2024-09-06:
I'm using Linux Mint 21 and the wheel used to work perfectly after installing tmff2 on previous install of the distro. Now for some reason I can't get it to work. I remember having an older version of protontricks on my previous install of Mint which might be the system package version and it seemed to work well under that one. Is it possible to get it to work under protontricks from flathub?

Thanks in advance.

Kimplul, 2024-09-07:
Hm, there's been a string of similar sounding issues lately, see #118, #112 and #90 for example. I've so far been unable to replicate on my machine running Debian, but I'm wondering if there's been some update to Steam or something that's messing with the wheel.

m4xed0ut, 2024-09-07:
Thanks for the response. I'll look into those three issues when I get the desire to plug my wheel again :). It's good to see this being talked about at the very least. I'll definitely let you know whenever I go through these issues. I've been mostly blaming updating protontricks without any second thought (and now I can't go back to the 1.7.0 software package version lol), but I don't really know if it's really the case.