Kimplul/hid-tmff2 #121: Compiler not found+Error 127 and multiple error 2

Fire-Angel-13, 2024-10-03:
elise@elise-MS-7C91:~$ git clone --recurse-submodules https://github.com/Kimplul/hid-tmff2.git
cd hid-tmff2
make
sudo make install
Clonage dans 'hid-tmff2'...
remote: Enumerating objects: 1179, done.
remote: Counting objects: 100% (423/423), done.
remote: Compressing objects: 100% (156/156), done.
remote: Total 1179 (delta 307), reused 336 (delta 259), pack-reused 756 (from 1)
Réception d'objets: 100% (1179/1179), 2.13 Mio | 14.13 Mio/s, fait.
Résolution des deltas: 100% (702/702), fait.
Sous-module 'deps/hid-tminit' (https://github.com/Kimplul/hid-tminit.git) enregistré pour le chemin 'deps/hid-tminit'
Clonage dans '/home/elise/hid-tmff2/deps/hid-tminit'...
remote: Enumerating objects: 151, done.        
remote: Counting objects: 100% (31/31), done.        
remote: Compressing objects: 100% (16/16), done.        
remote: Total 151 (delta 20), reused 17 (delta 15), pack-reused 120 (from 1)        
Réception d'objets: 100% (151/151), 42.93 Kio | 976.00 Kio/s, fait.
Résolution des deltas: 100% (86/86), fait.
Chemin de sous-module 'deps/hid-tminit' : '17d6cad0b508417193ebf4d23f67c845487e9932' extrait
make -C deps/hid-tminit KDIR="/lib/modules/6.8.0-45-generic/build" 
make[1] : on entre dans le répertoire « /home/elise/hid-tmff2/deps/hid-tminit »
make -C /lib/modules/6.8.0-45-generic/build M=/home/elise/hid-tmff2/deps/hid-tminit modules
make[2] : on entre dans le répertoire « /usr/src/linux-headers-6.8.0-45-generic »
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           
  CC [M]  /home/elise/hid-tmff2/deps/hid-tminit/hid-tminit.o
/bin/sh: 1: gcc-12: not found
make[4]: *** [scripts/Makefile.build:243 : /home/elise/hid-tmff2/deps/hid-tminit/hid-tminit.o] Erreur 127
make[3]: *** [/usr/src/linux-headers-6.8.0-45-generic/Makefile:1925 : /home/elise/hid-tmff2/deps/hid-tminit] Erreur 2
make[2]: *** [Makefile:240 : __sub-make] Erreur 2
make[2] : on quitte le répertoire « /usr/src/linux-headers-6.8.0-45-generic »
make[1]: *** [Makefile:4 : all] Erreur 2
make[1] : on quitte le répertoire « /home/elise/hid-tmff2/deps/hid-tminit »
make: *** [Makefile:16 : deps/hid-tminit] Erreur 2
[sudo] Mot de passe de elise :          
make -C deps/hid-tminit KDIR="/lib/modules/6.8.0-45-generic/build" install
make[1] : on entre dans le répertoire « /home/elise/hid-tmff2/deps/hid-tminit »
make -C /lib/modules/6.8.0-45-generic/build M=/home/elise/hid-tmff2/deps/hid-tminit modules_install
make[2] : on entre dans le répertoire « /usr/src/linux-headers-6.8.0-45-generic »
  DEPMOD  /lib/modules/6.8.0-45-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2] : on quitte le répertoire « /usr/src/linux-headers-6.8.0-45-generic »
make[1] : on quitte le répertoire « /home/elise/hid-tmff2/deps/hid-tminit »
make -C /lib/modules/6.8.0-45-generic/build M=/home/elise/hid-tmff2 modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-6.8.0-45-generic »
  DEPMOD  /lib/modules/6.8.0-45-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1] : on quitte le répertoire « /usr/src/linux-headers-6.8.0-45-generic »
depmod -A

I use Zorin OS 17.2

Kimplul, 2024-10-03:
Zorin OS seems to be Ubuntu-based, so you'll probably have to run something like
```
sudo apt install build-essential
``` 

It should install all required compilers and build tools. If not, you can try installing the specified version of `gcc` with
```
sudo apt install gcc-12
```

Fire-Angel-13, 2024-10-03:
Installing gcc12 worked, thank you

Kimplul, 2024-10-03:
Cheers, I'll close this then. Feel free to open up new issues if and when you encounter any.

closed 2024-10-03 by Kimplul