Kimplul/hid-tmff2 #122: Small "deadzone" when changing wheel direction (T300)

kiiro94, 2024-10-16:
First of all, thanks for the driver, it works amazingly.

One little thing that I noticed, is that whereas on Windows the wheel is responsive even with micro movements (down to the 0.1% on the control panel), on Linux there's a small "deadzone" if and only if I change the direction from left to right , or right to left.
As soon as I change the direction of travel of the wheel, there's a small deadzone.

I can kinda show it to you using Assetto Corsa content manager: I'm moving the wheel in the most linear way possible, and you can see a small pause in the input movement when the wheel goes from left to right, or right to left.

It feels like a couple degrees max of deadzone, but you can feel it in games, especially when going fast on a straight.

https://github.com/user-attachments/assets/5223efe9-a85e-4134-86c9-e9acb7de3a0c

kiiro94, 2024-10-16:
Fixed it by running 
evdev-joystick --e /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Racing_Wheel_FFB-event-joystick --d 0 --f 0

It works until next reboot/unplug of the wheel, and now the wheel behaves exactly like on Windows.

closed 2024-10-16 by kiiro94

kiiro94, 2024-10-17:
> Glad you got it working. Just to check, is this with the `udev` rule installed as outlined in the README?
> 
> https://github.com/Kimplul/hid-tmff2?tab=readme-ov-file#common-issues-and-notes

The udev rule did not work on my system (Fedora 41 beta). Dunno if I did something wrong. I had to execute that particular command to fix the deadzone.

kiiro94, 2024-10-18:
> I updated the `udev` part of the README, would you mind checking if it works for you?

No, I copied all the rules just to be sure, but they are not working.

Kimplul, 2024-10-18:
Interesting. Which wheel are you using? Does `dmesg` show the product ID when you plug the wheel in?