Kimplul/hid-tmff2 #123: No Force Feedback in Assetto Corsa

emma-abb, 2024-10-21:
I installed through the AUR and my T300RS is working fine except that there is no force feedback (trying to play Assetto Corsa). Furthermore all the force feedback options in oversteer are greyed out.

I don't really know which logs or configs are relevant sorry but just ask and I'll send whatever you need

Kimplul, 2024-10-21:
Oversteer being greyed out sounds like the module might not be loading correctly. Could you post the output of `sudo dmesg -w` when you plug the wheel in?

emma-abb, 2024-10-21:
```
[ 9447.426022] usb 1-6.2: new full-speed USB device number 21 using xhci_hcd
[ 9447.524013] usb 1-6.2: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 9447.524025] usb 1-6.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 9447.524029] usb 1-6.2: Product: Thrustmaster FFB Wheel
[ 9447.524032] usb 1-6.2: Manufacturer: Thrustmaster
[ 9447.531217] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.2/1-6.2:1.0/0003:044F:B65D.0017/input/input40
[ 9447.531699] hid-thrustmaster 0003:044F:B65D.0017: input,hidraw3: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-6.2/input0
[ 9447.549984] hid-thrustmaster 0003:044F:B65D.0017: Wheel with model id 0x204 is a Thrustmaster T300 Ferrari Alcantara Edition
[ 9447.550239] hid-thrustmaster 0003:044F:B65D.0017: Success?! The wheel should have been initialized!
[ 9447.730582] usb 1-6.2: USB disconnect, device number 21
[ 9448.176454] usb 1-6.2: new full-speed USB device number 22 using xhci_hcd
[ 9448.277511] usb 1-6.2: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[ 9448.277523] usb 1-6.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 9448.277527] usb 1-6.2: Product: Thrustmaster T300RS Racing wheel
[ 9448.277530] usb 1-6.2: Manufacturer: Thrustmaster
[ 9448.283990] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.2/1-6.2:1.0/0003:044F:B66E.0018/input/input41
[ 9448.284355] hid-generic 0003:044F:B66E.0018: input,hidraw3: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-6.2/input0

```

Kimplul, 2024-10-21:
`hid-generic` means that the module is indeed not being loaded at all. I'm not maintainer of the package and I'm not too familiar with Arch packaging, so as a stopgap would you mind installing the driver via the instructions in the README? That would at least show whether the module or the package is broken.

emma-abb, 2024-10-21:
Yay it's working now :D
I used the DKMS instructions on the github

Thank you so much for your help <3

Kimplul, 2024-10-21:
Right, sounds like there's something iffy with the package. I'll close this and look into the package with another issue tracking it.

Thanks for the report, feel free to open up new issues if/when you encounter them.

closed 2024-10-21 by Kimplul