Kimplul/hid-tmff2 #124: no force feedback and B1 isnt respondive in Oversteer

Goldentimsteer, 2024-11-03:
i have a t300rs wheel and i have been using Hid-tmff2 for a while until one day it stopped working, i have tested in both assetto corsa and horizon 5, the wheel works fine but i don't have any FFB, i have tried reinstalling the git but i didn't get anywhere, I have oversteer installed and i can see that the wheel is working and when I'm in oversteer my accelerator does not input anything but in game it does, also brake and clutch are switched. i am unable to play horizon 5 easily because b2 (right paddle) is either always pushed or not pushed at all. the biggest issue is just that FFB isn't working anymore.

Kimplul, 2024-11-03:
Hi, my initial guess would be that the driver for whatever reason is not being loaded anymore. Would you mind running `sudo dmesg -w`, plugging in the wheel, and posting the output? It should be able to confirm/deny my suspicion.

 >  [...] have tried reinstalling the git but i didn't get anywhere [...]

Do you mean the installation failed or something else? If it did fail, please post the error message as well.

> [...] b2 (right paddle) is either always pushed or not pushed at all [...]

Sorry, I don't entirely understand this bit. Do you mean that regardless of if you're pressing the paddle, in Oversteer it is stuck either 'pressed' or 'not pressed'? If so, that *might* indicate that the wheel has a loose connection to the base or that the wheel's hardware is simply broken. Unfortunately it's not exactly unheard of for Thrustmaster wheels to be a bit flaky, I've had a few break on me already. Maybe try reinstalling the wheel to the base?

Goldentimsteer, 2024-11-03:
> Hi, my initial guess would be that the driver for whatever reason is not being loaded anymore. Would you mind running `sudo dmesg -w`, plugging in the wheel, and posting the output? It should be able to confirm/deny my suspicion.
[dmesg -w.txt](https://github.com/user-attachments/files/17612455/dmesg.-w.txt)

Kimplul, 2024-11-04:
I see `loading of unsigned module is rejected`, which would indeed mean that the module is not being loaded. You'll either need to sign the module or allow unsigned modules. Signing modules is arguably the safer method, but requires a bit of manual intervention, whereas disabling UEFI secure boot is the 'quick and dirty' way to allow unsigned modules.
See https://wiki.ubuntu.com/UEFI/SecureBoot/DKMS for example, although there are a bunch of other resources as well.

For future reference, which distro are you using?

Goldentimsteer, 2024-11-05:
> I see `loading of unsigned module is rejected`, which would indeed mean that the module is not being loaded. You'll either need to sign the module or allow unsigned modules. Signing modules is [...]
ok i will look into that and see what i can get working
> For future reference, which distro are you using?
i am using pop os (Debian)