Kimplul/hid-tmff2 #125: AUR package issues

Kimplul, 2024-11-03:
@LukasLichten There was recently an issue (#123) about the driver not working when installed through the AUR package, I tried replicating on a virtual machine but seemed to work just fine. I'm not too familiar with the AUR, do you have some input on what could be happening?

On a sidenote, apparently there's at least one improvement suggestion on the package: https://aur.archlinux.org/packages/hid-tmff2-dkms-git#comment-991633, probably not related...?

creeper82, 2024-12-23:
I agree. Luckily manual reinstallation made the module work again. After an hour of debugging I finally found these issues. Thank God... I was already sure that Oversteer app is bugged

LukasLichten, 2024-12-24:
> @LukasLichten There was recently an issue (#123) about the driver not working when installed through the AUR package, I tried replicating on a virtual machine but seemed to work just fine. I'm not too familiar with the AUR, do you have some input on what could be happening?

I don't know how I missed the email notifying me of this, but well I did miss it

> On a sidenote, apparently there's at least one improvement suggestion on the package: https://aur.archlinux.org/packages/hid-tmff2-dkms-git#comment-991633, probably not related...?

This was honestly in large parts just a nitpick, a lot of wheel driver packages are automatically inserting the kernel module, I had forked the .install from another such driver.  
The only issues this caused is that if you installed a newer kernel version it would pull the kernel module to install it for the new kernel, so you end up without a driver till reboot.  
  
But something that probably did/is causing issues is the `hid_thrustmaster` module loading itself, as you had to amply the blacklist manually as specified in [Common Issues](https://github.com/Kimplul/hid-tmff2?tab=readme-ov-file#common-issues-and-notes). I was able to use the kernel module just fine this way.  
  
The [Arch DKMS package guidelines](https://wiki.archlinux.org/title/DKMS_package_guidelines), which are quite hard to find (compared to the standard AUR guidelines), did however also specify a way to include the blacklist.conf, so starting with the newest version of the PKGBUILD the previous issue is addressed, and the blacklist should (hopefully) make it work now for everyone. [The Commit](https://aur.archlinux.org/cgit/aur.git/commit/?h=hid-tmff2-dkms-git&id=325764bc8715e83a7057d2647c5208e24c9f9cd9).