Kimplul/hid-tmff2 #126: Deadzone @ T128

Rolli700, 2024-11-17:
Hello first and a big thank you to Kimplul and Contributors of this Project.

I have successfully installed the dkms module Version of the driver. The Device is listed on lsusb. Using the Device by Calibrate Game Controller Tool all Buttons are Working, also all axes, but there is alwas this Deadzone issue.
I tested with hid_thrustmaster black- as also !blacklisted. I'd also set the udev rule as mentioned for the Deadzone.
The Game im testing on is EuroTruckSimulator2, where the wheel is recognized and also the ForceFeedback is working, but it keeps having this 90° deadzone on the Y-Axis (wheel). Is there anything else i can try to set this to 0?

Running ETS2 on Debian12,  which is running natively on linux without proton, T128  Wheel Playstation Version, recognized as FFB-Wheel.

Tropingenie, 2024-11-21:
Also seeing this on a Thrustmaster T300, Nobara 40 running ETS2 natively. `jstest-gtk` and Oversteer I can see the wheel is responsive.

Rolli700, 2024-11-22:
Thnk you for your response. I'm trying this udev-rules:

SUBSYSTEM=="input", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b696", RUN+="/usr/bin/evdev-joystick --evdev %E{DEVNAME} --deadzone 0"

SUBSYSTEM=="input", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b696", RUN+="/usr/bin/jscal -s 6,1,1,32767,32767,16384,16384,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912 /dev/input/js%n"

#SUBSYSTEM=="input", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b696", RUN+="/usr/bin/evdev-joystick --s '%E{DEVNAME}' --f 0 --d 0"

The outcommented one i found googling about and was the first try as seen on some other website. But the rules doesnt't seem to affect anything. Did i miss something? Also i disconnected the 2nd Gamepad (logitech vendor here) as mentioned, did a reboot and retried. There is no change about the deadzone. Thrustmaster Driver still blacklisted, don't really understand if this is correct in my case.
Is there anything else i can try, like editing some config on foot? 

I noticed this: Using the KDE Calibrate Game Controller Tool just for checking i can see that the wheel (/dev/input/js0) is not responsive within the Deadzone. The value starts changing at the same point ~+/- 45° from center as the ingame response to the wheel input. (so trying to calibrate is not an option). The value ranges from left -32767 to 0 in the center to 32767 at right.

Kimplul, 2024-11-22:
> Is there anything else i can try, like editing some config on foot?

Just to check, did you try running the `evdev-joystick` and `jscal` commands from a terminal as well? I'm wondering if there's maybe some error that pops up, or if the `udev` rule might be failing to trigger for some reason. Don't have a T128 myself so I can't check, but the T300 not working is a bit weird.

The `evdev` path should be something like `/dev/input/by-id/usb-Thrustmaster_*`, tab around a bit and you should find it.

@Tropingenie You might also want to try running the command manually if you haven't tried that already.

> Thrustmaster Driver still blacklisted, don't really understand if this is correct in my case.

Should be fine either way, the `hid-thrustmaster`module is just an upstream version of `hid-tminit` that's by now a bit outdated. This driver automatically installs an up to date version as an external module.

Rolli700, 2024-11-22:
Kimplul! My Hero!
I tried the command on cli, i was not about to think of this myself:

$ sudo /usr/bin/jscal -s 6,1,1,32767,32767,16384,16384,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912 /dev/input/js1
jscal: joystick has different number of axes (11) than specified in command line (6)

The solution was to expand the command by a few axis. This currently works for me:

$ sudo /usr/bin/jscal -s 11,1,1,32767,32767,16384,16384,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912 /dev/input/js1

Thanks a lot!!!

Tropingenie, 2024-11-23:
Using the udev rule suggested in https://github.com/Kimplul/hid-tmff2/issues/42#issuecomment-1319939390 fixed it.