Kimplul/hid-tmff2 #127: Forcing hid_tmff_new to be used for the T248 wheel

mattmcadoo, 2024-11-18:
I'm having a problem where the T248 wheel is not acting like it's using the correct kernel driver.  It shows it being picked up in dmesg but lsmod is not showing it's used.  lshw is showing the usbhid driver being used.  According to this StackExchange post, there is a way of forcing a PCI device to use a specific driver, but I'm not seeing anything in the corresponding /sys/bus/usb/drivers that looks like kernel module.

Here's my kernel seeing the device and loading the module:
```
[  462.053600] usb 3-2.4: USB disconnect, device number 7
[  464.802583] usb 3-2.4: new full-speed USB device number 9 using xhci_hcd
[  464.913742] usb 3-2.4: New USB device found, idVendor=044f, idProduct=b696, bcdDevice= 1.00
[  464.913752] usb 3-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  464.913756] usb 3-2.4: Product: Thrustmaster Advance Racer
[  464.913759] usb 3-2.4: Manufacturer: Thrustmaster
[  465.002605] input: Thrustmaster Thrustmaster Advance Racer as /devices/pci0000:00/0000:00:03.1/0000:02:00.0/0000:03:08.0/0000:05:00.1/usb3/3-2/3-2.4/3-2.4:1.0/0003:044F:B696.0009/input/input21
[  465.002758] hid-generic 0003:044F:B696.0009: input,hidraw7: USB HID v1.00 Joystick [Thrustmaster Thrustmaster Advance Racer] on usb-0000:05:00.1-2.4/input0
[  465.036315] hid_tmff_new: loading out-of-tree module taints kernel.
[  465.067671] input: Thrustmaster Thrustmaster Advance Racer as /devices/pci0000:00/0000:00:03.1/0000:02:00.0/0000:03:08.0/0000:05:00.1/usb3/3-2/3-2.4/3-2.4:1.0/0003:044F:B696.0009/input/input22
[  465.067791] tmff2 0003:044F:B696.0009: input,hidraw7: USB HID v1.00 Joystick [Thrustmaster Thrustmaster Advance Racer] on usb-0000:05:00.1-2.4/input0
[  465.095875] tmff2 0003:044F:B696.0009: force feedback for T248
```

This is lshw showing the driver as usbhid:
```
T:  Bus=03 Lev=02 Prnt=11 Port=03 Cnt=01 Dev#=  9 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=044f ProdID=b696 Rev=01.00
S:  Manufacturer=Thrustmaster
S:  Product=Thrustmaster Advance Racer
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=01(O) Atr=03(Int.) MxPS=  64 Ivl=4ms
E:  Ad=82(I) Atr=03(Int.) MxPS=  64 Ivl=2ms
```

Listing of /sys/bus/usb/drivers:
```
threadripper /sys/bus/usb/drivers # l
total 0
drwxr-xr-x 2 root root 0 Nov 17 20:43 'Philips webcam'
drwxr-xr-x 2 root root 0 Nov 17 20:43  aqc111
drwxr-xr-x 2 root root 0 Nov 17 20:43  asix
drwxr-xr-x 2 root root 0 Nov 17 20:43  ati_remote2
drwxr-xr-x 2 root root 0 Nov 17 20:43  ax88179_178a
drwxr-xr-x 2 root root 0 Nov 17 20:43  catc
drwxr-xr-x 2 root root 0 Nov 17 20:43  cdc_ether
drwxr-xr-x 2 root root 0 Nov 17 20:43  cdc_ncm
drwxr-xr-x 2 root root 0 Nov 17 20:43  cdc_subset
drwxr-xr-x 2 root root 0 Nov 17 20:43  em28xx
drwxr-xr-x 2 root root 0 Nov 17 20:43  hub
drwxr-xr-x 2 root root 0 Nov 17 20:43  iforce
drwxr-xr-x 2 root root 0 Nov 17 20:43  kaweth
drwxr-xr-x 2 root root 0 Nov 17 20:43  lan78xx
drwxr-xr-x 2 root root 0 Nov 17 20:43  net1080
drwxr-xr-x 2 root root 0 Nov 17 20:43  pegasus
drwxr-xr-x 2 root root 0 Nov 17 20:43  r8152
drwxr-xr-x 2 root root 0 Nov 17 20:43  r8152-cfgselector
drwxr-xr-x 2 root root 0 Nov 17 20:43  r8153_ecm
drwxr-xr-x 2 root root 0 Nov 17 20:43  rtl8150
drwxr-xr-x 2 root root 0 Nov 17 20:43  s2255
drwxr-xr-x 2 root root 0 Nov 17 20:43  snd-usb-audio
drwxr-xr-x 2 root root 0 Nov 17 20:43  uas
drwxr-xr-x 2 root root 0 Nov 17 20:43  usb
drwxr-xr-x 2 root root 0 Nov 17 20:43  usb-storage
drwxr-xr-x 2 root root 0 Nov 17 20:43  usb_serial_simple
drwxr-xr-x 2 root root 0 Nov 17 20:43  usbfs
drwxr-xr-x 2 root root 0 Nov 17 20:43  usbhid
drwxr-xr-x 2 root root 0 Nov 17 20:43  usblp
drwxr-xr-x 2 root root 0 Nov 17 20:43  usbserial_generic
drwxr-xr-x 2 root root 0 Nov 17 20:43  usbtv
drwxr-xr-x 2 root root 0 Nov 17 20:43  uvcvideo
drwxr-xr-x 2 root root 0 Nov 17 20:43  zaurus
```

mattmcadoo, 2024-11-18:
sdl2-jstest does show the wheel and doing the test for it, moving the wheel does register the movement, but this is the only thing that does, every game both native Linux and in Steam/Proton doesn't register the wheel movements.  Pedals are hit or miss on being picked up.
```
$ sdl2-jstest -l
Found 3 joystick(s)

Joystick Name:     'Thrustmaster SimTask FarmStick Right'
Joystick GUID:     03008bc34f0400001604000011010000
Joystick Number:    0
Number of Axes:     8
Number of Buttons: 33
Number of Hats:     0
Number of Balls:    0
GameControllerConfig:
  missing (see 'gamecontrollerdb.txt' or SDL_GAMECONTROLLERCONFIG)

Joystick Name:     'Thustmaster T500 RS Gear Shift'
Joystick GUID:     030053514f04000060b6000011010000
Joystick Number:    1
Number of Axes:     2
Number of Buttons: 10
Number of Hats:     0
Number of Balls:    0
GameControllerConfig:
  missing (see 'gamecontrollerdb.txt' or SDL_GAMECONTROLLERCONFIG)

Joystick Name:     'Thrustmaster Advance Racer'
Joystick GUID:     030048cd4f04000096b6000000010000
Joystick Number:    2
Number of Axes:     9
Number of Buttons: 26
Number of Hats:     1
Number of Balls:    0
GameControllerConfig:
  missing (see 'gamecontrollerdb.txt' or SDL_GAMECONTROLLERCONFIG)
```

Kimplul, 2024-11-18:
That's pretty weird, thanks for the report.

> According to this StackExchange post, there is a way of forcing a PCI device to use a specific driver, but I'm not seeing anything in the corresponding /sys/bus/usb/drivers that looks like kernel module.

No link, but regardless I'm not sure the best way forward would be to force bind devices to this driver.

> This is lshw showing the driver as usbhid:

Shows `usbhid` for me as well, presumably `lshw` shows which driver is responsible for the physical operation of the device and not logical functionality, which is what this driver would be. I guess you could think of it like this driver piggybacking off `usbhid`?

Anycase, as for actual troubleshooting, which distro are you using and what happens if you unplug all other joysticks?

mattmcadoo, 2024-11-18:
Ah, here's the link:  https://unix.stackexchange.com/questions/134878/make-linux-load-specific-driver-for-given-device-realtek-nic  It does talk about using the bind/unbind for the driver.

I'm using Gentoo with a 6.11.9-gentoo-x86_64 kernel.  Removing the other devices doesn't seem to help with getting the wheel motion recognized.  I tried in SuperTuxKart.  The pedals are recognized, but not the steering wheel itself.

Kimplul, 2024-11-18:
Thanks for the info. I've heard of some games being confused by multiple different joysticks sending input at the same time, so glad that doesn't seem to be the case.

Oh, I just noticed, you're the one who opened up the issue about Gentoo last time, hello again! I still have the VM I set up last time, I'll have a go at booting it and see if I can replicate. I don't think I ever tried running any games in the VM so if this turns out to be a 'followup' of sorts, sorry, apparently didn't do my due diligence last time.

mattmcadoo, 2024-11-21:
I've tried some additional things without success, but figure they should be mentioned here.

I use Hyprland in Wayland, so I was thinking that it might be part of the problem (it still might be at this point) so I've tried to run the games through gamescope to see if the wheel would be picked up any differently, but no change.  https://github.com/ValveSoftware/gamescope  The wheel in Farming Simulator 25 still gets registered as the "X" button, and the brake pedal gets registered as the "Y" button.  The accelerate pedal seems to get registered as a pedal.  No improvement in this case.  The T248 wheel is the Xbox version, I'm always seeing those button names shown in the games.

Also, I had tried the install the Windows drivers for the wheel via protontricks as was suggested in the README, that was before opening this issue and I didn't notice anything change.

Here's the listing of the packages and versions of (relevant) Hyprland and Wayland things I have installed currently:
```
dev-libs/wayland-1.23.1
dev-libs/wayland-protocols-1.38
dev-util/hyprwayland-scanner-0.4.2
dev-util/wayland-scanner-1.23.1
gui-libs/egl-wayland-1.1.17_pre20241101
x11-base/xwayland-24.1.4
dev-libs/hyprland-protocols-0.4.0
dev-libs/hyprlang-0.5.3
dev-util/hyprwayland-scanner-0.4.2
gui-apps/nwg-dock-hyprland-0.3.2
gui-libs/hyprcursor-0.1.10
gui-libs/hyprutils-0.2.6
gui-libs/xdg-desktop-portal-hyprland-1.3.8
gui-wm/hyprland-0.45.1
gui-wm/hyprland-contrib-0.1-r1
sys-auth/hyprpolkitagent-0.1.1
```

Kimplul, 2024-11-21:
Thanks for the extra info, I just got a desktop up and running, intending to try some games. SuperTuxKart has some issues compiling shaders, probably something weird with the VM but might still take a little while to sort out. `sdl2-jstest` did work as you described, all buttons and axes seem to show up just fine.

I have the PS version of the T248, have you tried setting it to PC mode? I'd be a bit surprised if it did anything but at least it's something you can try while I'm trying to replicate. I might also have a look at Wayland at some point, I guess it could bring with it some input handling differences though I wouldn't necessarily hold my breath.

> Cool deal, let me know if you need any of the USE flag settings I have or any other debugging I can send over.

Missed this message the first time around but USE flags would be helpul, yes please.

mattmcadoo, 2024-11-22:
```
[ebuild   R    ] gui-libs/hyprutils-0.2.6::gentoo  0 KiB
[ebuild   R    ] dev-util/hyprwayland-scanner-0.4.2::gentoo  0 KiB
[ebuild   R    ] dev-util/wayland-scanner-1.23.1::gentoo  0 KiB
[ebuild   R    ] dev-libs/hyprlang-0.5.3::gentoo  0 KiB
[ebuild   R    ] sys-auth/hyprpolkitagent-0.1.1::guru  0 KiB
[ebuild   R    ] dev-libs/wayland-1.23.1::gentoo  USE="-doc -test" ABI_X86="32 (64) (-x32)" 0 KiB
[ebuild   R    ] dev-libs/wayland-protocols-1.38::gentoo  USE="-test" 0 KiB
[ebuild   R    ] dev-libs/hyprland-protocols-0.4.0::gentoo  0 KiB
[ebuild   R    ] gui-libs/hyprcursor-0.1.10::gentoo  0 KiB
[ebuild   R    ] media-libs/mesa-24.2.7::gentoo  USE="X llvm lm-sensors (opengl) proprietary-codecs vaapi vdpau vulkan vulkan-overlay wayland zstd -d3d9 -debug -opencl -osmesa (-selinux) -test -unwind -valgrind -xa" ABI_X86="32 (64) (-x32)" CPU_FLAGS_X86="sse2" LLVM_SLOT="18 -15 -16 -17" VIDEO_CARDS="radeonsi -d3d12 (-freedreno) -intel -lavapipe (-lima) -nouveau -nvk (-panfrost) -r300 -r600 -radeon (-v3d) (-vc4) -virgl (-vivante) -vmware -zink" 0 KiB
[ebuild   R    ] gui-libs/egl-wayland-1.1.17_pre20241101::gentoo  ABI_X86="(64) -32 (-x32)" 0 KiB
[ebuild   R    ] x11-base/xwayland-24.1.4::gentoo  USE="libei systemd (-selinux) -test -unwind -xcsecurity" 0 KiB
[ebuild   R    ] gui-libs/xdg-desktop-portal-hyprland-1.3.8::guru  USE="systemd (-elogind)" 0 KiB
[ebuild   R    ] gui-wm/hyprland-0.45.2::gentoo  USE="X systemd -legacy-renderer" 0 KiB
[ebuild   R    ] gui-apps/nwg-dock-hyprland-0.3.2::guru  0 KiB
[ebuild   R    ] gui-wm/hyprland-contrib-0.1-r1::guru  USE="grimblast hyprprop scratchpad shellevents swap" 0 KiB

Global USE flags:

USE="X a52 aac alsa apache2 archive awt -bindist caps cdda cddb \
     -consolekit curl device-mapper development dvd extensions extraengine \
     ffmpge flac fpm gd git glamor gnome-keyring gstreamer gtk3 \
     http ipv6 javascript json -kde -libav libnotify login lxc \
     matroska mms mmx mmxext mp3 mpeg mysql mysqli ncat ncurses -ndiff \
     networkmanager nfs nfsv4 nping offensive ogg opengl openh264 \
     rust-analyzer opus pdo php postproc profiling python -qt4 \
     redis sockets speex spice sqlite sse sse2 sse3 sse4.1 sse4.2 ssl \
     ssse3 ssh staging -subversion systemd sysvipc -telemetry \
     terminal theora threads v4l vaapi vdpau vim vim-syntax vlc \
     vorbis vpx -uxa webp tesseract wireshark-plugins x264 x265 xattr \
     xcomposite xinerama xml contrib xsl xslt xvfb xv xvid xvmc zip \
     slib zsh-completions context zstd trayicon epspdf extra \
     games graphics humanities music omega pdfannotextractor pstricks \
     publishers science tex4ht texi2html syntax-check soap mysqlnd \
     jemalloc xetex vnc lm_sensors hddtemp sensord btrfs graph \
     rrdcgi -share acl gcj nfsv41 -system-mitkrb5 pgo aio http3 \
     nginx widevine postgres uuid thunar network appindicator calendar \
     gdata pda xface tools visio dia boto -tracker xmlwriter avif \
     qml libass svc ao xrandr printsupport iproute2 down-root vte freerdp \
     gles2 egl gltf raw jpeg2k update_drivedb -examples elfutils \
     sysv-utils lzma -ruby gles atasmart tor-hardening -webkit mtp \
     contribdnn s3tc aalib xpm opencl -zeromq modplug sendto odf mod \
     -model-viewer vulkan openexr jpegxl gles1 libkms user-session \
     -bluetooth quic pdfimport hwaccel heif lto system-harfbuzz system-jpeg \
     bash-completion hip system-libvpx postscript system-icu \
     system-sqlite system-libevent gpg system-ffmpeg base vkd3d \
     zsh-completion tfshark lz4 maxminddb tray notification \
     kernel-install boot mediaplayer ofx wayland tproxy chromium drm raycast \
     jack fdk imagemagick gimp -javamail libvirt btpclient eap-tls \
     rocm rav1e svt-av1 gentoo-vm ofono ldac hcitop -packagekit \
     connection-sharing jingle omemo -abi_x86_32 lm-sensors -d3d9 \
     vulkan-overlay -enchant rust-src usbredir -netifrc ocr vector-icons \
     ozone srt browser hardened autotype dav1d qemu xephyr uki \
     ukify pipewire dbus -gnome-online-accounts amf amr amrenc \
     shaderc libdrm libtesseract openal gmp rubberband clippy opencv \
     faad drun ffmpeg dracut speech kate shout compositor rustfmt \
     -virtualbox uvm system-llvm -screencast iptables nftables hdri \
     proton uefi nvme flashrom amdgpu"
     
 Per package USE flags:

>=x11-base/xwayland-23.2.2 libei
media-gfx/blender openvdb_abi_11 llvm -openvdb -nanovdb
media-sound/lame sndfile
media-libs/embree raymask
media-libs/openimageio color-management openvdb_abi_11
media-libs/openpgl tbb
media-libs/opensubdiv tbb
media-gfx/openvdb openvdb_abi_11
[... 112 lines cut ...]
net-im/prosody -idn
sys-libs/zlib minizip

>=x11-libs/libdrm-2.4.117 video_cards_radeon
>=net-dns/dnsmasq-2.89-r1 script
>=net-libs/gnutls-3.8.1-r1 pkcs11
>=media-libs/opus-1.4 custom-modes
>=dev-libs/libpcre2-10.42-r1 pcre32
>=media-libs/suil-0.10.20 gtk2
>=media-libs/freetype-2.13.2 brotli
>=dev-libs/folks-0.15.6 eds
>=gnome-extra/evolution-data-server-3.50.1 vala
>=net-libs/libssh-0.10.5 server
>=dev-db/postgresql-16.1 kerberos
>=dev-libs/libical-3.0.17 vala
>=media-libs/freetype-2.13.2 harfbuzz
>=media-libs/libsdl2-2.28.5 haptic
>=net-misc/spice-gtk-0.42-r3 usbredir
>=app-misc/tracker-3.6.0 vala
>=dev-python/PyQt5-5.15.10-r1 sql
>=media-libs/imgui-1.90.6 glfw
>=dev-python/PyQt5-5.15.11 declarative
>=dev-libs/sleef-3.6.1-r2 cpu_flags_x86_avx512f
>=net-libs/libssh-0.10.6-r1 static-libs
```

mattmcadoo, 2024-11-22:
The wheel is set for PC mode.

Kimplul, 2024-11-22:
Alright, took a while to get VirGL set up but I can say I've at least partially replicated the issue. `sdl2-joydev` seems to work, but for example Speed dreams fails to read input at all from the wheel. SuperTuxKart seems to work just fine, interestingly. I tried Rally Dreams on Steam, and it does detect input for all the relevant axes but some buttons threw it off. The wheel was apparently being treated as a gamepad of some kind, so that might be a clue. I'll keep looking into this.

Kimplul, 2024-12-08:
Hello again, been a while. No real progress so far, unfortunately. I noticed that SpeedDreams did actually pick up some input, but for whatever reason it's treating the axes as buttons and effectively only reacting to when the wheel is turned all the way to the left or the right. I'm a bit stumped as to why, since SpeedDreams uses SDL, but neither `sdl2-jstest` nor `sdl-jstest` show anything like it. `evtest` and `jstest` also look perfectly normal.

I might look into if maybe SDL is initialized somehow differently between `sdl2-jstest` and SpeedDreams, I guess the significant difference is that one is windowed and the other isn't? Maybe? https://hardwaretester.com/gamepad seems to work just fine, though.

Kimplul, 2024-12-08:
Slight update, SpeedDreams started working after I deleted its config directory, `~/.speed-dreams`. Not entirely certain why, and Steam still seems to behave the same, so I guess this just means that SpeedDreams maybe wasn't a particularly good point of comparison.

mattmcadoo, 2024-12-11:
Glad to hear that you're able to replicate what I was experiencing.  It is definitely weird that low-level utils will see wheel as a range whereas all the games themselves will see it as mere binary buttons.

I'll update to the latest 6.12 kernel this weekend and see how that works out for reference.

mattmcadoo, 2024-12-26:
I didn't get around to testing this last week, but I was able to tonight.  Using the latest commit, 11cca6d4e9b49ec3779573f68917e362e4d65d36 as of right now, and a 6.12.6 kernel from the gentoo-sources "package" I am able to get the wheel working in 'Super Tux Kart' and 'American Truck Simulator', both native Linux games.  In 'American Truck Simulator' I got the wheel to register with the game after turning the wheel all way in one direction until it reached the end of the wheel's mechanical limitation.  No amount of lesser turns in either direction got it to register.  Once that was seen, then the wheel is super sensitive in the game to the slightest movements.  (IIRC, I did the same procedure in Super Tux Kart and got it to register, but the sensitivity in game is very low and I don't see a way of changing it in game.)

On the Proton side:  Farming Simulator 25 is still a no-go.  The wheel will only register for turning to the right, but not the left.  Even the pedals act up when they are applied.  IIRC, I did install the Thrustmaster drivers in Proton via protontricks previously.  Also to note, I'm not using gamescope anymore for the Windows games since Wayland support in Proton has gotten better.  (I've set Steam to use the new "Proton Hotfix" branch for all my games now.)

Definitely making progress and getting some good wins in.

Merry Christmas and Happy Yule!

mattmcadoo, 2024-12-26:
For Farming Simulator 25, I did try adding `PROTON_ENABLE_HIDRAW="0x044f/0xb696"` to the Steam launch options, but no change in how the game saw it.