Kimplul/hid-tmff2 #130: T300 not being recognized

Cimmerian-Iter, 2024-11-30:
Hello, steam deck user here, I installed the module through the AUR repo, and when i plug the T300 with F1 2011 wheel addon it says this

 9650.045735] usb 1-1.3: USB disconnect, device number 9
[ 9652.812924] usb 1-1.3: new full-speed USB device number 11 using xhci_hcd
[ 9652.925507] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 9652.925518] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 9652.925523] usb 1-1.3: Product: Thrustmaster FFB Wheel
[ 9652.925527] usb 1-1.3: Manufacturer: Thrustmaster
[ 9653.024867] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.3/1-1.3:1.0/0003:044F:B65D.001F/input/input58
[ 9653.025183] hid-thrustmaster 0003:044F:B65D.001F: input,hidraw6: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:04:00.3-1.3/input0
[ 9653.046527] hid-thrustmaster 0003:044F:B65D.001F: Unknown wheel's model id 0x203, unable to proceed further with wheel init
[ 9673.341434] usb 1-1.3: USB disconnect, device number 11

Is it the fact that I plugged the F1 Wheel addon that is the issue? The T300 is on V34 and has the switch to PC. 

Thanks

Kimplul, 2024-11-30:
> Is it the fact that I plugged the F1 Wheel addon that is the issue?

Kind of. The in-kernel `hid_thrustmaster` is an older version of `hid_tminit` that this driver depends on, specifically a version old enough that it treats your whole `0x0203` as a model number, when we later found out it's actually `0x02` for the base and `0x03` for the wheel. The newer `hid_tminit` should handle this situation correctly and is automatically installed when this driver is installed, so you should only need to blacklist `hid_thrustmaster`, see https://github.com/Kimplul/hid-tmff2?tab=readme-ov-file#common-issues-and-notes

There was an attempt to push a newer version into the kernel, but nothing seems to have happened in a while: https://github.com/scarburato/hid-tminit/issues/11

Cimmerian-Iter, 2024-12-01:
Ok thanks, I did wonder what happened to hid tminit, hope that there's progress on it side to have a smoother installation process.

So i did echo 'blacklist hid_thrustmaster' | sudo tee /etc/modprobe.d/hid_thrustmaster.conf, but i still have hid_thrustmaster on. Is it the blacklist that failed?

Kimplul, 2024-12-01:
Did the command succeed and did you reboot?

Cimmerian-Iter, 2024-12-01:
I did reboot, but i rebooted again and now it "work"?

[   82.377643] usb 1-1.2: USB disconnect, device number 4
[   84.390813] usb 1-1.2: new full-speed USB device number 5 using xhci_hcd
[   84.513604] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[   84.513618] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   84.513622] usb 1-1.2: Product: Thrustmaster FFB Wheel
[   84.513626] usb 1-1.2: Manufacturer: Thrustmaster
[  592.585789] usb 1-1.2: USB disconnect, device number 5
[  593.815141] usb 1-1.2: new full-speed USB device number 6 using xhci_hcd
[  593.925573] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  593.925586] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  593.925590] usb 1-1.2: Product: Thrustmaster FFB Wheel
[  593.925594] usb 1-1.2: Manufacturer: Thrustmaster

Oversteer doesn't open, but from what i've seen the fact that i changed my system language to french through kde made the whole thing explode for whatever reason. I booted up project cars 2 and it doesn't register any input

Cimmerian-Iter, 2024-12-01:
I will retry next week, Aur package in a steam deck doesn't blend well, i'll try the dkms way

Kimplul, 2024-12-01:
> I did reboot, but i rebooted again and now it "work"?

If that's the whole output, something is still messed up since there should be a message about `hid_tminit` at the very least. As you mentioned, the steam deck is not a typical computer so might be a good idea to try DKMS and go from there.

FWIW there might also be some issue with the AUR package, I've pinged the maintainer but no response yet: https://github.com/Kimplul/hid-tmff2/issues/125

Cimmerian-Iter, 2024-12-07:
It definitely isn't, first of all valve thought it was good to host outdated neptune headers in their repo, so your command in the readme is now useless for steam deck. Instead now we have to manually download the headers from their mirror https://steamdeck-packages.steamos.cloud/archlinux-mirror/jupiter-main/os/x86_64/ and install locally with pacman. 
 
Now that I have the correct headers, I have installed the aur package

(130)(deck@steamdeck SteamDeck_rEFInd)$ uname -r
6.5.0-valve22-1-neptune-65-g9a338ed8a75e
(deck@steamdeck SteamDeck_rEFInd)$ sudo dkms status
hid-tmff2/0.82: added
hid-tmff2/r270.412eac2, 6.5.0-valve22-1-neptune-65-g9a338ed8a75e, x86_64: installed

So it match my kernel version. But when loading the module I have this error

[  179.401277] usb 1-1.1: Product: Thrustmaster FFB Wheel
[  179.401282] usb 1-1.1: Manufacturer: Thrustmaster
[  179.486095] hid_tminit_new: loading out-of-tree module taints kernel.
[  179.486106] hid_tminit_new: module verification failed: signature and/or required key missing - tainting kernel
[  179.486612] BPF:      type_id=5861 bits_offset=0
[  179.486616] BPF:  
[  179.486618] BPF: Invalid name
[  179.486620] BPF: 
[  179.486623] failed to validate module [hid_tminit_new] BTF: -22

I guess the verification failed is just something we can ignore, but if it fail to validate is it because the package is malformed? Or is it still tied to signature? Secure boot isn't enabled

Cimmerian-Iter, 2024-12-07:
With dkms (not using aur package)

[   81.357149] usb 1-1.1: USB disconnect, device number 3
[   83.359493] usb 1-1.1: new full-speed USB device number 10 using xhci_hcd
[   83.482606] usb 1-1.1: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[   83.482622] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   83.482629] usb 1-1.1: Product: Thrustmaster FFB Wheel
[   83.482635] usb 1-1.1: Manufacturer: Thrustmaster
[   83.585637] BPF:      type_id=5861 bits_offset=0
[   83.585646] BPF:  
[   83.585650] BPF: Invalid name
[   83.585653] BPF: 
[   83.585657] failed to validate module [hid_tminit_new] BTF: -22

No signature issue, 

(deck@steamdeck SteamDeck_rEFInd)$ sudo dkms status
hid-tmff2/0.82, 6.5.0-valve22-1-neptune-65-g9a338ed8a75e, x86_64: installed

Kimplul, 2024-12-07:
Oh, interesting. Looks like the deck has some additional security checks that causes the `hid-tminit` module to not be loaded. I guess since the module name doesn't match the filename? Haven't seen that before.

I had a steam deck installation on a virtual machine but didn't see this behaviour, so this seems to be a newer addition. I might try reinstalling it but for now I pushed a test commit to a new `deck` branch in https://github.com/Kimplul/hid-tminit/tree/deck, would you mind checking if that lets `hid-tminit` load? Just run

+ `make`
+  `sudo make install`

You might want to uninstall `hid-tmff2` from DKMS as well to make sure all old versions of `hid-tminit` are removed. This won't make the wheel fully work, but if it gets the wheel to boot up we should be pretty close.

> It definitely isn't, first of all valve thought it was good to host outdated neptune headers in their repo, so your command in the readme is now useless for steam deck.

Thanks for the info, I guess you're on `linux-neptune-65` based on the logs you've sent. Would `sudo pacman -S linux-neptune-headers` work instead?

Cimmerian-Iter, 2024-12-08:
linux-neptune-headers-5.13.0.valve37-1 this is what neptune headers does install. No I checked some reddit about dkms and steam deck and yeah we need to manually download the headers

I will test the commit next week sorry only have access to the wheel when im home in the weekend

Kimplul, 2024-12-08:
> linux-neptune-headers-5.13.0.valve37-1 this is what neptune headers does install. No I checked some reddit about dkms and steam deck and yeah we need to manually download the headers

Oh wow, that's unfortunate. Could you link to the post? I'm curious about the details.

> I will test the commit next week sorry only have access to the wheel when im home in the weekend

Take your time, there's no rush.

Cimmerian-Iter, 2024-12-08:
https://www.reddit.com/r/SteamDeck/comments/y6uudu/comment/iuy1mod/?utm_source=share&utm_medium=mweb3x&utm_name=mweb3xcss&utm_term=1&utm_content=share_button

It's just other user who had the same problem with headers when they needed them to use dkms on deck. They found out that the repo of valve didn't have the correct headers

Cimmerian-Iter, 2024-12-13:
make -C /lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/build M=/home/deck/Downloads/hid-tminit-deck modules_install
make[1] : on entre dans le répertoire « /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/build »
  INSTALL /lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/updates/hid-tminit-new.ko
  SIGN    /lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/updates/hid-tminit-new.ko
At main.c:167:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: ./certs/signing_key.pem
  ZSTD    /lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/updates/hid-tminit-new.ko.zst
  INSTALL /lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/updates/usb-tminit-new.ko
  SIGN    /lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/updates/usb-tminit-new.ko
At main.c:167:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: ./certs/signing_key.pem
  ZSTD    /lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/updates/usb-tminit-new.ko.zst
  DEPMOD  /lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e
make[1] : on quitte le répertoire « /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/build »

during sudo make install

Now when plugging the wheel

[  117.599415] usb 1-1.3: new full-speed USB device number 13 using xhci_hcd
[  117.726471] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  117.726488] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  117.726494] usb 1-1.3: Product: Thrustmaster FFB Wheel
[  117.726500] usb 1-1.3: Manufacturer: Thrustmaster
[  117.835992] hid_tminit_new: loading out-of-tree module taints kernel.
[  117.836034] hid_tminit_new: module verification failed: signature and/or required key missing - tainting kernel
[  117.836492] BPF:      type_id=5861 bits_offset=0
[  117.836496] BPF:  
[  117.836498] BPF: Invalid name
[  117.836500] BPF: 
[  117.836504] failed to validate module [hid_tminit_new] BTF: -22

Still fail, does that have to do with the errors in make install where it also complained about keys?

Kimplul, 2024-12-15:
> Still fail, does that have to do with the errors in make install where it also complained about keys?

I'm not entirely sure, they're related to signing the modules ("I compiled this file, here's my key" kind of thing), the BPF error of `Invalid name` doesn't really sound like the keys would be relevant.

Searching around on the internet, there's at least this: https://www.reddit.com/r/pop_os/comments/xnjbx7/after_automatic_popos_update_invalid_name_any_idea/ The comments are related to `Pop!_OS` so maybe directly applicable to the Steam Deck, but seems to indicate that the kernel headers and running kernel are out of sync?

Cimmerian-Iter, 2025-01-10:
Closing the issue. Getting the headers manually did do that out of sync issue, instead using the 65 neptune from command line fixed.

I'll do a guide for steam deck, so that users knows what kernel version they have and which neptune version to get. It would streamline the process and not us having to open an issue each time for that.

[  328.572887] usb 1-1.3: new full-speed USB device number 9 using xhci_hcd
[  328.686553] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b66d, bcdDevice= 1.00
[  328.686566] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  328.686571] usb 1-1.3: Product: Thrustmaster Racing Wheel FFB
[  328.686575] usb 1-1.3: Manufacturer: Thrustmaster
[  328.789573] input: Thrustmaster Thrustmaster Racing Wheel FFB as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.3/1-1.3:1.0/0003:044F:B66D.000D/input/input42
[  328.790181] hid-generic 0003:044F:B66D.000D: input,hiddev100,hidraw11: USB HID v1.10 Gamepad [Thrustmaster Thrustmaster Racing Wheel FFB] on usb-0000:04:00.3-1.3/input0
[  328.804108] hid_tmff_new: loading out-of-tree module taints kernel.
[  328.804124] hid_tmff_new: module verification failed: signature and/or required key missing - tainting kernel
[  328.860481] input: Thrustmaster Thrustmaster Racing Wheel FFB as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.3/1-1.3:1.0/0003:044F:B66D.000D/input/input43
[  328.860916] tmff2 0003:044F:B66D.000D: input,hiddev100,hidraw11: USB HID v1.10 Gamepad [Thrustmaster Thrustmaster Racing Wheel FFB] on usb-0000:04:00.3-1.3/input0
[  328.865530] tmff2 0003:044F:B66D.000D: force feedback for T300RS

Thanks a lot kim I really appreciate your support