Kimplul/hid-tmff2 #131: Hello, I want to thank everybody in this thread and everybody involved in this project. After switching to Linux, I thought my expensive wheel setup would never be used again. After many months of periodically trying to get my TX working, I finally was able to get it working -- WITH force feedback -- in ETS2 for the first time last night.

Kimplul, 2024-12-07:
Hello, I want to thank everybody in this thread and everybody involved in this project. After switching to Linux, I thought my expensive wheel setup would never be used again. After many months of periodically trying to get my TX working, I finally was able to get it working -- WITH force feedback -- in ETS2 for the first time last night. 

Although there is one problem, the steering deadzone. I tried the udev rules listed in the readme, and even successfully ran the jscal and evdev-joystick commands in my own shell, but the deadzone did not change at all. (And yes, my steering deadzone in ETS2 was set to 0%.)

_Originally posted by @allylikesu in https://github.com/Kimplul/hid-tmff2/issues/48#issuecomment-2524580415_

Kimplul, 2024-12-07:
Hello @allylikesu, I moved your comment to a new thread.

In a recent issue (#126) it was noted that ETS2 apparently still uses `jsdev`, so would you mind running the `jscal` command from the udev rules file for your wheel manually in a shell and posting the output? I'm expecting there to be en error of some kind, possibly related to the number of axes the wheel has.

allylikesu, 2024-12-08:
Hello, my problem seems to have fixed itself. I'm pretty sure the problem was with `hid_thrustmaster` not being blacklisted properly. I noticed in my dmesg logs that it was showing up as hid-thrustmaster (with a dash), instead of hid_thrustmaster (with an underscore) as in the example command in the "'Common Issues and Notes" section. After creating a new hid-thrustmaster.conf file that blacklists hid-thrustmaster (note the dashes), the deadzone problem seems to have been resolved.
It was quite odd, as in dmesg I was seeing both hid-thrustmaster and tmff messages, and the wheel was showing up correctly (with the correct device id of b669), and force feedback was working, but hid-thrustmaster was still there to mess some things up I guess.

In conclusion, it might be a good idea to have a sentence after the blacklist command that says something along the lines of "the driver may also be hid-thrustmaster with a dash instead of an underscore. check your system and modify the command accordingly".

Kimplul, 2024-12-08:
Glad it works, but unfortunately I don't quite believe the solution you provided to be true. Thrustmaster wheels are kind of silly in that they initially show up as a 'Generic Thrusmaster FFB wheel`, and need a driver to 'tell' them what model they actually are, T300/T248/etc. `hid-thrustmaster` is an in-kernel driver that does just that, but it unfortunately doesn't handle a few cases properly, so this driver uses `hid-tminit` as a dependency and the blacklist just tells Linux to prefer `hid-tminit` over `hid-thrustmaster`. Once the wheel figures out what model it actually is, is 'reconnects' itself, and at that point this driver takes over and the `udev` rules etc. notice a 'new' device and take effect. If you were experiencing force feedback, it shouldn't really matter how the wheel was initialized, the same `udev` rules should still trigger. I wonder if you maybe rebooted and that reloaded the `udev` rules or something?

As for the underscore vs dash, Linux replaced dashes in the filename with underscores when  loading modules, and I believe the rules treat them interchangeably. Not entirely sure why, though.

Anycase, I'll close this since things seem to work, but if the deadzone turns up again, feel free to reopen.

closed 2024-12-08 by Kimplul