Kimplul/hid-tmff2 #133: Does not work on bazzite.

Iwo24pl, 2024-12-14:
How can i install it on bazzite? is shows some kernel issues.

Kimplul, 2024-12-14:
Haven't tried Bazzite, please show the commands and output that you tried, including the error message(s). Preferably as just raw text within a code block.

Iwo24pl, 2024-12-15:
make -C deps/hid-tminit KDIR="/lib/modules/6.11.10-304.bazzite.fc41.x86_64/build" 
make[1]: *Enter to catalog '/var/home/iwo/hid-tmff2/deps/hid-tminit'
make -C /lib/modules/6.11.10-304.bazzite.fc41.x86_64/build M=/var/home/iwo/hid-tmff2/deps/hid-tminit modules
make[2]: *Enter to catalog '/var/home/iwo/hid-tmff2/deps/hid-tminit'
make[2]: *** /lib/modules/6.11.10-304.bazzite.fc41.x86_64/build: Nie ma takiego pliku ani katalogu. Stop.
make[2]: *Exiting catalog '/var/home/iwo/hid-tmff2/deps/hid-tminit'
make[1]: *** [Makefile:4: all] Błąd 2
make[1]: *Exiting catalog '/var/home/iwo/hid-tmff2/deps/hid-tminit'
make: *** [Makefile:20: deps/hid-tminit] *Error 2

Iwo24pl, 2024-12-15:
Single * = translated from Polish

Iwo24pl, 2024-12-15:
Also Bazzite is based on Fedora Atomic, so I'm not sure if it will work, because its an ostree-based os

Iwo24pl, 2024-12-15:
I just installed Linux Mint. Most of the apps did not work on Bazzite.

Iwo24pl, 2024-12-15:
-

closed 2024-12-15 by Iwo24pl

Iwo24pl, 2024-12-16:
I had them installed.

Iwo24pl, 2024-12-16:
On normal Fedora there is dims, not on atomic