Kimplul/hid-tmff2 #134: T300 RS - no FFB

creeper82, 2024-12-23:
I'm on Arch, installed dkms manually as the AUR package has some problems apparently. Everything works fine, but FFB does not work at all. The only thing that works is the autocenter option in e.g. Oversteer

Whenever I use `fftest`, it shows the effect list with no errors, however when I play the effect, nothing happens. None of the games can apply the force feedback as well

How can I debug this further? A few months ago I had everything configured without any slightest issue. Now, I had a long break, some updates released in the meantime, and for whatever reason it doesn't work anymore. At all.

creeper82, 2024-12-23:
Maybe the error is related?
[image]

Kimplul, 2024-12-23:
> Everything works fine, but FFB does not work at all.

So everything, in fact, doesn't work fine. :P

> Maybe the error is related?

Don*t think so, the errors in the image are from `hid-thrusmtaster`, which is a separate driver from this one. Thrustmaster wheels are kind of silly, when you first plug one in it doesn't really know what wheel it is, and requires a driver to send some initialization commands that causes it to 'reboot' as the correct wheel. Unfortunately, this 'reboot' is done such that it break the USB protocol, which sometimes causes the USB subsystem to report some errors, which is the `URB to change wheel mode...` error.

The second error is because when you install this driver, it automatically installs an external version of `hid-thrustmaster` (`hid-tminit` to be precise) just so that we always make sure wheels get initialized. It's just reporting that you have a module built into the kernel as well as an external `hid-thrustmaster`.

FFB starts happening once `hid-thrustmaster` is done, and this `tmff2` driver is loaded, which seems to happen based on the `force feedback for T300RS` message. Good call looking at `dmesg`, but in the future, please just post the raw text from terminals, makes my life a bit easier. :)

> How can I debug this further?

If everything worked a few months ago, it *could* be that I've introduced some bug in some recent-ish commit. You might want to uninstall the `dkms` version and try checking out some old enough commit from this repo and manually install it to see if the wheel starts working. If it does, it's just down to finding the commit that introduced the bug. You'd probably want to reboot between each installation, but you might also get away with just running `sudo modpobe -r hid-tmff-new && sudo depmod -A && sudo modprobe hid-tmff-new`.

If no commit seems to work, could be that some other system package that was updated is interfering with the wheel, though finding out which one might be a bit tricky. A fairly common culprit would be something to do with `udev`, so you might want to try playing with the rules in `/etc/udev/rules.d/99-thrustmaster.rules` (assuming you do `sudo make udev-rules` first). I did recently make some changes to the rules, so this could be plausible, though I don't entirely see why *just* the FFB would stop working.

It is also possible that there's some problem with the wheel itself, like hardware going bad. If you want to check this possibility, you might want to try connecting the wheel to a Windows machine, real or virtual.

creeper82, 2024-12-24:
I think we can close this for now. Miraculously, when booting the system today, force feedback works. I didn't do a single thing. Seems like Linux decided to let me enjoy the christmas. If the issue returns somehow, I'll just create a new one.

closed 2024-12-24 by creeper82