Kimplul/hid-tmff2 #137: Still getting error on dkms after installing headers

jbirdforest, 2025-01-05:
Hello, on steamdeck.  I installed the linux headers and am getting the following message after installing the dkms package:

(deck@steamdeck hid-tmff2)$ git clone --recurse-submodules https://github.com/Kimplul/hid-tmff2.git
cd hid-tmff2
sudo ./dkms/dkms-install.sh
sudo make udev-rules # optional but should fix some common issues
Cloning into 'hid-tmff2'...
remote: Enumerating objects: 1229, done.
remote: Counting objects: 100% (290/290), done.
remote: Compressing objects: 100% (80/80), done.
remote: Total 1229 (delta 248), reused 213 (delta 210), pack-reused 939 (from 2)
Receiving objects: 100% (1229/1229), 2.14 MiB | 4.14 MiB/s, done.
Resolving deltas: 100% (729/729), done.
Submodule 'deps/hid-tminit' (https://github.com/Kimplul/hid-tminit.git) registered for path 'deps/hid-tminit'
Cloning into '/home/deck/hid-tmff2/hid-tmff2/hid-tmff2/hid-tmff2/deps/hid-tminit'...
remote: Enumerating objects: 155, done.        
remote: Counting objects: 100% (35/35), done.        
remote: Compressing objects: 100% (20/20), done.        
remote: Total 155 (delta 22), reused 17 (delta 15), pack-reused 120 (from 1)        
Receiving objects: 100% (155/155), 44.27 KiB | 4.92 MiB/s, done.
Resolving deltas: 100% (88/88), done.
Submodule path 'deps/hid-tminit': checked out '17d6cad0b508417193ebf4d23f67c845487e9932'
Error! DKMS tree already contains: hid-tmff2-0.82
You cannot add the same module/version combo more than once.
Sign command: /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/build/scripts/sign-file
Binary /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/build/scripts/sign-file not found, modules won't be signed
Error! Your kernel headers for kernel 6.5.0-valve22-1-neptune-65-g9a338ed8a75e cannot be found at /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/build or /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/source.
Please install the linux-headers-6.5.0-valve22-1-neptune-65-g9a338ed8a75e package or use the --kernelsourcedir option to tell DKMS where it's located.
Sign command: /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/build/scripts/sign-file
Binary /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/build/scripts/sign-file not found, modules won't be signed
Error! Your kernel headers for kernel 6.5.0-valve22-1-neptune-65-g9a338ed8a75e cannot be found at /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/build or /usr/lib/modules/6.5.0-valve22-1-neptune-65-g9a338ed8a75e/source.
Please install the linux-headers-6.5.0-valve22-1-neptune-65-g9a338ed8a75e package or use the --kernelsourcedir option to tell DKMS where it's located.
install -m 0644 udev/99-thrustmaster.rules /etc/udev/rules.d/

Kimplul, 2025-01-05:
Which headers did you install? Seems like `linux-neptune-65-headers` would be the appropriate package to install, but you might also want to look through #130, similar issue related to the Steam Deck.

jbirdforest, 2025-01-05:
I installed the 65 headers

jbirdforest, 2025-01-05:
I think it worked this time:

(deck@steamdeck ~)$ sudo pacman -S linux-headers
[sudo] password for deck: 
warning: linux-headers-6.0.2.arch1-1.1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) linux-headers-6.0.2.arch1-1.1

Total Installed Size:  145.19 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                                     [####################################] 100%
(1/1) checking package integrity                                   [####################################] 100%
(1/1) loading package files                                        [####################################] 100%
(1/1) checking for file conflicts                                  [####################################] 100%
(1/1) checking available disk space                                [####################################] 100%
:: Processing package changes...
(1/1) reinstalling linux-headers                                   [####################################] 100%
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Updating module dependencies...
(deck@steamdeck ~)$ sudo pacman -S linux-neptune-65-headers
warning: linux-neptune-65-headers-6.5.0.valve24-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) linux-neptune-65-headers-6.5.0.valve24-1

Total Installed Size:  116.97 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                                     [####################################] 100%
(1/1) checking package integrity                                   [####################################] 100%
(1/1) loading package files                                        [####################################] 100%
(1/1) checking for file conflicts                                  [####################################] 100%
(1/1) checking available disk space                                [####################################] 100%
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
:: Processing package changes...
(1/1) reinstalling linux-neptune-65-headers                        [####################################] 100%
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Updating module dependencies...
(deck@steamdeck ~)$ sudo pacman -S joyutils
warning: joyutils-1.8.1-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) joyutils-1.8.1-1

Total Installed Size:  0.17 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                                     [####################################] 100%
(1/1) checking package integrity                                   [####################################] 100%
(1/1) loading package files                                        [####################################] 100%
(1/1) checking for file conflicts                                  [####################################] 100%
(1/1) checking available disk space                                [####################################] 100%
:: Processing package changes...
(1/1) reinstalling joyutils                                        [####################################] 100%
:: Running post-transaction hooks...
(1/2) Reloading device manager configuration...
(2/2) Arming ConditionNeedsUpdate...
(deck@steamdeck ~)$ git clone --recurse-submodules https://github.com/Kimplul/hid-tmff2.git
fatal: destination path 'hid-tmff2' already exists and is not an empty directory.
(128)(deck@steamdeck ~)$ git clone --recurse-submodules https://github.com/Kimplul/hid-tmff2.git
cd hid-tmff2
sudo ./dkms/dkms-install.sh
sudo make udev-rules # optional but should fix some common issues
fatal: destination path 'hid-tmff2' already exists and is not an empty directory.
Error! DKMS tree already contains: hid-tmff2-0.82
You cannot add the same module/version combo more than once.
Sign command: /usr/lib/modules/6.5.0-valve23-1-neptune-65-g385b5e207ae2/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Building module:
Cleaning build area...
'make' KDIR=/usr/lib/modules/6.5.0-valve23-1-neptune-65-g385b5e207ae2/build......
Signing module /var/lib/dkms/hid-tmff2/0.82/build/deps/hid-tminit/hid-tminit-new.ko
Signing module /var/lib/dkms/hid-tmff2/0.82/build/deps/hid-tminit/usb-tminit-new.ko
Signing module /var/lib/dkms/hid-tmff2/0.82/build/hid-tmff-new.ko
Cleaning build area...

hid-tminit-new.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.5.0-valve23-1-neptune-65-g385b5e207ae2/updates/dkms/

usb-tminit-new.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.5.0-valve23-1-neptune-65-g385b5e207ae2/updates/dkms/

hid-tmff-new.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.5.0-valve23-1-neptune-65-g385b5e207ae2/updates/dkms/
depmod...
install -m 0644 udev/99-thrustmaster.rules /etc/udev/rules.d/

Cimmerian-Iter, 2025-01-10:
Does the module work now for you? When I tried last time with the correct header, I had an out of sync issue between headers and kernel

[  117.836492] BPF:      type_id=5861 bits_offset=0
[  117.836496] BPF:  
[  117.836498] BPF: Invalid name
[  117.836500] BPF: 
[  117.836504] failed to validate module [hid_tminit_new] BTF: -22

I will get back to the T300 (I was hoping a new steam update comes to restore rootfs back to before i was doing shenigans and start clean )

jbirdforest, 2025-01-10:
Yes, it worked.  For Steam deck I installed the 65 headers. Then installed the joyutils then dkms install.  You can see my sudo commands in the post above.

jbirdforest, 2025-01-10:
Also after installing the joyutils for the next two commands I believe I cut and pasted the entire box including the gitclone commands, (just cut and paste from the install instructions).  Hope this helps!

Cimmerian-Iter, 2025-01-10:
I will clear any head I have, and redo from start, maybe I can fix the out of sync issue. If it doesn't, I'll restore rootfs. Thanks for the tip

Cimmerian-Iter, 2025-01-10:
it worked thanks, i got the 65 neptune headers manually from steam repo but yeah getting them directly from command line fixed the out of sync issue and now the module works correctly

jbirdforest, 2025-01-10:
Awesome!  I use a t248x on Steam deck and it works great with Assetto Corsa, Dirt Rally 2.0 and WRC Generations.  Let me know if you find any other games that work for it on Steam!