Kimplul/hid-tmff2 #138: Getting the wheel to work in steam

Cimmerian-Iter, 2025-01-10:
Hello, Now that the first step is done, I'll try getting the wheel to work correctly on my deck

[  366.496818] usb 1-1.3: USB disconnect, device number 5
[  369.266438] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[  369.372694] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  369.372706] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  369.372708] usb 1-1.3: Product: Thrustmaster FFB Wheel
[  369.372710] usb 1-1.3: Manufacturer: Thrustmaster
[  369.457628] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.3/1-1.3:1.0/0003:044F:B65D.000E/input/input44
[  369.457865] hid-thrustmaster 0003:044F:B65D.000E: input,hidraw2: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:04:00.3-1.3/input0
[  369.477704] usbhid 1-1.3:1.0: Wheel with (model, attachment) = (0x2, 0x3) is a Thrustmaster T300RS (F1 attachment). attachment_found=1
[  369.479783] usbhid 1-1.3:1.0: Success, the wheel should have been initialized!
[  369.568937] usb 1-1.3: USB disconnect, device number 6
[  370.286576] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
[  370.412700] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[  370.412712] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  370.412718] usb 1-1.3: Product: Thrustmaster T300RS Racing wheel
[  370.412723] usb 1-1.3: Manufacturer: Thrustmaster
[  370.502197] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.3/1-1.3:1.0/0003:044F:B66E.000F/input/input45
[  370.502423] tmff2 0003:044F:B66E.000F: input,hidraw2: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:04:00.3-1.3/input0
[  370.507712] tmff2 0003:044F:B66E.000F: force feedback for T300RS

T300 F1 addon is recognized, but in oversteer I have this message 
[image]

So I can't change much settings.

fftest to see if it works, rumble effect work, the others doesn't but i guess it has to do with this 
[image]

Installed windows driver in steam games but the wheel is simply not detected, not a single input. Using proton-ge, I checked other issue and it's either down to proton version and steam input.

/home/deck/.steam/steam/compatibilitytools.d/GE-Proton9-20/files/bin/wine control joy.cpl shows nothing

Finally tried speed dream. It detect input? but in game nothing moves, no steering no accelerator. 

So there's something that prevents access to the wheel correctly? I did the udev rules. But i'm kinda clueless about why i'm not even having correct linux support before tackling windows support.

Kimplul, 2025-01-10:
Sounds very similar to #127, unfortunately I don't have a solution for it at the moment. There was also #132 which I at least initially thought might've been related, but I'm not sure.

Kimplul, 2025-01-11:
> so it's a case of two driver for a single device and it doesn't pick tmff2 if i understand correctly?

Not quite, the initial question was how to force `hid-tmff2` to be used for the device, but as I pointed out a bit down the thread, this driver is being loaded properly but the input issue is in some other layer, but I don't quite know which or what the issue exactly is.

Please do report back if updating the deck fixes this issue for you, at least it'll be a bit more data to (hopefully) debug further.

Cimmerian-Iter, 2025-02-28:
so deck update came, cleaned my rootfs so i can test again.

Before doing the echo blacklist command

`[  653.293338] usb 1-1.1: USB disconnect, device number 8
[ 1005.246535] usb 1-1.1: new full-speed USB device number 9 using xhci_hcd
[ 1005.356213] usb 1-1.1: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 1005.356231] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1005.356238] usb 1-1.1: Product: Thrustmaster FFB Wheel
[ 1005.356244] usb 1-1.1: Manufacturer: Thrustmaster
[ 1005.458656] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.1/1-1.1:1.0/0003:044F:B65D.000A/input/input35
[ 1005.459223] hid-thrustmaster 0003:044F:B65D.000A: input,hidraw6: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:04:00.3-1.1/input0
[ 1005.478216] hid-thrustmaster 0003:044F:B65D.000A: Wheel with model id 0x206 is a Thrustmaster T300RS
[ 1005.480249] hid-thrustmaster 0003:044F:B65D.000A: Success?! The wheel should have been initialized!
[ 1005.491650] hid_tminit_new: loading out-of-tree module taints kernel.
[ 1005.491664] hid_tminit_new: module verification failed: signature and/or required key missing - tainting kernel
[ 1005.493597] Error: Driver 'hid-thrustmaster' is already registered, aborting...
[ 1005.549717] usb 1-1.1: USB disconnect, device number 9
[ 1006.266438] usb 1-1.1: new full-speed USB device number 10 using xhci_hcd
[ 1006.396238] usb 1-1.1: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[ 1006.396254] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1006.396261] usb 1-1.1: Product: Thrustmaster T300RS Racing wheel
[ 1006.396266] usb 1-1.1: Manufacturer: Thrustmaster
[ 1006.482916] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.1/1-1.1:1.0/0003:044F:B66E.000B/input/input36
[ 1006.483481] hid-generic 0003:044F:B66E.000B: input,hidraw6: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:04:00.3-1.1/input0
[ 1006.553927] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.1/1-1.1:1.0/0003:044F:B66E.000B/input/input37
[ 1006.556529] tmff2 0003:044F:B66E.000B: input,hidraw6: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:04:00.3-1.1/input0
[ 1006.561253] tmff2 0003:044F:B66E.000B: force feedback for T300RS`

and after

`[   81.130302] usb 1-1.1: USB disconnect, device number 3
[   83.385688] usb 1-1.1: new full-speed USB device number 6 using xhci_hcd
[   83.495954] usb 1-1.1: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[   83.495969] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   83.495977] usb 1-1.1: Product: Thrustmaster FFB Wheel
[   83.495983] usb 1-1.1: Manufacturer: Thrustmaster
[   83.585647] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.1/1-1.1:1.0/0003:044F:B65D.0009/input/input35
[   83.586014] hid-thrustmaster 0003:044F:B65D.0009: input,hidraw3: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:04:00.3-1.1/input0
[   83.606938] usbhid 1-1.1:1.0: Wheel with (model, attachment) = (0x2, 0x6) is a Thrustmaster T300RS. attachment_found=1
[   83.608981] usbhid 1-1.1:1.0: Success, the wheel should have been initialized!
[   83.690444] usb 1-1.1: USB disconnect, device number 6
[   84.416164] usb 1-1.1: new full-speed USB device number 7 using xhci_hcd
[   84.528947] usb 1-1.1: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[   84.528962] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   84.528968] usb 1-1.1: Product: Thrustmaster T300RS Racing wheel
[   84.528974] usb 1-1.1: Manufacturer: Thrustmaster
[   84.623691] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1-1.1/1-1.1:1.0/0003:044F:B66E.000A/input/input36
[   84.623996] tmff2 0003:044F:B66E.000A: input,hidraw3: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:04:00.3-1.1/input0
[   84.628937] tmff2 0003:044F:B66E.000A: force feedback for T300RS
(deck@steamdeck hid-tmff2)$ 
`

so indeed i don't have the error about hid=trhsutmaster already registered, so it should have unloaded the default driver, but still same issue in game, oversteer and that linux car demo.

What is weird is that i have the exact same problem with the lgff4 driver that was made for logitech G29. 

I will try to investigate the G29 and see if i can find something useful for the T300