Kimplul/hid-tmff2 #139: Richard Burns Rally

jbirdforest, 2025-01-14:
Hello I have been able to get the driver to work. I recently created a bottle to run Richard Burns Rally and the wheel and pedals work, however there is no ffb.  I was wondering if anyone has been able to make these Linux drivers work with RBR and if you have any tips?  I am running on Steam deck with arch Linux

jbirdforest, 2025-01-15:
I found a solution.  The following method (custom wine proton) works w/ bottles on Steam Deck and allows FFB to work w/ RBR.  https://gitlab.com/as400l/wine-rbr

closed 2025-01-15 by Kimplul

Kimplul, 2025-01-15:
Haven't played RBR in a long while, should maybe have a go at it at some point again, fun game. The author of the patch seems to be intending on getting it upstreamed, so I'd expect this to resolve itself without workarounds in due time, which is good. Glad you found the fork!