Kimplul/hid-tmff2 #140: Bad return status for module build on kernel: 6.12.8-2-default (x86_64)

dlmw, 2025-01-17:
I've uninstalled the module that I installed with DKMS. `sudo dkms status` doesn't return anything, indicating that the module has been uninstalled.

When I run `sudo ./dkms/dkms-install.sh` here's the ouput I get:
```
Creating symlink /var/lib/dkms/hid-tmff2/0.81/source -> /usr/src/hid-tmff2-0.81

Sign command: /lib/modules/6.12.8-2-default/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
'make' KDIR=/lib/modules/6.12.8-2-default/build

Error! Bad return status for module build on kernel: 6.12.8-2-default (x86_64)
Consult /var/lib/dkms/hid-tmff2/0.81/build/make.log for more information.

Sign command: /lib/modules/6.12.8-2-default/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
'make' KDIR=/lib/modules/6.12.8-2-default/build

Error! Bad return status for module build on kernel: 6.12.8-2-default (x86_64)
Consult /var/lib/dkms/hid-tmff2/0.81/build/make.log for more information.
```

Here's the content of `/var/lib/dkms/hid-tmff2/0.81/build/make.log`:
```
DKMS make.log for hid-tmff2/0.81 for kernel 6.12.8-2-default (x86_64)
Fri Jan 17 10:44:18 AM CET 2025
make -C deps/hid-tminit KDIR="/lib/modules/6.12.8-2-default/build" 
make[1]: Entering directory '/var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit'
make -C /lib/modules/6.12.8-2-default/build M=/var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit modules
make[2]: Entering directory '/usr/src/linux-6.12.8-2-obj/x86_64/default'
  CC [M]  /var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit/hid-tminit.o
  MODPOST /var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit/Module.symvers
  CC [M]  /var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit/hid-tminit.mod.o
  CC [M]  /var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit/.module-common.o
  LD [M]  /var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit/hid-tminit.ko
  BTF [M] /var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit/hid-tminit.ko
Skipping BTF generation for /var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit/hid-tminit.ko due to unavailability of vmlinux
make[2]: Leaving directory '/usr/src/linux-6.12.8-2-obj/x86_64/default'
make[1]: Leaving directory '/var/lib/dkms/hid-tmff2/0.81/build/deps/hid-tminit'
make -C /lib/modules/6.12.8-2-default/build M=/var/lib/dkms/hid-tmff2/0.81/build modules
make[1]: Entering directory '/usr/src/linux-6.12.8-2-obj/x86_64/default'
  CC [M]  /var/lib/dkms/hid-tmff2/0.81/build/src/hid-tmff2.o
/var/lib/dkms/hid-tmff2/0.81/build/src/hid-tmff2.c:743:25: error: initialization of ‘const __u8 * (*)(struct hid_device *, __u8 *, unsigned int *)’ {aka ‘const unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)’} from incompatible pointer type ‘__u8 * (*)(struct hid_device *, __u8 *, unsigned int *)’ {aka ‘unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)’} [-Wincompatible-pointer-types]
  743 |         .report_fixup = tmff2_report_fixup,
      |                         ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/hid-tmff2/0.81/build/src/hid-tmff2.c:743:25: note: (near initialization for ‘tmff2_driver.report_fixup’)
make[3]: *** [/usr/src/linux-6.12.8-2/scripts/Makefile.build:230: /var/lib/dkms/hid-tmff2/0.81/build/src/hid-tmff2.o] Error 1
make[2]: *** [/usr/src/linux-6.12.8-2/Makefile:1960: /var/lib/dkms/hid-tmff2/0.81/build] Error 2
make[1]: *** [../../../linux-6.12.8-2/Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-6.12.8-2-obj/x86_64/default'
make: *** [Makefile:4: all] Error 2
```

Running `sudo dkms status` again I get: `hid-tmff2/0.81: added` as the output.

Kimplul, 2025-01-17:
There was recently a PR about this: #128 

Are you sure you have the most recent version of this driver? There should be a version check in `src/hid-tmff2.c` about 6.12.0 that fixes this.

Kimplul, 2025-01-17:
`build.log` looks clean, so not entirely sure what `dkms` is upset about. Did you remove all previous versions in `dkms`? Wondering if it's maybe conflating multiple runs or something.

dlmw, 2025-01-17:
Now when I run `sudo dkms status` I get `hid-tmff2/0.82: added`.

What would be the easiest way to ensure everything is uninstalled and not creating conflicts?

Kimplul, 2025-01-17:
```
sudo dkms remove hid-tmff2/0.81 --all
sudo dkms remove hid-tmff2/0.82 --all
```

Should uninstall/remove etc and get a clean start.

MmAaXx500, 2025-01-17:
I think there was a submodule update since then, `git submodule update` updates the submodule, and it should then build properly.

dlmw, 2025-01-17:
That was it!!! `git submodule update`, ran the same command as before and it instantly worked.

That's perfect, thanks.

I know how Git works but never worked with submodules so I wonder if it would be wise to add it in the README?

Kimplul, 2025-01-17:
Granted, the instructions are kind of written with the assumption that the user installs the driver once and then doesn't have to worry about updating, and the recent API change kind of breaks this assumption. I'll think of how I want to formulate the instructions.

Glad you caught what the issue was @MmAaXx500, just out of curiosity have you run into anything similar before? I wouldn't have thought the submodule would have been the culprit based on the logs provided.

For what it's worth, this issue would've been avoided had I upstreamed this driver into the kernel, maybe I should also have a think about it. Had some plans at some point but found bugs and got scared, recently all issues have been with external stuff so maybe the code is 'robust enough' to consider upstreaming.

Anyway, great that you got things working, I'll close the issue but feel free to open up new ones if/when you encounter any further problems.

closed 2025-01-17 by Kimplul

MmAaXx500, 2025-01-17:
> Glad you caught what the issue was [@MmAaXx500](https://github.com/MmAaXx500), just out of curiosity have you run into anything similar before? I wouldn't have thought the submodule would have been the culprit based on the logs provided.

Probably yes, but the thought process here was that the log mentioned to check the module names --> with the `0.81` to `0.82` change the `usb-tminit` module was introduced --> it's easy to forget a submodule update and there is no `usb-tminit` mentioned in the make.log --> it must be the old submodule, because I built it to test it and there was no issue.