Kimplul/hid-tmff2 #142: hid-tmff2/0.82 is not installed but DKMS install fails

Tropingenie, 2025-01-22:
So my FFB stopped working properly on my Thrustmaster T300 recently, and I suspect it was due to Nobara updating from Fedora 40 to 41. Came across #51 and tried to reinstall through dkms, didn't work. Now, that would be fine, except the errors are both that I do not have hid-tmff2/0.82 installed, and that I do (see below).

Let me know what else is needed to help, new-ish to Linux so I can never remember what commands to try off the top of my head.

```
░▒▓  …/hid-tmff2    master !  ▓▒░ ◦•······························································································································•◦ ░▒▓   83ms ▓▒░
··• sudo dkms uninstall -m "hid-tmff2" -v "0.82"
Module hid-tmff2/0.82 is not installed for kernel 6.12.10-200.nobara.fc41.x86_64 (x86_64). Skipping...

░▒▓  …/hid-tmff2    master !  ▓▒░ ◦•·····························································································································•◦ ░▒▓   483ms ▓▒░
··• ./dkms/dkms-install.sh 
You must run this script as root!

░▒▓  …/hid-tmff2    master !  ▓▒░ ◦•·······························································································································•◦ ░▒▓   5ms ▓▒░
··• sudo ./dkms/dkms-install.sh 

Error! DKMS tree already contains: hid-tmff2/0.82
You cannot add the same module/version combo more than once.

Sign command: /lib/modules/6.12.10-200.nobara.fc41.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s).... done.

Error! Build of hid-tminit-new.ko failed for: 6.12.10-200.nobara.fc41.x86_64 (x86_64)
Make sure the name of the generated module is correct and at the root of the
build directory, or consult make.log in the build directory
/var/lib/dkms/hid-tmff2/0.82/build for more information.

Sign command: /lib/modules/6.12.10-200.nobara.fc41.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s).... done.

Error! Build of hid-tminit-new.ko failed for: 6.12.10-200.nobara.fc41.x86_64 (x86_64)
Make sure the name of the generated module is correct and at the root of the
build directory, or consult make.log in the build directory
/var/lib/dkms/hid-tmff2/0.82/build for more information.
```

Kimplul, 2025-01-22:
Have a look at #140, I believe you're running into the same issue.

The DKMS errors are a bit confusing, but it's probably just that DKMS keeps track of three values, the name of the module, version of the module, and version of the kernel. Presumably the driver failed to build for your newest kernel due to an API change in 6.12, so you don't have `hid-tmff2/0.82/6.12-...` 'installed', but you do have `hid-tmff2/0.82`/6.11-...` (or whatever the previous name was) and the error messages are just reporting this kind of odd situation poorly.

Tropingenie, 2025-02-03:
Confirmed working on Tuesday, FFB is working in Euro Truck (which I know was working prior to the kernel update that confused dkms).

closed 2025-02-03 by Tropingenie

Beanow, 2025-04-13:
> Ran git submodule update to pull in hid-tminit

Thanks from the future, that was my issue too!