Kimplul/hid-tmff2 #144: hid-tmff2/0.82 system freeze

pkmed, 2025-01-26:
Hi, my system freezes after a random amount of time during races in Dirt rally 1 & 2 and RBR RSF.
Proton/wine version seems has no impact on the issue.
Steering wheel and gamepad are connected simultaneously.
When the freeze occurs here's what happens:
- video stops completely
- audio starts looping the last 0.5 seconds
- FFB stays present in the same direction it was before the freeze
- input from keyboard and mouse doesn't do anything

Unplugging the wheel doesn't do anything. 
When my gamepad isn't connected it looks like i have a little more time before the freeze.

My system:
OS: Nobara Linux 41 (KDE Plasma) x86_64
Kernel: Linux 6.12.9-202.nobara.fc41.x86_64
DE: KDE Plasma 6.2.5
WM: KWin (Wayland)
CPU: AMD Ryzen 5 2600 (12) @ 3.40 GHz
GPU: NVIDIA GeForce RTX 4060
Wheel: Thrustmaster t300 rs gt
Gamepad: Logitech Dual Action

[after_core_dump.log](https://github.com/user-attachments/files/18551164/after_core_dump.log)

Kimplul, 2025-01-26:
That's not good, thanks for the report. I guess it's a threading bug, `spin_lock` and `spin_lock_irqsave` in `src/hid-tmff2.c` would probably need to be `spin_lock_irqsave` and `spin_lock_irqrestore`, respectively. The kernel docs mention that `spin_lock` can be slightly faster, which is why I went with it, but it can only be used when not in interrupt context, which I didn't think `start/upload/etc` were. I guess I must've misunderstood something (or Nobara uses an RT kernel? Apparently the RT patches slightly change locking semantics, not sure).

I pushed a commit to the `spinlock` branch, could you try it out?

pkmed, 2025-01-26:
@Kimplul everything looks good so far, no freezes after around an hour. Gonna test it for couple more days to be sure that everything is ok.

Kimplul, 2025-01-26:
Sounds good. I'll optimistically merge into `master`, at least the patch doesn't seem to be causing any issues for me and it can always be reverted if needed. Might also be good to expose more people to this fix earlier rather than later.

MaxiFtt, 2025-03-18:
Hi, I ran into the same issue with same symptoms using 
- Debian GNU/Linux 12 (bookworm)
- Linux 6.1.0-28-amd64

Then I switched to the `spinlock` branch and everything worked fine. 
I just wanted to comment my experience so @Kimplul you have more feedback. Thanks so much for the fix!

Kimplul, 2025-03-18:
Cheers @MaxiFtt, although `master` should have the fix applied so I don't really understand why it's showing symptoms but `spinlock` doesn't. Pretty sure Debian doesn't have RT patches enabled so at lest we can rule that theory out, I suppose.

Did you install the driver a while ago or was this a new install?

MaxiFtt, 2025-03-19:
The last commit I have in `master` is 5e877441d538b579f1412c8286005b45d83c3ee4 from December 28, then i switched to the `spinlock` branch with a3e70e77d95c12e3fe11e0a7ccb67e7195e0e1ab from January 26.

Kimplul, 2025-03-19:
Makes sense, `git pull` in `master` and you should be golden. Still kind of surprised about Debian.

Kimplul, 2025-04-12:
Haven't heard any updates in a while, I'll assume this can be considered resolved.

closed 2025-04-12 by Kimplul