Kimplul/hid-tmff2 #145: Losing FFB in Assetto Corsa Evo when at limit (Thrustmaster T300RS)

Pengukov, 2025-01-28:
## Issue

When playing Assetto Corsa Evo (early access v0.13) when reaching maximum force the FFB just completely drops to zero (-1 in logs).

For example driving around a corner, when at about the grip limit/getting understeer the wheel instantly loses all FFB for a short period of time (ranging from very short to about a second or so). 
Sometimes there was a "knocking" effect where FFB would drop and activate 2-4x a second while driving around a corner.

To avoid reaching maximum FFB it is possible to reduce the ingame FFB strength to about 60-65. This does not work by reducing the FFB gain in Oversteer.

Other games (Assetto Cora/Automobilista 2/native BeamNG.drive) do not have this issue, nor does it occur on Windows.

### Logs

The ffbwrap logs show that the force drops to -1 instead of reaching at 32767 (no occurrence of 32767): 
[FFB_ACE.log-20250126122545_bug.log](https://github.com/user-attachments/files/18577713/FFB_ACE.log-20250126122545_bug.log)
```
000094499695 > UPLOAD id:0 dir:16429 length:0 delay:0 type:CONSTANT level:-32614 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000094499701 < 0 id:0
000094502707 > UPLOAD id:0 dir:16429 length:0 delay:0 type:CONSTANT level:-31382 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000094502715 < 0 id:0
000094505716 > UPLOAD id:0 dir:16429 length:0 delay:0 type:CONSTANT level:-1 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000094505722 < 0 id:0
000094508696 > UPLOAD id:0 dir:16429 length:0 delay:0 type:CONSTANT level:-1 attack_length:0 attack_level:0 fade_length:0 fade_level:0
```

While the log from Assetto Corsa (via Content Manager) does not show this behavior and reaches 32767: 
[FFB_AC.log-20250126144719_normal.log](https://github.com/user-attachments/files/18577714/FFB_AC.log-20250126144719_normal.log)
```
000035919049 > UPLOAD id:0 dir:16429 length:0 delay:0 type:CONSTANT level:31593 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000035919056 < 0 id:0
000035922066 > UPLOAD id:0 dir:16429 length:0 delay:0 type:CONSTANT level:32767 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000035922076 < 0 id:0
000035925110 > UPLOAD id:0 dir:16429 length:0 delay:0 type:CONSTANT level:32586 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000035925120 < 0 id:0
000035928067 > UPLOAD id:0 dir:16429 length:0 delay:0 type:CONSTANT level:32767 attack_length:0 attack_level:0 fade_length:0 fade_level:0
```

### How to reproduce

FFB gain 100% in Oversteer and ACE (with Thrustmaster T300RS). 
Driving with Mazda MX5 at second gear redlining and steering either fast in a direction or just taking a corner at about the steering limit.

### Tried

Changing Proton versions 
Installing Thrustmaster driver into prefix
ffbwrap --offset-fix --update-fix

Kimplul, 2025-01-28:
Hi, thanks for the report. `ffbwrap` sits between the game and the driver (roughly speaking), and since `ffbwrap` is showing a value of `-1`, I'm fairly confident there's a bug in Proton or the game itself, and I would recommend posting this to https://github.com/ValveSoftware/Proton/issues/8395.

Pengukov, 2025-01-28:
I kinda thought so, will post there thanks.

JacKeTUs, 2025-01-28:
That's the game + Wine issue, so feel free to close this
https://gitlab.winehq.org/wine/wine/-/merge_requests/7161

closed 2025-01-28 by Kimplul