Kimplul/hid-tmff2 #146: T248X not fully recognized

sizqui001, 2025-02-24:
Hello,
first of all thanks for this project! I am trying to use the XBOX version of T248 and it seems I am missing something. To be fair my troubleshooting skills are rather limited... 
Issue: wheel seems to be recognized by the system (Arch-based, Linux 6.12.16-1-lts x86_64), but will not be detected by sdl2-jstest or oversteer
I would appreciate if you can share some pointers!

What I have been able to investigate:

- DKMS install looks good:
```
sudo dkms status
hid-tmff2/r280.a9312ea, 6.12.16-1-lts, x86_64: installed
hid-tmff2/r280.a9312ea, 6.13.4-arch1-1, x86_64: installed

```
- The wheel appears in dmesg: 
```
[  165.437736] usb 1-1.4: new full-speed USB device number 8 using ehci-pci
[  165.526716] usb 1-1.4: New USB device found, idVendor=044f, idProduct=b69a, bcdDevice= 1.01
[  165.526728] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  165.526733] usb 1-1.4: Product: Thrustmaster T248X GIP Racing Wheel
```
- I noticed the idProduct was not present in  `/lib/udev/rules.d/99-thrustmaster-wheel-perms.rules`, so I tried adding it:
```
# Thrustmaster T248 Racing Wheel (USB)
ATTRS{idProduct}=="b696", RUN+="/bin/sh -c 'cd %S%p; chmod 666 range gain spring_level damper_level friction_level'"
ATTRS{idProduct}=="b69a", RUN+="/bin/sh -c 'cd %S%p; chmod 666 range gain spring_level damper_level friction_level'"

```
- `lshw` recognizes the USB connection but it appears as _Generic USB device_. Not sure if this is meaningful
- Of course I tried rebooting, reconnecting with/without the PC running. 
- I thought about giving `tmdrv` a shot but that doesn't seem to be my problem

Thanks in advance!

Kimplul, 2025-02-24:
Hi, thanks for the report, appreciate it. Really detailed, clear and well written as well, always nice to see!

> lshw recognizes the USB connection but it appears as Generic USB device. Not sure if this is meaningful

That means that this driver is not being used for the wheel. Based on your logs, it seems that the Xbox version has a different USB PID than the PC version that I have, so there's currently nothing telling Linux that this driver should be picked for the wheel.

I created a new branch, `t248x`, to accept devices with the ID from the `dmesg` log, try installing that branch and report back. Probably best you uninstall the existing `dkms` version as well, to make sure the newer module is loaded. In general I would also recommend trying out untested modules in virtual machines or something as a precaution, but that might be a bit overkill for this situation. Still, be warned.

sizqui001, 2025-02-25:
OK, so after some more time trying to investigate **I noticed the wheel was in XBOX mode, instead of PC.** Rookie mistake, apologies for that! In bold, just in case anyone else suffers the same "issue".
After switching it got properly recognized:
```
[ 1657.140833] usb 1-1.2: new full-speed USB device number 9 using ehci-pci
[ 1657.230360] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b696, bcdDevice= 1.00
[ 1657.230373] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1657.230378] usb 1-1.2: Product: Thrustmaster Advance Racer
[ 1657.230381] usb 1-1.2: Manufacturer: Thrustmaster
[ 1657.232473] input: Thrustmaster Thrustmaster Advance Racer as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:044F:B696.000B/input/input33
[ 1657.232929] tmff2 0003:044F:B696.000B: input,hidraw7: USB HID v1.00 Joystick [Thrustmaster Thrustmaster Advance Racer] on usb-0000:00:1a.0-1.2/input0
[ 1657.260548] tmff2 0003:044F:B696.000B: force feedback for T248

```
Quick test in Assetto Corsa is looking good as well! Oversteer was a bit misleading on the pedal ranges though.
Anyway, thanks a lot for your support. Not only we were able to make it work, but I learnt along the way!

closed 2025-02-25 by sizqui001

Kimplul, 2025-02-25:
Well done sorting this out. I'll have to keep the PC mode in mind for the future, probably add some note to the docs.

Looking back at your initial message, I realized that your wheel is showing up as a raw USB device, whereas the changes I made in `t248x` were as if it was a HID device. Since they're different categories as far as the kernel is concerned, no wonder it wasn't being picked up.

Just to correct myself for any possible future readers, I referred to my wheel as the PC version, but that should've been PS. Apparently both the Xbox and the PS version have a PC mode, and at least for the Xbox version that mode seems to be required.