Kimplul/hid-tmff2 #150: Driver for t128 steering wheel no longer works on kernel 6.13.7

GGTNT, 2025-03-14:
Hey, I recently bought a new PC with fairly recent components that require me to be on a recent kernel. Unfortunately, when I tried to install the driver (like here https://github.com/Kimplul/hid-tmff2/issues/67) on the new kernel (here 6.13.7), it displays an error when I execute the `make `command: ```
make -C deps/hid-tminit KDIR="/lib/modules/6.13.7-061307-generic/build” 
make[1]: enter the “/home/theo/hid-tmff2/deps/hid-tminit” directory
make -C /lib/modules/6.13.7-061307-generic/build M=/home/theo/hid-tmff2/deps/hid-tminit modules
make[2]: enter the “/home/theo/hid-tmff2/deps/hid-tminit” directory
make[2]: *** /lib/modules/6.13.7-061307-generic/build: No files or folders of this name. Stop.
make[2]: Exit the “/home/theo/hid-tmff2/deps/hid-tminit” directory.
make[1]: *** [Makefile:5 : all] Error 2
make[1]: exit directory “/home/theo/hid-tmff2/deps/hid-tminit”.
make: *** [Makefile:16: deps/hid-tminit] Error 2
 
```

Thank you in advance for your help

Kimplul, 2025-03-14:
Which distro are you on? It kind of sounds like you might not have kernel headers installed, or possibly a mismatch between the running kernel and the installed headers.

GGTNT, 2025-03-14:
I use zorin os, but as the kernel is quite old, I had to install (with mainline) the latest kernel (6.13.7). I saw that there were some broken dependencies with the linux header, but since it worked anyway, I didn't ask any questions. Do you think that's where the error is coming from?

Kimplul, 2025-03-14:
Haven't used with ZorinOS, so I don't really know for sure but I'd wager a guess that yeah, broken header are probably the cause for the build to fail.

I don't really know what you mean by `with mainline`, did you compile the kernel yourself or are there multiple kernel versions packaged in Zorin and the newest one has broken headers or something else entirely?

GGTNT, 2025-03-14:
Oh, apologies for the confusion, Mainline is a program that makes it easy to download and compile a new kernel. (Like [here](https://ubuntuhandbook.org/index.php/2020/08/mainline-install-latest-kernel-ubuntu-linux-mint/) ). 

So I have several kernels installed but it boots on the most recent one when the system boots. It gave me an error saying that there were broken dependencies with the linux header, but since it was booting anyway, I didn't wonder. But maybe the problem comes from what I understood in your message.

GGTNT, 2025-03-14:
Okay, thank you for your advice. I will try to reinstall the kernel and check if all the dependencies are properly resolved and that there are no more errors. I am currently not at home, but I will update you on the progress of the issue this evening.

GGTNT, 2025-03-15:
Hey, I switched distros in the end because it was too complex with certain dependencies to have the latest kernel on Zorin OS. I was then advised to switch to Cachy OS, but now I’m encountering another error when I run:
`
git clone --recurse-submodules -b t128 https://github.com/Kimplul/hid-tmff2.git
cd hid-tmff2
make
`
`
make -C deps/hid-tminit KDIR="/lib/modules/6.13.7-2-cachyos/build"
make[1]: Entering directory '/home/theo/hid-tmff2/deps/hid-tminit'
make -C /lib/modules/6.13.7-2-cachyos/build M=/home/theo/hid-tmff2/deps/hid-tminit modules
make[2]: Entering directory '/usr/lib/modules/6.13.7-2-cachyos/build'
make[3]: Entering directory '/home/theo/hid-tmff2/deps/hid-tminit'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: clang version 19.1.7
  You are using:           gcc (GCC) 14.2.1 20250207
  CC [M]  hid-tminit.o
gcc: error: unrecognized option '-mretpoline-external-thunk' in command line
gcc: error: unrecognized option '-fsplit-lto-unit' in command line
gcc: error: unrecognized option '-fdebug-info-for-profiling' in command line
gcc: error: unrecognized option '-mllvm' in command line
gcc: error: unrecognized option '-mllvm' in command line
gcc: error: unrecognized option '-improved-fs-discriminator=true' in command line
gcc: error: unrecognized option '-fbasic-block-sections=labels' in command line
make[5]: *** [/usr/lib/modules/6.13.7-2-cachyos/build/scripts/Makefile.build:196: hid-tminit.o] Error 1
make[4]: *** [/usr/lib/modules/6.13.7-2-cachyos/build/Makefile:1997: .] Error 2
make[3]: *** [/usr/lib/modules/6.13.7-2-cachyos/build/Makefile:251: __sub-make] Error 2
make[3]: Leaving directory '/home/theo/hid-tmff2/deps/hid-tminit'
make[2]: *** [Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/usr/lib/modules/6.13.7-2-cachyos/build'
make[1]: *** [Makefile:5: all] Error 2
make[1]: Leaving directory '/home/theo/hid-tmff2/deps/hid-tminit'
make: *** [Makefile:16: deps/hid-tminit] Error 2
`

Kimplul, 2025-03-15:
Haven't used CachyOS either so not 100% sure but based on the error message,  the kernel was compiled with `clang`, but you don't seem to have it installed so the kernel's build system tries to use `gcc`. Modules need to be built with the same command line flags as the kernel, and apparently the CachyOS kernel uses some flags that `gcc` doesn't recognize, hence the error. Maybe start with installing `clang`, I believe the build system should pick it up automatically.

GGTNT, 2025-03-15:
I'm not sure I understood your message correctly, clang and gcc are already installed. Should I add arguments to the make command?

Kimplul, 2025-03-15:
Oh, interesting, guess I've misunderstood some part of the kernel build system. You can force `clang` to be used by `make CC=clang`, which might work. Compiling the kernel itself with `clang` is done with `make LLVM=1`, but when I tried it on this module it still used `gcc` so there's definitely a gap in my knowledge at this point.

GGTNT, 2025-03-15:
Oh, right, technologies change so quickly that it’s hard to stay up to date. Anyway, I tried the commands but I’m getting an error:

```
❯ make CC=clang
make -C deps/hid-tminit KDIR="/lib/modules/6.13.7-2-cachyos/build" 
make[1]: entering directory '/home/theo/hid-tmff2/deps/hid-tminit'
make -C /lib/modules/6.13.7-2-cachyos/build M=/home/theo/hid-tmff2/deps/hid-tminit modules
make[2]: entering directory '/usr/lib/modules/6.13.7-2-cachyos/build'
make[3]: entering directory '/home/theo/hid-tmff2/deps/hid-tminit'
  CC [M]  hid-tminit.o
ld: unrecognized emulation mode: llvm
Supported emulations: elf_x86_64 elf32_x86_64 elf_i386 elf_iamcu i386pep i386pe elf64bpf
make[5]: *** [/usr/lib/modules/6.13.7-2-cachyos/build/scripts/Makefile.build:196: hid-tminit.o] Error 1
make[5]: *** Deleting file 'hid-tminit.o'
make[4]: *** [/usr/lib/modules/6.13.7-2-cachyos/build/Makefile:1997: .] Error 2
make[3]: *** [/usr/lib/modules/6.13.7-2-cachyos/build/Makefile:251: __sub-make] Error 2
make[3]: leaving directory '/home/theo/hid-tmff2/deps/hid-tminit'
make[2]: *** [Makefile:251: __sub-make] Error 2
make[2]: leaving directory '/usr/lib/modules/6.13.7-2-cachyos/build'
make[1]: *** [Makefile:5: all] Error 2
make[1]: leaving directory '/home/theo/hid-tmff2/deps/hid-tminit'
make: *** [Makefile:16: deps/hid-tminit] Error 2
```

Then I tried with LLVM=1:

❯ make LLVM=1
```
make -C deps/hid-tminit KDIR="/lib/modules/6.13.7-2-cachyos/build" 
make[1]: entering directory '/home/theo/hid-tmff2/deps/hid-tminit'
make -C /lib/modules/6.13.7-2-cachyos/build M=/home/theo/hid-tmff2/deps/hid-tminit modules
make[2]: entering directory '/usr/lib/modules/6.13.7-2-cachyos/build'
make[3]: entering directory '/home/theo/hid-tmff2/deps/hid-tminit'
  CC [M]  hid-tminit.o
  MODPOST Module.symvers
  CC [M]  hid-tminit.mod.o
  CC [M]  .module-common.o
  LD [M]  hid-tminit.ko
  BTF [M] hid-tminit.ko
make[3]: leaving directory '/home/theo/hid-tmff2/deps/hid-tminit'
make[2]: leaving directory '/usr/lib/modules/6.13.7-2-cachyos/build'
make[1]: leaving directory '/home/theo/hid-tmff2/deps/hid-tminit'
make -C /lib/modules/6.13.7-2-cachyos/build M=/home/theo/hid-tmff2 modules
make[1]: entering directory '/usr/lib/modules/6.13.7-2-cachyos/build'
make[2]: entering directory '/home/theo/hid-tmff2'
  CC [M]  src/hid-tmff2.o
src/hid-tmff2.c:740:18: error: incompatible function pointer types initializing 'const __u8 *(*)(struct hid_device *, __u8 *, unsigned int *)' (aka 'const unsigned char *(*)(struct hid_device *, unsigned char *, unsigned int *)') with an expression of type '__u8 *(struct hid_device *, __u8 *, unsigned int *)' (aka 'unsigned char *(struct hid_device *, unsigned char *, unsigned int *)') [-Wincompatible-function-pointer-types]
  740 |         .report_fixup = tmff2_report_fixup,
      |                         ^~~~~~~~~~~~~~~~~~
1 error generated.
make[4]: *** [/usr/lib/modules/6.13.7-2-cachyos/build/scripts/Makefile.build:196: src/hid-tmff2.o] Error 1
make[3]: *** [/usr/lib/modules/6.13.7-2-cachyos/build/Makefile:1997: .] Error 2
make[2]: *** [/usr/lib/modules/6.13.7-2-cachyos/build/Makefile:251: __sub-make] Error 2
make[2]: leaving directory '/home/theo/hid-tmff2'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: leaving directory '/usr/lib/modules/6.13.7-2-cachyos/build'
make: *** [Makefile:4: all] Error 2

```

Kimplul, 2025-03-15:
Oh, you shouldn't need to use the `t128` branch anymore, the T128 is merged into `master`. There was an API change in kernel 6.12 and that error was fixed in https://github.com/Kimplul/hid-tmff2/pull/128.

Good to know `LLVM=1` at least does something.

GGTNT, 2025-03-15:
Oh, great, cloning the master directly and using make LLVM=1, It works :D 

Thanks a lot for your help

Kimplul, 2025-03-15:
No worries, glad we figured this out.

closed 2025-03-15 by Kimplul