Kimplul/hid-tmff2 #151: T-GT II - Enchantments

samsunix, 2025-03-17:
Hello again,

I would like to request little T-GT2 enchantment.
Following controls highlighted on [picture](https://imgur.com/1UWF2Zg) are not functional and not detected by oversteer.

They are not absolutely crucial but 'nice to have'. Let me know what you need and I will do my best to provide.

Kimplul, 2025-03-17:
Presumably the T-GT II has an `rdesc` that's different from the one(s) in https://github.com/Kimplul/hid-tmff2/blob/a9312ead5720922e1c06b541ea48b6db3f289d36/src/tmt300rs/hid-tmt300rs.c#L87, so first order of business would probably be to try and get hold of it. With the driver uninstalled (or disabled or whatever, as long as it doesn't get loaded) you should be able to query the raw `rdesc` from the wheel with
```
cat /sys/kernel/debug/hid/XXXX\:VID\:PID.YYYY/rdesc
```

I believe `XXXX = 044f` and `YYYY = b66e` but you might have to dig around a bit (there are other ways to get the `rdesc`, for example capturing packets with WireShark but I think the above method is probably the easiest).

With a 'proper' `rdesc`, we would then need to figure out when we're talking to a T-GT II versus a regular T300. I would expect that information to be available via `t300rs_get_attachment()`, you might want to temporarily add in some print statements [here](https://github.com/Kimplul/hid-tmff2/blob/a9312ead5720922e1c06b541ea48b6db3f289d36/src/tmt300rs/hid-tmt300rs.c#L1409) (again, there are other ways to get the information like sending raw USB packets from the command line but I think the 'hack' above is probably easiest).

samsunix, 2025-03-18:
I dunno how to do pull request but maybe better if you check these entries first before I try and compile driver again because I have no clue what the heck I am doing.
```
0x09, 0x40,  /* Usage (Knob 1 Rotation) */
0x15, 0x00,  /* Logical minimum (0) */
0x26, 0xff, 0x03,  /* Logical maximum (1023) */
0x75, 0x10,  /* Report size (16) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */

0x09, 0x41,  /* Usage (Knob 1 Button Press) */
0x15, 0x00,  /* Logical minimum (0) */
0x25, 0x01,  /* Logical maximum (1) */
0x75, 0x01,  /* Report size (1) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */

0x09, 0x42,  /* Usage (Knob 2 Rotation) */
0x15, 0x00,  /* Logical minimum (0) */
0x26, 0xff, 0x03,  /* Logical maximum (1023) */
0x75, 0x10,  /* Report size (16) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */

0x09, 0x43,  /* Usage (Knob 2 Button Press) */
0x15, 0x00,  /* Logical minimum (0) */
0x25, 0x01,  /* Logical maximum (1) */
0x75, 0x01,  /* Report size (1) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */

0x09, 0x44,  /* Usage (Knob 3 Rotation) */
0x15, 0x00,  /* Logical minimum (0) */
0x26, 0xff, 0x03,  /* Logical maximum (1023) */
0x75, 0x10,  /* Report size (16) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */

0x09, 0x45,  /* Usage (Knob 3 Button Press) */
0x15, 0x00,  /* Logical minimum (0) */
0x25, 0x01,  /* Logical maximum (1) */
0x75, 0x01,  /* Report size (1) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */

0x09, 0x46,  /* Usage (Knob 4 Rotation) */
0x15, 0x00,  /* Logical minimum (0) */
0x26, 0xff, 0x03,  /* Logical maximum (1023) */
0x75, 0x10,  /* Report size (16) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */

0x09, 0x47,  /* Usage (Knob 4 Button Press) */
0x15, 0x00,  /* Logical minimum (0) */
0x25, 0x01,  /* Logical maximum (1) */
0x75, 0x01,  /* Report size (1) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */

0x09, 0x50,  /* Usage (Joystick 1 Mouse) */
0x15, 0x00,  /* Logical minimum (0) */
0x25, 0x01,  /* Logical maximum (1) */
0x75, 0x01,  /* Report size (1) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */

0x09, 0x51,  /* Usage (Joystick 2) */
0x15, 0x00,  /* Logical minimum (0) */
0x25, 0x01,  /* Logical maximum (1) */
0x75, 0x01,  /* Report size (1) */
0x95, 0x01,  /* Report count (1) */
0x81, 0x02,  /* Input (Variable, Absolute) */
```

samsunix, 2025-03-26:
Any update / comments regarding?

samsunix, 2025-03-26:
'''
_pkgbase=hid-tmff2
pkgname="${_pkgbase}-dkms"
pkgver=0.8
pkgrel=1
pkgdesc="Linux kernel module for Thrustmaster T300RS (DKMS)"
arch=('x86_64')
url="https://github.com/Kimplul/${_pkgbase}"
license=('GPL-3.0')
depends=('dkms')
conflicts=("${_pkgbase}")
# install=${pkgname}.install
source=("git+${url}")
#md5sums=(use 'updpkgsums')
md5sums=(SKIP)

prepare() {
  cd ${_pkgbase}
  git submodule update --init --recursive

  # Apply the patch
  patch -p1 << 'EOF'
diff --git a/src/tmt300rs/hid-tmt300rs.c b/src/tmt300rs/hid-tmt300rs.c
index f83e3af..153d8be 100644
--- a/src/tmt300rs/hid-tmt300rs.c
+++ b/src/tmt300rs/hid-tmt300rs.c
@@ -1407,6 +1407,10 @@ static int t300rs_get_attachment(struct t300rs_device_entry *t300rs)
                goto out;
        }

+       hid_info(t300rs->hdev, "attachment: %x model: %x\n",
+                       response->a.attachment,
+                       response->a.model);
+
        if (response->type == cpu_to_le16(0x49)) {
                attachment = response->a.attachment;
        } else if (response->type == cpu_to_le16(0x47)) {
EOF
}

package() {
#  install -Dm644 ${_pkgbase}/dkms.conf "${pkgdir}"/usr/src/${_pkgbase}-${pkgver}/dkms.conf
install -Dm644 ${_pkgbase}/dkms/dkms.conf "${pkgdir}"/usr/src/${_pkgbase}-${pkgver}/dkms.conf
  # Set name and version
  sed -e "s/@_PKGBASE@/${_pkgbase}/" \
      -e "s/@PKGVER@/${pkgver}/" \
      -i "${pkgdir}"/usr/src/${_pkgbase}-${pkgver}/dkms.conf

  # Copy sources (including Makefile)
  cp -r ${_pkgbase}/* "${pkgdir}"/usr/src/${_pkgbase}-${pkgver}/
}
'''

No change. nobs and buttons don't respond.

samsunix, 2025-03-26:
Oops, seem I missed one step. let me try again

Kimplul, 2025-03-26:
> No change. nobs and buttons don't respond.

The patch I mentioned isn't mean to fix anything quite yet, it's just to print out to `dmesg` some values that the wheel returns for a specific USB request. I'm hoping that we can detect which wheel we're talking to based on these values, so we can select the correct `rdesc` and *that way* get the buttons working.

So for now, please check `dmesg` and try and find the `attachment`/`model` values.

samsunix, 2025-03-26:
```
~/hid-tmff2-dkms master*
❯ sudo dmesg | grep "attachment"

[sudo] smx-käyttäjän salasana: 
[    8.984894] tmff2 0003:044F:B66D.0002: attachment: 8 model: 5
```

Kimplul, 2025-03-26:
> [    8.984894] tmff2 0003:044F:B66D.0002: attachment: 8 model: 5

Seems promising, an attachment ID of 8 is different from the two other known IDs, so we should be able to check against this. I'll look into modifying `t300rs_wheel_fixup` (or see if there's a better way to set the `rdesc`). There also also a few other wheels that use the T248 product ID, and whatever solution we end up with might benefit those wheels as well, which is nice.

> [sudo] smx-käyttäjän salasana: 

Suomi-jumpscare :)

samsunix, 2025-03-26:
Do you still need something else or can I 'safely' return my AUR package and wait update? :)

Kimplul, 2025-05-11:
Alright, hello again. Sorry, took a lot longer to get to this than I was anticipating. I pushed an initial commit to the `tgtII` branch that has the modifications to the `rdesc`, which should hopefully allow you to test if all the buttons show up properly.

It's not quite a finished thing, to start with I just hard-coded the `rdesc` so the driver thinks all devices with the T248 ID are T-GT II's, if the `rdesc` end up working I'll start working on the more generic 'dispatch' mechanism I was talking about earlier, but for now I just want to know if all the buttons work.

samsunix, 2025-05-11:
Hey,

I was forced to move back to windows temporarily but I'll test it with "live-cd" asap and get back to you.

Appreciated 👍🏻 Kiitos.