Kimplul/hid-tmff2 #152: FFB -Force Feedback not present/working on latest Batocera v42 Butterfly

casefresh, 2025-03-28:
Hello -
Here’s my results using this wheel on the larsen. acosta v42 Butterfly. You can find the conversation on the Discord Batocera - steering wheels server / sub server 

I tested the Thrustmaster T248 on Batocera Butterfly v42 beta (latest version as of yesterday) to confirm how force feedback (FFB) behaves. I played both random non-FFB steering wheel games and FFB-supported steering wheel games (Gran Turismo 3 & 4 on PS2, F-Zero GX on GameCube) to compare results.

General Wheel Functionality
	•	Drivers, I believe, are working correctly, but please confirm. (See my screenshot snippet from my earlier post for reference.)
	•	Menu navigation, buttons, and general wheel functions work fine.
	•	On a basic level, the wheel responds correctly to steering, acceleration, braking, and button inputs. No noticeable delay or control issues affecting gameplay.

Random Non-FFB Steering Wheel Games
	•	Most of the non-FFB games I’m playing are arcade-style, not simulation-style.
	•	Auto-rotation does NOT work—I had to manually set the wheel to 270° for proper control.
	•	FFB settings (1, 2, and 3) changed directly on the wheel are confirmed working:
	•	FFB 1 = Light steering feel
	•	FFB 2 = Moderate stiffness
	•	FFB 3 = Very stiff, more of a sportier feel
	•	These stiffness changes are happening within the wheel itself—the game isn’t controlling anything.
	•	There is no reaction to terrain, crashes, or game physics—which makes sense since these steering wheel games don’t support FFB.

F-Zero GX (GameCube) & Gran Turismo 3 & 4 (PS2) – FFB-Supported Steering Wheel Games
	•	Auto-rotation works properly—the game adjusts it automatically.
	•	The wheel is recognized, but there’s NO actual force feedback effects.
	•	FFB settings (1, 2, and 3) on the wheel don’t seem to change anything now that I can determine or feel.
	•	Basically, no change for me if I go between FFB 1, 2, or 3.
	•	I can select them on the wheel, but nothing feels any different per se.
	•	The game is overriding something, but I don’t think it’s actually doing any force feedback changes.
	•	I believe it’s taking control of the wheel, but I don’t know if switching between FFB 1, 2, and 3 on the wheel actually does anything now.
	•	Maybe on a really hard crash there’s a difference, but I don’t think so.
	•	What is clear is that the game is forcing some kind of override—feedback is coming back to the wheel because it’s not letting the wheel do its own thing like it did in non-FFB steering wheel games.
	•	That’s the smoking gun, in my opinion—because in non-FFB games, FFB 1, 2, and 3 clearly stiffened the wheel differently, but in GX and Gran Turismo, that effect disappears entirely. The wheel was adjusting stiffness on its own before, but in these games, that behavior is completely gone. The game is taking over something, but I don’t believe it’s actually applying any real force feedback in return.

Kimplul, 2025-03-28:
Hi, thanks for the report. I'm guessing you're using Dolphin for GameCube emulation and PCSX2 for PS2 emulation? I don't have any experience with them personally, and a quick internet search brings up other people with wheel issues so I don't really even know how well FFB *should* work, unfortunately.

> Auto-rotation does NOT work—I had to manually set the wheel to 270° for proper control.

What does auto-rotation mean in this context?

> FFB settings (1, 2, and 3) changed directly on the wheel are confirmed working:

I'm guessing this is just autocentering, i.e. there's a constant force pulling the wheel to the middle, but please confirm. Unless told otherwise by an application, the wheel defaults to autocentering, so I suppose that's what's going on here as well. The emulator just fetches button inputs and leaves the FFB alone for the games that don't use FFB. Not really sure why the other games seem to try and take control of the FFB but not really doing anything with it, but I have a few checks that may or may not narrow things down:

+ The T248 has a few different modes that you can select in the firmware, there's been reports that only the PC mode currently works, see #146. Check that you're not in PS or XBox mode or something.
+ Check if FFB on your wheels works at all. Try something like `fftest /dev/input/by-id/usb-Thrustmaster_SOMETHING_-event-joystick` and run the sinusoidal effect, for example.

If need be, the emulators are open source and we could look at the code and see if they're maybe doing something the driver isn't expecting, but let's get the easy stuff out of the way first.

Also, please upload the images you're referring to somewhere other than Discord. You should be able to copy-paste them into GitHub comments, although if they're screenshots of terminal output, preferably just use a code block.

casefresh, 2025-03-29:
> Hi, thanks for the report. I'm guessing you're using Dolphin for GameCube emulation and PCSX2 for PS2 emulation? I don't have any experience with them personally, and a quick internet search brings [...]

Hello! Thanks for the reply.  it's pcsx2 and dolphin on BATOCERA v42 Buttefly (their beta before stable release) i'm using. tovarichtch on discord (one of the devs for batocera) that focuses on the controls to make it seamless across platform
emulators and run multiple peripherals like this wheel.

I was mistaken that i thought you were a dev in Batocera . i guess you developed this and the batocera devs inserted the module? But something isn't working quite right with the FFB option on this wheel within the environment i'm using it in (batocera, dolphin, pcsx2).  these 2 games above mentions do have the ability on batocera for ffb.

The ffb 1 2 3 settings are the 3 settings you can choose to feel different feedback types, with 3 being the most prominent.

this wheel has a small display on it.

so definitely batocera are keen tooled this correctly.  it's a popular wheel that works well and has great ffb.

tovarichtch is on discord batocera server. check it out, it's a great system to play games with. it has done a great job with lightguns and they are implementing The linux lindbergh emulator and it's working fantastic.

ps. i'll try and figure out how to archer some terminal screenshots and vids of the sinus thing

casefresh, 2025-03-29:
am i able to post a onedrive links here? i'm just in the midst of figuring out how to upload you some stuff to this thread.

Kimplul, 2025-03-29:
> I was mistaken that i thought you were a dev in Batocera . i guess you developed this and the batocera devs inserted the module?

Seems like it. I have no affiliation with Batocera, I just wrote this driver and people seem to be using it. That's about it :)

> this wheel has a small display on it.

Yep, I have a T248 myself I can test with. The display isn't really fully functional at the moment, I haven't implemented the interface that games use to talk to the display.

> am i able to post a onedrive links here? i'm just in the midst of figuring out how to upload you some stuff to this thread.

Anything that doesn't require logging in should be fine. I'm pretty sure you can just copy-paste/drag-and-drop image/video files into the GitHub comment box, might be the easiest way.

Did you check the mode of the wheel?

casefresh, 2025-03-30:
sounds good, i'll try on my desktop and not my iphone later. If you have time, I recommend checking Batocera out and setting it up.

casefresh, 2025-03-30:
yep I have the wheel in PC mode which is required for how Batocera has implemented or how the module works.  it's all good the wheel except for the FFB. works great on all the other fronts for plug and play into Batocera

you should check out the sega lindbergh implementation they have done and how well they have that working with controls in Batocera. you'll need to run the butterfly version though (their beta).