Kimplul/hid-tmff2 #153: Troubles installing drivers on Bazzite

Markosha2004, 2025-04-03:
I tried both dkms and manual install, but it throws errors such as "cannot create file: read-only file system". Discord community of that distro is no help.

Kimplul, 2025-04-03:
There's an open issue about this over in the bazzite repo: https://github.com/ublue-os/bazzite/issues/901#issuecomment-2054077901

TL;DR: It's technically possible to compile and load external modules in bazzite, but the distro is built such that the average user really shouldn't do so. The best solution would probably be for the bazzite devs to bake these external modules into the system, but it seems that hasn't yet happened. Not sure why, but probably the usual lack of time/resources/motivation.

Any particular reason why you chose bazzite specifically? For now it might be easier to just use some other gaming-focused distro, I saw some mentions of nobara. No personal experience with it though, I just use regular Debian.

Markosha2004, 2025-04-03:
> There's an open issue about this over in the bazzite repo: [ublue-os/bazzite#901 (comment)](https://github.com/ublue-os/bazzite/issues/901#issuecomment-2054077901)
> 
> TL;DR: It's technically possible to compile and load external modules in bazzite, but the distro is built such that the average user really shouldn't do so. The best solution would probably be for the bazzite devs to bake these external modules into the system, but it seems that hasn't yet happened. Not sure why, but probably the usual lack of time/resources/motivation.
> 
> Any particular reason why you chose bazzite specifically? For now it might be easier to just use some other gaming-focused distro, I saw some mentions of nobara. No personal experience with it though, I just use regular Debian.

I just wanted to try it out. I usually use Mint, and it so much easier with drivers and shit. I'll try Nobara next

Markosha2004, 2025-04-16:
I'm closing this issue since it was established that bazzite is immutable system and drivers can't be installed

closed 2025-04-16 by Markosha2004