Kimplul/hid-tmff2 #155: A few issues

Oweblow, 2025-04-15:
I tried this driver on my Linux Mint installation so there were fewer games I had to boot into Windows for using my T128 with.
For the most part, it's pretty good, and I'm thankful that you've took your time to code this.

However, there are 2 bugs I've managed to pick up on in my testing.

For instance, in Forza Horizon 4: when I switch from the keyboard to the wheel, my car automatically starts accelerating and doesn't stop.
The game acts like my gas pedal doesn't exist (Oversteer can see it) but the break pedal works fine.
It's not a mapping issue as it's the same save file imported from my Windows installation.

Secondly, in BeamNG.drive: it runs amazingly for the most part, but the wheel feels strangely loose, as if it's insensitive to the surrounding environment. Could just be me though.

That's all so far.
Again, thanks for coding this driver to begin with, Windows sucks so it's nice for more of my stuff to be supported within Linux.
It's slowly getting rid of the need to dual-boot.

Oweblow, 2025-04-15:
System info if a ya need it for some reason:
[image]

Kimplul, 2025-04-15:
> It's not a mapping issue as it's the same save file imported from my Windows installation.

I'm not sure this proves much, unfortunately. Pedal axis mapping turns out to be surprisingly obtuse, for example: https://github.com/Kimplul/hid-tmff2/issues/31#issuecomment-985543679

Anyway, did you already try remapping acceleration? I think this is the first time anyone's brought up Forza Horizon 4, so I don't really have any good reference for how well it should work.

> Secondly, in BeamNG.drive: it runs amazingly for the most part, but the wheel feels strangely loose, as if it's insensitive to the surrounding environment. Could just be me though.

BeamNG has showed up quite a lot, searching for it in the bug tracker gives 8 open issues (including this one) and 7 closed ones. Although I don't remember anyone mentioning it feeling loose, which is kind of interesting, but unfortunately not a lot to go off. Ideally there would be a direct side-by-side comparison, like "on this track when I run into this wall the wheel jumps like this on Windows but doesn't react on Linux" or something along those lines. At that point we could try and pinpoint if the difference is in this driver or in how Proton translates FFB effects or if it's in BeamNG itself.

Out of curiosity, did you run the native BeamNG version or through Proton?

samsunix, 2025-04-15:
@Oweblow 
No direct experience with FH4 but FH5 is one big question mark when it comes wheel support. I think it safe to assume it's same with FH4. What I found helpful after launching game, try different usb ports for a wheel. This 'sometimes' trigger game detect wheel correctly (It was same on windows side).

Also try setup dead zones, this sometimes help narrow the issue with pedal. And yes I would recommend mapping pedals again as this could be easiest solution.

I would also give try different proton versions eg. Glorious Eggroll and if you are willing try arch / cachyos: cachyos-proton.
(hyprland example takes 1.2Gb Ram and 0% CPU when running idle - which is very nice. 

That said lack of proper controller support and missing VR support, I personally drop game from my game library. Fact is this is "xbox game" so game-pad is safest choose. 

If you like open world racing keep eye on Assetto Corsa Evo as they (hopefully) release open world map in upcoming months. 

@Kimplul _pst pst, is there any updates coming up?_ ;)

Kimplul, 2025-04-15:
> @Kimplul pst pst, is there any updates coming up? ;)

Perhaps. Easter's coming up, meaning a long weekend, so I'll probably have time to look at my backlog a bit.

Oweblow, 2025-04-15:
@Kimplul I tried remapping the acceleration in Forza Horizon 4 before making this bug report, and it was acting like it couldn't see my acceleration pedal (I pressed down on it when I was remapping it, but the prompt was still there). The brake pedal works fine.
I should probably mention that the loose wheel issue only happens when I'm running the Windows version of BeamNG.drive underneath GEProton 9.25, if I'm running the native Linux port stashed in the files: the wheel feels fine.
I actually have the normal Assetto Corsa in my game library, just trying to get it to run underneath Linux with Proton.
To be honest, I only like playing most of these games to explore the map, racing with a wheel is hellish (racing with a keyboard is better).

Oweblow, 2025-04-15:
Also, OMSI 2 works perfectly fine with this driver (and wheel)!

Oweblow, 2025-04-18:
Anyway, about trying Arch Linux/CachyOS, I think I'm good.
I don't want to have to set up Linux again, and if I had to and went with an Arch distro, it'd probably be Manjaro Cinnamon so it doesn't break every time there's a new update released (cough cough normal Arch).
I like using Debian distros anyway.

Also, Hyperland looks really nice but I think I'll stick to a normal desktop environment with bells and whistles (like Cinnamon) because it's more convinient.

[image]
Do you like it?

Anyway, I'll go about trying the CachyOS fork of Proton.

Oweblow, 2025-04-18:
I tried the CachyOS fork of Proton on FH4, and: it didn't make a difference! ...sadly.
I would've also tried with BeamNG.drive but I've chosen to just keep using the native Linux port.

I think it's probably a bug with the driver.

Kimplul, 2025-04-19:
> (cough cough normal Arch)

Dunno about that, I've had Arch for years without any significant issues. I think it being unstable is kind of just a meme and shouldn't be taken too seriously, at least if you know what you're doing.

> Do you like it?

Looks nice. I think the shading is a bit too strong in the upper right corner, but the `owen` in the opposite corner is a kind of neat touch.

Anycase, nothing wrong with a bit of chit-chat, but this issue is a bit all over the place and hard to follow. IIUC the main issue here is FH4 not handling the gas pedal properly, and you've tried a couple different versions of Proton to debug it (the CachyOS fork and something else...?)

> I think it's probably a bug with the driver.

Possible, sure, but I don't think there's enough data to make any conclusions yet. For instance, we don't know if this happens with other wheels, or if it only happens with Thrustmaster wheels or just the T128. I tried to look through https://www.protondb.com/app/1293830 but no mention of the word `wheel` so hard to say at the moment.

Oweblow, 2025-04-19:
Oki doki.
Anyway, cool driver!

Oweblow, 2025-04-19:
> I tried the CachyOS fork of Proton on FH4, and: it didn't make a difference! ...sadly.
Also, yeah.