Kimplul/hid-tmff2 #157: T300 defaults to 900° rotation

Mattyan1989, 2025-05-03:
First of all, thank you for your work on this driver, which allowed me to finally ditch Windows since simracing was the only reason I was dual booting. However, I have encountered a small issue with my T300GT Edition. Every time I turn on my computer, the steering wheel sets itself to 900° of rotation. I usually prefer it to be at its maximum (1080°) as default and set it lower if I ever need it for a specific game. Normally, I use Oversteer to load a profile with the default rotation setting, but I often forget to do this until I get in game and notice the wheel is set wrong. Could it be possible to set the rotation at a driver level somehow so it is automatic every time I connect it or reboot? Thank you.

Kimplul, 2025-05-03:
Try creating a file `/etc/modprobe.d/hid-tmff-new.conf`, and populate it with
```
options hid-tmff-new range=1080
```

This should tell Linux to set the range to 1080 whenever the module is loaded. I haven't tested it, so please do report back if it worked.

Mattyan1989, 2025-05-04:
It works, thank you! I was curious, so I checked if there were other parameters.
These are my findings in case someone comes across this in the future:
`modinfo hid-tmff-new | grep '^parm'`
These are available:
open_mode - Whether to send mode change commands on open/close (int)
timer_msecs - Timer resolution in msecs (int)
spring_level - Level of spring force (0-100), as per Oversteer standards (int)
damper_level - Level of damper force (0-100), as per Oversteer standards (int)
friction_level - Level of friction force (0-100), as per Oversteer standards (int)
range - Range of wheel, depends on the wheel. Invalid values are ignored (int)
alt_mode - Alternate mode, eg. F1 mode (int)
gain - Level of gain (0-65535) (int)

If more parameters are necessary, they need to be written on the same line, for example:
`options hid-tmff-new range=1080 spring_level=50`

Any modification can be tested without rebooting or disconnecting the wheel by reloading the module:
`sudo rmmod hid-tmff-new`
and
`sudo modprobe hid-tmff-new`

closed 2025-05-04 by Kimplul