Kimplul/hid-tmff2 #160: T128X no longer recognized

ghost-dad, 2025-05-22:
Hi,

I've plugged in my T128X wheel after a good while and saw the wheel not being detected by Oversteer anymore. I do expect this to be related to kernel updates and API changes reported by some other people in here (eg. #149, #150, #140, #142), but nevertheless am facing issues in getting the wheel to work.

### Current state
I'm on CachyOS, all up to date.

**Oversteer** is not seeing any device at all. **lsusb** sees the wheel just fine: `Bus 001 Device 016: ID 044f:b69c ThrustMaster, Inc. Thrustmaster T128X GIP Racing Wheel`. *dkms* status seems to think everything is up to date: `git submodule update --recursive --remote`. **dmesg** input line seems to treat the wheel as a generic Xbox gamepad of all things.

```
[  109.727204] usb 1-4: new full-speed USB device number 10 using xhci_hcd
[  109.854026] usb 1-4: New USB device found, idVendor=044f, idProduct=b69c, bcdDevice= 1.01
[  109.854042] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  109.854049] usb 1-4: Product: Thrustmaster T128X GIP Racing Wheel
[  109.854055] usb 1-4: Manufacturer: Thrustmaster
[  109.854059] usb 1-4: SerialNumber: 0000D32EE1D7FF91
[  109.876638] input: Generic X-Box pad as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/input/input30
[  109.876722] usbcore: registered new interface driver xpad
[  109.879664] usbcore: registered new interface driver usb-thrustmaster
[  119.333859] [UFW BLOCK] IN=eno1 OUT= MAC=e0:d5:5e:aa:e1:c3:ac:f8:cc:8a:bc:e0:08:00 SRC=193.39.142.150 DST=192.168.0.194 LEN=132 TOS=0x08 PREC=0x00 TTL=52 ID=34941 PROTO=UDP SPT=48646 DPT=46839 LEN=112 
[  133.617356] [UFW BLOCK] IN=eno1 OUT= MAC=01:00:5e:00:00:01:ac:f8:cc:8a:bc:e0:08:00 SRC=192.168.0.1 DST=224.0.0.1 LEN=36 TOS=0x00 PREC=0xC0 TTL=1 ID=0 PROTO=2
```

Booting up a game seems to confirm this: instead of working as a wheel with no FFB (as usually happens when something is off with the drivers), stuff is binded all over the place in a way that really does seem like it was perceived as a gamepad.

###What I've done
Basically reinstalled everything: the driver, Oversteer itself, the dependencies. Cleared old files with `sudo dkms remove hid-tmff2/0.82 --all` and purged whatever dregs of AUR packages I've had back in the day (I believe I've already overriden them with DKMS version sometime back in 2024 when troubleshooting some other issue).

I've ran `git pull` and `git submodule update --recursive --remote` to make sure stuff is up to date. Not an expert on git, but the terminal seems to have indeed confirmed pulling some files, so that seems good.

The message I get when installing is
```Deprecated feature: CLEAN (/usr/src/hid-tmff2-0.82/dkms.conf)
Creating symlink /var/lib/dkms/hid-tmff2/0.82/source -> /usr/src/hid-tmff2-0.82
Deprecated feature: CLEAN (/var/lib/dkms/hid-tmff2/0.82/source/dkms.conf)
Sign command: /usr/lib/modules/6.14.7-5-cachyos/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module(s)...... done.
Signing module /var/lib/dkms/hid-tmff2/0.82/build/deps/hid-tminit/hid-tminit-new.ko
Signing module /var/lib/dkms/hid-tmff2/0.82/build/deps/hid-tminit/usb-tminit-new.ko
Signing module /var/lib/dkms/hid-tmff2/0.82/build/hid-tmff-new.ko
Deprecated feature: CLEAN (/var/lib/dkms/hid-tmff2/0.82/source/dkms.conf)
Installing /usr/lib/modules/6.14.7-5-cachyos/updates/dkms/hid-tminit-new.ko.zst
Installing /usr/lib/modules/6.14.7-5-cachyos/updates/dkms/usb-tminit-new.ko.zst
Installing /usr/lib/modules/6.14.7-5-cachyos/updates/dkms/hid-tmff-new.ko.zst
Running depmod... done.
```

It was messier before (not running dkms remove on earlier files, etc.). That repeated 'deprecated feature: CLEAN' line for dmks.conf is sure raising an eyebrow, but other than that it doesn't show any outright errors.

### What now?
Any ideas how to proceed further? My gut says there's probably something left that refuses to update cleanly, but I'm running out of ideas as to where to prod.

Kimplul, 2025-05-22:
Seems like `xpad` is taking control of the wheel. I suspect it's related to the fairly recent commit https://github.com/paroj/xpad/commit/0e6a710bf8d2b19e64c04ad0f6b307c27b6846bc

A quick workaround would be to just disable the `xpad` module with something like
```
echo 'blacklist xpad' | sudo tee /etc/modprobe.d/xpad.conf
```

Might need a reboot. If the wheel starts working, I would suggest opening up another issue over in https://github.com/paroj/xpad and let them know that Thrustmaster wheels are already spoken for :)

There was a similar issue but with PlayStation wheels over in https://github.com/libsdl-org/SDL/pull/12173 that is maybe of interest.

ghost-dad, 2025-05-22:
Worked like a charm, case closed.

One worry I had was that I'll have to keep creating and deleting that blacklist file to actually use my xinput gamepad for other games, but a quick test seems to have it working just fine.

Thanks for the quick response!

closed 2025-05-22 by ghost-dad