Kimplul/hid-tmff2 #168: TS-PC install issue

Kimplul, 2025-07-25:
Hi, I moved your issue to a new thread since the original is getting a bit messy 

> Hello there,
> 
> First, I'm a beginner regarding Linux. I use Zorin OS 17.3 Core on Ryzen 9 5900 + 32GB RAM + RTX4070.
> I own a TS-PC with both the Ferrari 488 Challenge Wheel and TM OPEN WHEEL ADD-ON.
> I mostly play on BeamNG

Welcome to the Linux community, sorry you had to jump into the deep end with kernel modules. Hopefully you consider it a learning experience and not a too bad inconvenience :)

> I installed Oversteer from @Potajito and followed every debian related step from https://github.com/Kimplul/hid-tmff2 since I read the TS-PC branch was included in the master.

Sorry, the TS-PC still lives in its own branch, `tspc`. TS-PC support is not quite as 'polished' as I'd like to allow it to be in the `master` branch, so you'll have to run something like

```
git clone https://github.com/Kimplul/hid-tmff2.git
cd hid-tmff2
git checkout tspc
git submodule update --init
sudo ./dkms/dkms_install.sh
sudo make udev-rules
```

Although admittedly it seems that a fair number of people have the TS-PC and it might benefit from getting merged into `master`, I might look into it in the near future.

Out of curiosity, what made you think the TS-PC was supported in `master`? If it's something I can edit to be more clear, I shall.

> My issue is I cannot see my wheel in oversteer, I see it in Joystick but I have the strange jump in the response around the wheel's range center. Also pedals are treated as buttons instead of slider.

I believe these problems should disappear with the `tspc` branch. In short, by not using the `tspc` branch, your driver doesn't support the TS-PC, and the wheel without a driver behaves badly. I can go into a bit more detail if you're curious, but it's not important to understand by any means.

> I read the entire thread but did not understood everything, am I missing something? What information should I brought to better explain my situation?

I don't know if this is your first time opening up an issue, but if it is, well done. These are the types of questions maintainers love to see :)

Hopefully switching to the `tspc` branch fixes the issues you're having, but if it doesn't, please show the output of `sudo dmesg -w` when plugging in the wheel.

> Thank you. 

 _Originally posted by @Jaileh in [#65](https://github.com/Kimplul/hid-tmff2/issues/65#issuecomment-3120273511)_

Jaileh, 2025-07-26:
This is the error I tried to correct:
```
jaileh@jaileh-desktop:~/hid-tmff2$ sudo ./dkms/dkms-install.sh
Error! DKMS tree already contains: hid-tmff2-0.82
You cannot add the same module/version combo more than once.
Module hid-tmff2/0.82 already built for kernel 6.8.0-64-generic (x86_64).
Module hid-tmff2/0.82 already installed on kernel 6.8.0-64-generic (x86_64).
```

mitusgamer12, 2025-09-03:
Hello everyone. I just bought a TS-PC, and I'm trying to get it workiong on linux too. I know close to nothing about drivers on linux, so I will just tell what I did. I followed these commands: 

> git clone https://github.com/Kimplul/hid-tmff2.git
cd hid-tmff2
git checkout tspc
git submodule update --init
sudo ./dkms/dkms_install.sh
sudo make udev-rules

the driver seems to have installed properly, but i get the same errors jaileh had. Oversteer can't see the wheel and the pedals are buttons. The ffb doesn't work on any game and it's set to only 180º, and I can't change it. here is what dmesg -w returns:

`[    3.117711] hub 6-0:1.0: USB hub found
[    3.117717] hub 6-0:1.0: 2 ports detected
[    3.360465] usb 1-4: new full-speed USB device number 2 using xhci_hcd
[    3.364442] usb 3-2: new full-speed USB device number 2 using xhci_hcd
[    3.364462] usb 5-1: new full-speed USB device number 2 using xhci_hcd
[    3.517498] usb 3-2: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 7.00
[    3.517506] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.517509] usb 3-2: Product: Thrustmaster FFB Wheel
[    3.517511] usb 3-2: Manufacturer: Thrustmaster
[    3.517605] usb 5-1: New USB device found, idVendor=058f, idProduct=9254, bcdDevice= 3.12
[    3.517609] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.517611] usb 5-1: Product: Generic USB Hub
[    3.517613] usb 5-1: Manufacturer: ALCOR
[    3.528478] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    3.529346] ata1.00: ATA-8: WDC WD10EZEX-00RKKA0, 80.00A80, max UDMA/133
[    3.530244] ata1.00: 1953525168 sectors, multi 16: LBA48 NCQ (depth 32), AA
[    3.531172] ata1.00: configured for UDMA/133
[    3.531338] scsi 0:0:0:0: Direct-Access     ATA      WDC WD10EZEX-00R 0A80 PQ: 0 ANSI: 5
[    3.566597] hub 5-1:1.0: USB hub found
[    3.569503] hub 5-1:1.0: 4 ports detected
[    3.880495] usb 5-1.3: new low-speed USB device number 3 using xhci_hcd
[    3.997590] usb 5-1.3: New USB device found, idVendor=10d5, idProduct=55a2, bcdDevice= 0.01
[    3.997598] usb 5-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.997601] usb 5-1.3: Product: 2Port KVMSwitcher
[    3.997604] usb 5-1.3: Manufacturer: Device
[    3.997606] usb 5-1.3: SerialNumber: 1"²
[    4.000433] ata2: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[    4.007040] ata2.00: ATA-7: SAMSUNG HD103SI, 1AG01118, max UDMA7
[    4.007468] ata2.00: 1953525168 sectors, multi 16: LBA48 NCQ (depth 32), AA
[    4.014119] ata2.00: configured for UDMA/133
[    4.014292] scsi 1:0:0:0: Direct-Access     ATA      SAMSUNG HD103SI  1118 PQ: 0 ANSI: 5
[    4.480491] ata5: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    4.481608] ata5.00: ATA-10: SanDisk SSD PLUS 240GB, UF2204RL, max UDMA/133
[    4.481616] ata5.00: 468862128 sectors, multi 1: LBA48 NCQ (depth 32)
[    4.483314] ata5.00: Features: Dev-Sleep
[    4.491070] ata5.00: configured for UDMA/133
[    4.491265] scsi 4:0:0:0: Direct-Access     ATA      SanDisk SSD PLUS 04RL PQ: 0 ANSI: 5
[    4.802008] ata6: SATA link down (SStatus 0 SControl 330)
[    4.804746] r8169 0000:06:00.0 enp6s0: renamed from eth0
[    4.811823] sd 1:0:0:0: [sdb] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    4.811823] sd 4:0:0:0: [sdc] 468862128 512-byte logical blocks: (240 GB/224 GiB)
[    4.811831] sd 0:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    4.811833] sd 4:0:0:0: [sdc] Write Protect is off
[    4.811833] sd 1:0:0:0: [sdb] Write Protect is off
[    4.811835] sd 1:0:0:0: [sdb] Mode Sense: 00 3a 00 00
[    4.811835] sd 4:0:0:0: [sdc] Mode Sense: 00 3a 00 00
[    4.811837] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    4.811850] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    4.811850] sd 0:0:0:0: [sda] Write Protect is off
[    4.811852] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    4.811865] sd 4:0:0:0: [sdc] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    4.811872] sd 1:0:0:0: [sdb] Preferred minimum I/O size 512 bytes
[    4.811872] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    4.811887] sd 4:0:0:0: [sdc] Preferred minimum I/O size 512 bytes
[    4.811913] sd 0:0:0:0: [sda] Preferred minimum I/O size 4096 bytes
[    4.818239] hid: raw HID events driver (C) Jiri Kosina
[    4.873424]  sdc: sdc1 sdc2 sdc3 sdc4
[    4.873673] sd 4:0:0:0: [sdc] Attached SCSI disk
[    4.921723]  sdb:
[    4.921758] sd 1:0:0:0: [sdb] Attached SCSI disk
[    4.928316]  sda: sda1
[    4.928509] sd 0:0:0:0: [sda] Attached SCSI disk
[    4.945425] usbhid 5-1.3:1.1: couldn't find an input interrupt endpoint
[    4.945483] usbcore: registered new interface driver usbhid
[    4.945486] usbhid: USB HID core driver
[    4.947670] input: Device 2Port KVMSwitcher as /devices/pci0000:00/0000:00:08.1/0000:08:00.4/usb5/5-1/5-1.3/5-1.3:1.0/0003:10D5:55A2.0002/input/input3
[    5.104599] hid-generic 0003:10D5:55A2.0002: input,hidraw0: USB HID v1.00 Keyboard [Device 2Port KVMSwitcher] on usb-0000:08:00.4-1.3/input0
[    5.106389] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-2/3-2:1.0/0003:044F:B65D.0001/input/input4
[    5.106495] hid-thrustmaster 0003:044F:B65D.0001: input,hidraw1: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:08:00.3-2/input0
[    5.128318] hid-thrustmaster 0003:044F:B65D.0001: Unknown wheel's model id 0x609, unable to proceed further with wheel init
[    6.839447] [drm] amdgpu kernel modesetting enabled.
`

any help is really appreciated. Thanks in advance!

Kimplul, 2025-09-03:
Hi @mitusgamer12, try running
```
echo 'blacklist hid_thrustmaster' | sudo tee /etc/modprobe.d/hid_thrustmaster.conf
```

It looks like the wheel is being picked up by the in-kernel `hid-thrustmaster`, which doesn't know what to do with the TS-PC. Adding it to a blacklist should hopefully allow `hid-tminit` to take over.

Kimplul, 2025-09-03:
@mitusgamer12 Did you already reboot? I'm guessing `hid-thrustmaster` stayed loaded and 'blocked' `hid-tminit` (they share the same module identifier at the moment, it's admittedly a bit confusing), rebooting should clear things.

mitusgamer12, 2025-09-03:
Thanks again for the reply. I rebooted, but the error persists. By the way, I forgot to mention that I'm on Debian 13. Maybe I could try to completely remove the hid-thrustmaster driver. Is there any way to do so? Thanks!

Kimplul, 2025-09-03:
Oh, interesting. I guess that makes sense if you keep the wheel plugged in during a reboot, since the kernel probably has `hid-thrustmaster` in its initrd and sees the blacklist only later on during boot? I always unplug my controllers so I hadn't even considered that it might be an issue. Thanks for bringing it up, this is very useful information.

Hopefully the nonsense with blacklisting modules will be resolved in the near-ish future, there are some upstreaming efforts ongoing but I'm not entirely sure about their current status.

mitusgamer12, 2025-09-04:
Yeah, I keep all my peripherals connected all the time, so it makes sense that it worked for you but not for me. I'm glad I can be of help!
Also, thanks a lot for these amazing drivers, I've been playing a bit and it feels just like, if not even better than on windows. Keep it on!

Jaileh, 2025-11-13:
In case anyone need to get the TS-PC driver on a fresh debian (or derivated) config as I do, here is the full script

```bash
# prerequisites
sudo apt install linux-headers-$(uname -r)
sudo apt install joystick -y
sudo apt install git -y
sudo apt install make -y
sudo apt install dkms -y

# actual driver installation
git clone https://github.com/Kimplul/hid-tmff2.git
cd hid-tmff2
git checkout tspc
git submodule update --init
sudo ./dkms/dkms-install.sh
sudo make udev-rules
echo 'blacklist hid_thrustmaster' | sudo tee /etc/modprobe.d/hid_thrustmaster.conf
```